trabucayre/openFPGALoader #336: Tang Primer 20K

vpecanins, 2023-04-21:
Hello,

I just wanted to let you know that i couldn't use the latest version of openFPGALoader to work on the Sipeed Tang Primer 20K board.

However, it works correctly (flash and sram programming) if i revert to the following commit:

git checkout 94d0ad6c851f691b27ed48884744dcaeb051e734

In the current version, the function Gowin::spi_put in gowin.cpp appears to be broken and causes stack smashing. 

Lastly, i wanted to let you know that at the moment, openFPGALoader seems to be the only tool for Sipeed/GOWIN boards since the original programmer tool from GOWIN does not work in Linux.

trabucayre, 2023-04-21:
Hello,
Thanks for this information!
Could you paste a dump?
This commit is related to a class unused by gowin part so I assume issue is somewhere between this commit and HEAD.

I have to investigate all commits to find the one with impact.

nickoe, 2023-08-24:
@vpecanins Is this symptom different for what is observed in https://github.com/trabucayre/openFPGALoader/issues/300 ?

wingrime, 2023-09-04:
I confirm issue on my PC, writing some bitstreams hungs

solution from #300 - reverting commit helped me