trabucayre/openFPGALoader #338: Supporting Kria , FTDI4232

SemiramisTehrani, 2023-04-25:
Hi ,
I know Kria is not listed in the supported device, But I ft4232 is listed in the cable list. Below is what I get 

$ openFPGALoader -c ft4232 design.bit
Jtag frequency : requested 6.00MHz -> real 6.00MHz
Error : no device found

$ openFPGALoader -V
openFPGALoader v0.10.0

Is there any way to add Kria to the supported devices list? or anyway around this?

thanks,
Semi

SemiramisTehrani, 2023-04-25:
I have FT4232H on my card.

trabucayre, 2023-04-25:
I haven't a Kria based board myself so I can't test.
But the simpliest way is to update [part.hpp](https://github.com/trabucayre/openFPGALoader/blob/master/src/part.hpp) with the idcode and 1/ try to detect device 2/try to load a bitstream into SRAM
SPI flash is another story because, I suppose, this device needs a bridge between JTAG and SPI flash.

SemiramisTehrani, 2023-04-25:
Hi , thanks for your quick response. I will follow your instructions and will post the results.  thanks, Semi

SemiramisTehrani, 2023-04-26:
Hi , 
I updated the part.hpp , using vivado to extract the idcode and ir values.  rebuild the tool , and tried to load a bitstream didn't got through because I still get " Error : no device found". 
I did scan usb , I got correct info of my ft4232 , vid, & vip valuse , and carrier card info. 
I also checked  cable.phh . Our device is ft4232H , do I need to add it to the cable.pph because ft4232 is only listed?

I also used a Zedboard as a sanity check , using hsa_2 for cable ( my cable is just simple USB) and it worked out. 

I appreciate your feedback. 

thanks, Semi

SemiramisTehrani, 2023-04-27:
Hi , 

I am using NVIDIA , running Linux Ubuntu 18.04  connecting to Kria (carrier card) with just a usb cable.  As you already mentioned the JTAG signals have a level shifter between FTDI and FPGA shown on the schematic. 

Here is  when I post lsusb 
Bus 001 Device 005 : ID 0403 : 6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC

And also my keyboard & mouse are listed that I think you are not interested ?

thanks,
Semi

SemiramisTehrani, 2023-04-27:
Hi again , 
I also just used JTAG-HS3 Rev.A (tiny adaptor) connecting to JTAG headheader on the carrier card and connect my usb cable to it. I have original adaptor , and one that I remove the EEPROM. I get same error result for both. 
lsusb reports
 Bus 001 Device 005 : ID 0403 : 6014 Future Technology Devices International, Ltd FT232H Single HS USB-UART/FIFO IC 

however when I use below command

openFPGALoader -c digilent-hs3  design.bit   
error : digilent-hs3 not found

thanks,
Semi

trabucayre, 2023-04-27:
your `lsusb`shows a classic ft4232 -> no informations about this interface :( (it's classic for most vendors: no pinout informations and no custom VID/PID). I have to search a bit more about this interface but without the board it's a bit more difficult.

for your second try `digilent-hs3`-> `digilent_hs3`

trabucayre, 2023-04-27:
Ok! This mean your able to communicate with the board.
I didn't notice before but kria is based on MPSoC device: there is 2 distincts idcode (one when the device is not configured, one after: see [this note](https://github.com/trabucayre/openFPGALoader/blob/master/src/part.hpp#L75)).

SemiramisTehrani, 2023-04-27:
Hi , thanks for your response. 
I added following to the part.hpp & it all worked :-) 
{0x04724093, {"xilinx", "zynqmp",      "xck26", 12}},
{0x08e48126, {"xilinx", "zynqmp_cfgn",      "xck26", 4}},

I had a typo  and a couple of other errors to fix when added above lines , finally it re-built with no errors. 

thanks,
Semi

trabucayre, 2023-04-27:
Great!
Could you open a PR to add this zynqmp version ?
kria needs to find correct FTDI configuration...

SemiramisTehrani, 2023-05-08:
Hi , 
I had to do a re-work on the base card to force the boot to be JTAG . By default the base card boot mode is QSPI. 

spiOverJtag : 
Now , I need to verify  spiOverJtag programming , already added the board info . Using below commands 

openFPGALoader -b kria  design.bit -----> successfully programmed

openFPGALoader -b kria -f design.bit
Error: Failed to claim FPGA device: Error: can't flash ZynqMP 

do I need to add a constraint file for my Kria FPGA?

thanks,
Semi