trabucayre/openFPGALoader #348: Unable to program Zedboard [Error: Failed to claim FPGA device: Error: can't flash Zynq7000]

gian-didom, 2023-06-01:
Hello,

I am trying to program a Zedboard with openfpgaloader. I exported the .blt file from Vivado.
Here is the error message:

`% openfpgaloader -b zedboard sampleProject.blt   
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Error: Failed to claim FPGA device: Error: can't flash Zynq7000`

It seems that the board is not seen properly. If I scan for connected devices, however, I see this output:

`% openfpgaloader --scan-usb
found 1 USB device
Bus device vid:pid       probe type      manufacturer serial               product
001 001    0x0403:0x6014 ft232H          Digilent     210248774932         Digilent USB Device`

If I try to select the cable instead of the board, the following output appears:

`openfpgaloader -c digilent_ad sampleProject.blt
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Error: no device found
`

If I run the detect command with the zedboard, I get:

`openfpgaloader -b zedboard --detect
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
index 0:
	idcode   0x4ba00477
	type     ARM cortex A9
	irlength 4
index 1:
	idcode 0x3727093
	manufacturer xilinx
	family zynq
	model  xc7z020
	irlength 6
`

I still have to figure out if the board is seen or not.

**System info**
**Laptop:** MacBook Pro M1 Pro
**OS:** MacOS Ventura

trabucayre, 2023-06-01:
Hello,
As mentioned in this [issue](https://github.com/trabucayre/openFPGALoader/issues/343) it's not an internal error but simply because SPI flash is connected to PS pins and can't be accessed from PL.
I have updated `xilinx.cpp` to have a more explicit message.

Thanks!

gian-didom, 2023-06-01:
Please note that I'm not using the -f flag

trabucayre, 2023-06-01:
Sorry. I have seen this error and immediately think about this issue.
Could you try with` -b zedboard` ?
I have retested and no issue, but I have used an Linux computer. It's weird because default behavior is volatile memory.

gian-didom, 2023-06-01:
I think that the script itself tries to write to volatile memory, because a message appears when using -f instead.
I will try with a different cable. Do both JTAG and UART ports of the ZedBoard need to be connected?
I'll also try with a Linux computer.

trabucayre, 2023-06-02:
I think I found why openFPGALoader try to write to the flash:
In your command line you have a bitstream called *sampleProject.blt* with **BLT** as extension instead of **BIT**.
If you check this couple of [lines](https://github.com/trabucayre/openFPGALoader/blob/master/src/xilinx.cpp#LL141C1-L156C3) target choise is based on user information + extension: a *.blt* extension isn't a know bitstream file and considered as a raw binary data (can't be used for SRAM mode) to write somewhere into the flash.