trabucayre/openFPGALoader #355: `JTAG init failed with` error. ByteBlasterII, Altera Max

raleighlittles, 2023-07-12:
I'm using the latest release build (v0.10.0 from the Releases page)

Trying to run this command:

```bash
$ ./openFPGALoader --detect --cable usb-blasterII --probe-firmware ~/Chrome-Downloads/blaster_6810.hex
```

I get this output:
```
USB-Blaster II firmware version: 1.27
JTAG init failed with: Unknown device with IDCODE: 0x041481bb (manufacturer: 0x0dd (), part: 0x20 vers: 0x0
```

I'm on Ubuntu 20.04.6 LTS with kernel 5.15.0-75

-----------------------------------------------------

Heres the lsusb output for my device

```
Bus 001 Device 013: ID 09fb:6010 Altera 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x09fb Altera
  idProduct          0x6010 
  bcdDevice            1.27
  iManufacturer           1 Altera
  iProduct                2 USB-BlasterII
  iSerial                 3 UBII-000015721
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0037
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x88  EP 8 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

The FPGA is an Altera Max V, with exact board info:

```
020A40DD   5M(1270ZF324|2210Z)/EPM2210
```

trabucayre, 2023-07-19:
In fact this issue is not fully related to the cable but to the device too: max V isn't supported.
A weird thing here: an extra LSB was added (  0x20A40DD == 0x041481bb >> 1).

trabucayre, 2023-07-24:
Could you also try with master branch: a recent commit may have fixed align issue.
For MAXV by itself I maybe need one to add / update support.