trabucayre/openFPGALoader #361: Support for Lattice ECP3?

michaeltraxler, 2023-08-05:
Lattice ECP5 and MachXO3 works very well with openFPGALoader. How can I help to add support for the Lattice ECP3? I thought it is the "same" as ECP5, but maybe not. At least I can help in testing...

trabucayre, 2023-08-06:
Most of lattice family have more or less same programming sequence.
So:
1. adding idcode in `part.hpp`
2. comparing for SRAM mode sequence for ECP5 and sequence for ECP3 and simply try to load a bitstream
3. same thing with SPI flash access (maybe simply trying since this part is not documented by Lattice).

ECP3 eval boards seems hard to find/expensive :-/

hattorihanso, 2023-11-28:
It is a French board:
http://tachyssema.com/old/download.php?view.15
I have recovered it from some "to be thrown away" pile, but it is working and has all three addon modules.

trabucayre, 2023-11-28:
Hi,
Yes: if it's possible to have physically access to a ECP3 board I'm interested.
I have compared ECP3's bitstream with one for ECP5: some area seems reversed, maybe it's just something required for parsing but not to load/write but without being able to test it's a bit hard to say.
Thanks!

joergfoertsch, 2024-07-23:
Hi, 
I saw there is an active branch trying to get the ECP3 programming to work.
Since I am also interested in this feature I tested the current status of said branch trying to program an LFE3-150EA adding it to part.hpp. 
The program ran through, but the ECP3 did not respond as if it were programmed.
Hence I was wondering if I did something wrong or if there is simply more to do/help to get this feature running :)
Many thanks for your efforts!

trabucayre, 2024-07-28:
Hi,
Yes I try to implement configuration flow but there is no documentation on how to load bitstream into memory and my only source is svf file.
For some reason when I try to use directly the svf file loading fails and loading process using implementation seems to works but the FPGA seems not using new bitstream...
So this branch is a WIP and I needs to find what is missing.

trabucayre, 2025-06-17:
@michaeltraxler , @joergfoertsch I know I have took lot of time to continue ECP3 support, but now I'm able to load a bitstream (SRAM).
The branch has been updated, most of the work was to fixes delay after some instruction/operation.

michaeltraxler, 2025-06-17:
Hello Gwenhael, 
these are great news! Thanks a lot for digging deep into this and in the end solving it! We definitely really appreciate it and will use it for sure in our setups!

joergfoertsch, 2025-07-01:
Many thanks, Gwenhael.
I also finally came to testing and it works like a charm :)