trabucayre/openFPGALoader #365: Not support chip

daobinhminh, 2023-08-10:
Why this source code not support for vu35p chip.
What I need to do to use with my vu35p

trabucayre, 2023-08-10:
I assume you receive a message about unknown device. `xcvu9p` and `xcvu37p` are known but idcode for 35p is missing.
The simple way to validate support is to add juste after this [line](https://github.com/trabucayre/openFPGALoader/blob/master/src/part.hpp#L99) an entry with the idcode of you chip and test if your able to load and flash a bitstream.

Usually idcode (ie support) are added once one has validate openFPGALoader is working with a family/device

daobinhminh, 2023-08-11:
I add idcode of vu35p and it load bit done. But it's still not working. Can you figure out what not true with this

{0x04b31093, {"xilinx", "virtexusp", "xcvu9p",  18}},
{0x14b79093, {"xilinx", "virtexusp", "xcvu37p", 18}},
{0x14b77093, {"xilinx", "virtexusp", "xcvu35p", 12}},

daobinhminh, 2023-08-11:
Can I load multiple bit using openFPGALoader tool. I load 1 .bit file it work but when I load 8 .bit file it not.

trabucayre, 2023-08-24:
I assume bit are working when you use vendor tools?
For multiple bitstreams could you be more precise?