trabucayre/openFPGALoader #367: Support load partial bitstream

ducdracaena, 2023-08-21:
Hello, my project use partial reconfigurtion for some location in the chip. However, when I try to load the static bitstream and partial bitstream. it doesnt work. So, how can I solve it?

Thanks,
Duc

trabucayre, 2023-08-24:
Could you precise vendor/family FPGA?
I have never tried that and I'm not sure to have hardware supporting this feature. But maybe support may be done without hardware.

ducdracaena, 2023-08-24:
The vendor is xilinx
family FPGA is virtex ultrascale + vu35p

ZhiyuanYuanNJ, 2024-10-23:
> Could you precise vendor/family FPGA? I have never tried that and I'm not sure to have hardware supporting this feature. But maybe support may be done without hardware.

Hi, How can I generate a bitstream file? The FPGA currently in use is compatible with Xilinx xc7vx690t, but I tried [spiOverJtag_xc7vx330tffg1157.bit.gz](https://github.com/trabucayre/openFPGALoader/blob/c0faf48d46aa807af3ce6a61300e2a64d71c240a/spiOverJtag/spiOverJtag_xc7vx330tffg1157.bit.gz), but it didn't work and may not match this FPGA.