trabucayre/openFPGALoader #373: idCode detection failed on ULX3S

gamelaster, 2023-09-06:
Hello, when using openFPGALoader v0.11.0, both on Windows and Linux:
```
openFPGAloader --board ulx3s ulx3s.bit
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
Open file: DONE
Parse file: DONE
mismatch between target's idcode and bitstream idcode
        bitstream has 0x41113043 hardware requires 0xffffffff
Error: Failed to program FPGA: std::exception
```
This means that hardware returns 0xffffffff idCode, what is obviously wrong. `fujprog` does not have any issues to detect proper idCode and program the bitstream.

trabucayre, 2023-09-06:
I have just checked with master without issue.
Could you provides output for `openFPGALoader --scan-usb` maybe you have more than one *ftdi* with `VID:PID = 0x0403:0x6011`.
If ti's your case could you try with `openFPGAloader --board ulx3s ulx3s.bit ----busdev-num=0xAA:0xBB` where AA is the value for the first column of `--scan-usb` and BB is the value for the second column.

Thanks

gamelaster, 2023-09-06:
Hi @trabucayre and thanks for fast reply.
I indeed have more compatible devices on my Windows machine, but on Linux, I had only this one FTDI device connected.
Logs:
```
$ openFPGAloader --scan-usb
found 21 USB device
Bus device vid:pid       probe type      manufacturer serial               product
Error: can't open device with vid:vid = 0x1366:0x0105. Error code -5 Entity not found
002 021    0x0403:0x6015 ft231X          FER-RADIONA-EMARD D00523               ULX3S FPGA 85K v3.0.8

$ openFPGAloader --board ulx3s ulx3s.bit --busdev-num=0x0403:0x6015
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
Open file: DONE
Parse file: DONE
mismatch between target's idcode and bitstream idcode
        bitstream has 0x41113043 hardware requires 0xffffffff
Error: Failed to program FPGA: std::exception
```

trabucayre, 2023-09-06:
Could you provides ulx3s version and the output of:
```
openFPGALoader -b ulx3s --detect --verbose-level 2
```

gamelaster, 2023-09-06:
@trabucayre (note, performed on Windows)
```
openFPGALoader -b ulx3s --detect --verbose-level 2
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
Raw IDCODE:
- 0 -> 0x41113043
- 1 -> 0xffffffff
- 2 -> 0xffffffff
- 3 -> 0xffffffff
- 4 -> 0xffffffff
found 1 devices
index 0:
        idcode 0x1113043
        manufacturer lattice
        family ECP5
        model  LFE5UM-85
        irlength 8
```

trabucayre, 2023-09-06:
For windows result seems fine.
Communication is ok and device detected too

gamelaster, 2023-09-06:
@trabucayre this is strange, because the final flashing doesn't work at all:
```
openFPGAloader --board ulx3s ulx3s.bit --verbose-level 2
Jtag probe limited to 3MHz
Jtag frequency : requested 6000000Hz -> real 3000000Hz
ret 0
Raw IDCODE:
- 0 -> 0x41113043
- 1 -> 0xffffffff
- 2 -> 0xffffffff
- 3 -> 0xffffffff
- 4 -> 0xffffffff
found 1 devices
index 0:
        idcode 0x1113043
        manufacturer lattice
        family ECP5
        model  LFE5UM-85
        irlength 8
File type : bit
Open file: DONE
Parse file: DONE
bitstream header infos
Part: LFE5U-85F-6CABGA381
idcode: 41113043
mismatch between target's idcode and bitstream idcode
        bitstream has 0x41113043 hardware requires 0xffffffff
Error: Failed to program FPGA: std::exception
```

Can be the issue, that I use openFPGAloader from OSS CAD suite? (When I will come back home, I will use my own built openFPGAloader on Linux)

gamelaster, 2023-09-06:
@trabucayre I tested openFPGAloader from pacman and also from ubuntu release on my Linux, and it works perfectly! To the opposite, openFPGAloader from oss-cad-suite shows same error. Thus openFPGAloader is broken in the oss-cad-suite.

trabucayre, 2023-09-06:
Have you tested to build manually from this repository?
It's weird: oss-cad-suite uses master so it must works as expected.

gamelaster, 2023-09-06:
@trabucayre I dissected the trouble, and it stops working when 9e91c31e31c0d343a921c69e005cc2a2601ef08e commit is applied.

gamelaster, 2023-09-06:
To be exact, after reverting this line in `jtag.cpp`, it stops working: https://github.com/trabucayre/openFPGALoader/commit/9e91c31e31c0d343a921c69e005cc2a2601ef08e#diff-b303e4ef06663a46e45d8384246a34771c844206fb3372941c3079461f8aafcfR623

trabucayre, 2023-09-06:
I have seen the same thing: fix pushed

closed 2023-09-06 by gamelaster

trabucayre, 2023-09-07:
Thanks for your feedback/help!