trabucayre/openFPGALoader #375: Timeout on Xilinx xc7a50tfgg484 with Spansion s25fl128l and digilent_hs3

peccehu, 2023-09-08:
Hi, I'm trying to get openFPGALoader to work with my custom board; running from a Debian x86 host.

First I had to add the flash definition for Spansion s25fl128l since this was missing. I modified the existing definition of s25fl256l based on the datasheet  ( https://paste.pics/PAPY6 )

```
	{0x016018, {
		.manufacturer = "spansion",
		.model = "S25FL128L",
		.nr_sector = 256,
		.sector_erase = true,
		.subsector_erase = false,
		.has_extended = true,
		.tb_otp = false,
		.tb_offset = (1 << 5),
		.tb_register = STATR,
		.bp_len = 3,
		.bp_offset = {(1 << 2), (1 << 3), (1 << 4), 0}}
	},
```

I haven't made a board definition, but I believe I'm supplying all the needed parameters on the command line?

[devel@hc3test]:~/GIT/openFPGALoader/build$ sudo ./openFPGALoader -c digilent_hs3 -f hc3_bb_wrapper.bit -v --bridge spiOverJtag_xc7a50tfgg484.bit --fpga-part xc7a50tfgg484
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
found 1 devices
index 0:
        idcode 0x362c093
        manufacturer xilinx
        family artix a7 50t
        model  xc7a50t
        irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2021/11/12
design_name: hc3_bb_wrapper
hour: 16:56:22
part_name: 7a50tfgg484
toolVersion: 0XFFFFFFFF;Version=2021.1
userID: TRUE
use: /home/devel/GIT/openFPGALoader/spiOverJtag/spiOverJtag_xc7a50tfgg484.bit
load program
Flash SRAM: [==================================================] 100.00%
Done
1 60 18 ff read 16018ff
Detected: spansion S25FL128L 256 sectors size: 128Mb
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
00000000 00000000 00000000 00
timeout: ff ff ff
ff
wait: Error
Erasing: [                                                  ] 0.00%timeout: ff ff ff
ff
wait: Error
Erasing: [==                                                ] 4.00%timeout: ff ff ff
ff
wait: Error
Erasing: [====                                              ] 8.00%timeout: ff ff ff
ff
wait: Error

I verified the xilinx_spiOverJtag.v and constr_xc7a_fgg484.xdc constraint used for spiOverJtag and the constraints match with my design (right pinout for the spi flash).

Anyway; I can flash the configuration when using Vivado Hardware manager so this verifies my hardware setup is correct.

Am I missing something in the SPI Flash definition? From the datasheet I see that there are some differences between s25fl128l and s25fl256l in register layout (they have seperate tables for both devices) :

(datasheet :  https://www.infineon.com/dgdl/Infineon-S25FL256L_S25FL128L_256-MB_(32-MB)_128-MB_(16-MB)_3.0_V_FL-L_FLASH_MEMORY-DataSheet-v09_00-EN.pdf?fileId=8ac78c8c7d0d8da4017d0ed40e335224  )

thank you,
Marc

peccehu, 2023-09-08:
I had programmed a board with Vivado and when I afterwards tried to flash with openFPGALoader (as in the log above) then I got timeouts during erasing.

After removing JTAG and cycling power on the board, it seemed however that the configuration was lost (based on LED behaviour and power consumption). So it seems that the SPI configuration flash was effectively erased with openFPGALoader; despite the timeout!

I needed to reprogram with Vivado to get the system up again.

trabucayre, 2023-09-10:
Sorry for delay to answer.
I have tried with a board based on xc7a50 device but not the same package nor the same flash without issue.
openFPGALoader is able to see the flash chip after loading spiOverJtag bridge so basic (flash access) is fine.
This chip look similar to any others.
I have to find if I have a board (with different device) but with this specific chip to see (with a physical access) if there is some subtilities here.

peccehu, 2023-09-11:
It seems my Purple scope probe in previous post was broken, so signal was measured on DQ1.
But now I replaced the scope probe and even then; I only get signal on DQ1 (SO) at the beginning of the program execution; and I'm quite sure this is when the serial flash JEDEC ID code (0x016018) is being read out. 
https://paste.pics/PBP43
as in the execution log : "1 60 18 ff read 16018ff"

kind regards,
Marc

peccehu, 2023-09-14:
Hi,
maybe S25FL128L is a bit different from other flashes.
In the datasheet, here is the description of the Block erase command (this is where the openFPGALoader fails) :
https://paste.pics/PCRUQ
WREN must be issued, and also nCS pin must be controlled in order for the block erase to go through.

I modified the program and in the function `SPIFlash::sectors_erase()`  I inserted `write_enable()` command before the `block64_erase()` call. After this I see the WEN bit going high in status register.

However, it seems this is not sufficient for the block erase. I believe nCS must also be controlled in the proper way, but I don't know how to do this.

Could you give me some pointers as to how I can control the SPI Flash nCS line from inside the code? I assume I need to call functions in the `FTDIpp_MPSSE` class which emulates the JTAG TAP interface?

thanks for any help,
Marc

trabucayre, 2023-09-14:
`write_enable` function is called before each `blockxx_erase` call. Description looks similar to others chip.
nCS is handled by FPGA so no way to have access (but spiOverJtag must be improvded).
I will order a couple of this chip to manage tests.

peccehu, 2023-09-14:
You are right; I commented out the line which I added; but left the added `display_status_reg(read_status_reg())`.
The result is :
      RDSR : 02
      WIP  : 0
      WEL  : 1
      BP   : 0
      TB   : 0
      SRWD : 0
Which indeed indicates that write enable is activated.

peccehu, 2023-09-25:
Hi, any luck with this? Could you try with samples of S25FL128L ?

trabucayre, 2023-09-26:
Hi,
I have ordered (and received) two chips, but not already time to solder one of them on something similar to a PMOD to test.

trabucayre, 2023-10-01:
Flash chip has finally been soldered (but not currently in a PMOD form factor) and tested using a tigard (ie. direct SPI access).
In this mode I haven't seen anything wrong: the chip is correctly detected, erased and reprogrammed.
I have to finalize my adaptor and test with a modified spiOverJtag bitstream to see if, something is wrong in this mode. But algorithm implemented in openFPGALoader is working with this device.

Could you check if you chip is configured in SPI or in QSPI? This parameter is stored in non-volatile memory and openFPGALoader support only SPI mode.

peccehu, 2023-10-02:
Thanks for the testing;
When generating the .mcs file in Vivado; I'm always selecting the SPIx4 option; so pretty sure it's using Quad-SPI.
I'm not sure how to change from QSPI to SPI?
kind regards,
Marc

trabucayre, 2024-06-07:
Can I close this issue ?

peccehu, 2024-06-07:
Yes, the issue was due to the fact that my bitstream was generated for SPIx4 instead of SPIx1.

thanks for the support,
Marc

closed 2024-06-07 by trabucayre