trabucayre/openFPGALoader #377: Unable to flash ecpix5

jeremyherbert, 2023-09-11:
Hi,

I am using the latest git master version but I am unable to program my ECPIX5. I get the output `unable to open ftdi device: -3 (device not found)`, however the cable is visible in my lsusb:

[image]

I have tried with:

`openFPGALoader --board ecpix5 --detect`

I have also tried with `openFPGALoader -c ft4232 --detect` and it works correctly:

```
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
index 0:
	idcode 0x1113043
	manufacturer lattice
	family ECP5
	model  LFE5UM-85
	irlength 8
```

However, when I try to write a bit file, I get the following output:

```
$ sudo openFPGALoader -c ft4232 top.bit

Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file: DONE
Parse file: DONE
Enable configuration: mpsse_write: fail to write with error -1 (usb bulk write failed)
mpsse_write: fail to write with error -1 (usb bulk write failed)
mpsse_read: fail to flush buffer with error: -1 (usb bulk write failed)
mpsse_write: fail to write with error -1 (usb bulk write failed)
mpsse_write: fail to write with error -1 (usb bulk write failed)
mpsse_read: fail to flush buffer with error: -1 (usb bulk write failed)
mpsse_write: fail to write with error -1 (usb bulk write failed)
mpsse_read: fail to flush buffer with error: -1 (usb bulk write failed)
FAIL
mpsse_write: fail to write with error -1 (usb bulk write failed)
mpsse_write: fail to write with error -1 (usb bulk write failed)
mpsse_read: fail to flush buffer with error: -1 (usb bulk write failed)
mpsse_write: fail to write with error -1 (usb bulk write failed)
mpsse_read: fail to flush buffer with error: -1 (usb bulk write failed)
displayReadReg
	Config Target Selection : 2
	PWD Protect
	Done Flag
	ISC Enable
	Write Enable
	Read Enable
	Busy Flag
	Fail Flag
	FFEA OTP
	No err
Error: Failed to program FPGA: std::exception
mpsse_write: fail to write with error -1 (usb bulk write failed)
mpsse_read: fail to flush buffer with error: -1 (usb bulk write failed)
Loopback failed, expect problems on later runs -1
unable to config pins : -1 unable to configure bitbang mode. Perhaps not a BM/2232C type chip?
```

trabucayre, 2023-09-12:
Hi,
I have tried to load and flash bitstream on **ecpix5** successfully.
One thing strange here: **ecpix5** has a *ft2232h* not a *ft4232*.

Anyway: errors reported are produces at *libftdi*/*libusb* level, try to change your USB cable or (I see you uses a **VMWARE* VM) trying first a direct access (msys2 package): sometime USB is unstable with a virtual machine (and if your VM report a ft4232 instead of a ft2232 maybe communication between libftdi and device is broken/bad).

jeremyherbert, 2023-09-12:
Thanks for your reply.

I have just tried on a non-VM system (Ubuntu 22.04) with a different cable and it still shows the same errors. And my ECPIX definitely has a FT4232 - I can see the markings on the IC directly.

trabucayre, 2023-09-12:
It's really weird: most of the time I see this type of error it's related to hardware... Could you try to reduce frequency?

For FT4232 vs FT2232 it's interesting, @lambdaconcept docs shows an FT2232 but maybe your board is a new release with a different cable.

trabucayre, 2023-09-12:
You use `sudo` to run openFPGALoader could you try:
- switching to root (`sudo /bin/bash`) or
- following this [page](https://trabucayre.github.io/openFPGALoader/guide/install.html#udev-rules) to have access right as simple user

I don't remember exactly but I have already seen issue when using sudo

jeremyherbert, 2023-09-12:
Yes, I was just trying sudo to try to see if there were any permissions issues. It does the same thing with non-sudo. My board is very recent, so I believe it is a new revision. Are there any other working boards that use an FT4232?

jeremyherbert, 2023-09-12:
Ok, interesting update. If I connect the board and try to flash a bitfile at 6MHz, not only do the errors come up but the device de-enumerates...

However, at 1MHz it works fine. Even more confusingly, after I flash it with 1MHz, I can then run the command again at 6MHz and it works fine... I can also flash it at 30MHz as long as I flash it with 1MHz first. 

Not quite sure what to make of this.

trabucayre, 2023-09-13:
Yes really interesting thing!
I have recently used a board with a ft4232 without noticing something like this -> I have to install an ubuntu (22.04) with docker to test if I'm able to reproduce this issue, and to see if it's openFPGALoader related or a regression somewhere between libftdi and libusb.

jeremyherbert, 2023-09-14:
So I have been playing with this a bit because I figured like you that it was a hardware problem. It behaved the same with multiple different cables on multiple PCs. Just on a whim I tried unplugging and plugging the cable on the device side ~50 times - and this seems to have fixed it! I guess there was maybe some flux or debris inside the connector from manufacturing. Sorry, I probably should have checked this under the microscope or something earlier. I'm happy for you to close this, but perhaps this is a good opportunity to add a new board or something for the "R03" version of the ECPIX5 because it has the FT4232?

trabucayre, 2023-09-14:
Great news! Thanks!
Yes sometime a simple debris may produces wrong behaviour :)
Yes. It make sense to add a new entry to this new ECPIX5's revision.
Thanks!

trabucayre, 2023-09-21:
I have [pushed](https://github.com/trabucayre/openFPGALoader/commit/afbf0c4ff8d54f60b8cf206b9338881b4b8921c6) a new board definition for *R03* variant.

Thanks again