trabucayre/openFPGALoader #378: Error: fail to open D:/a/msys64/ucrt64/share/openFPGALoader/spiOverJtag_xc6slx9tqg144.bit.gz

ottelo9, 2023-09-11:
I installed openFPGAloader in MSYS2 (pacman -S mingw-w64-ucrt-x86_64-openFPGALoader).

$ openFPGALoader -V
openFPGALoader v0.10.0

If I try to flash I get the error:

```
$ openFPGALoader -c dirtyJtag --fpga-part xc6slx9tqg144 -f "FWUPD_64HD_BETA7_v2.2.0.bin"
write to flash
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Open file DONE
Parse file DONE
use: D:/a/msys64/ucrt64/share/openFPGALoader/spiOverJtag_xc6slx9tqg144.bit.gz
Error: fail to open D:/a/msys64/ucrt64/share/openFPGALoader/spiOverJtag_xc6slx9tqg144.bit.gz
Error: Failed to program FPGA: Error: fail to open D:/a/msys64/ucrt64/share/openFPGALoader/spiOverJt
ag_xc6slx9tqg144.bit.gz
```

I thought this bug was fixed in v.10 #263 ?

trabucayre, 2023-09-13:
The unclear thing is it's possible to install msys2 in a different directory. Patch you mention is to convert relative path to absolute.
Could you try the binary produces by CI with the latest release: here the path is related and automatically converted as absolute at runtime.
Maybe msys2 package must be updated with `MSYS2_ARG_CONV_EXCL="-DDATA_DIR=" ` as done here by the CI (package must be, also, updated to the latest release too).

gregdavill, 2024-03-21:
The fix I'd original applied might not work in all cases, and this looks like one of those times. 

There might be a better way to handle this.  My patch used `cygpath` to do a runtime conversion, but it also fails if cygpath is not in the current environment path.