trabucayre/openFPGALoader #380: CH347 stuck on load for Xilinix and chip id mismatch on Lattice FPGA

LearnShareAlways, 2023-09-18:
I have Xilinx Kintex XC7K325 board and ECP5 color light 75b board and openFPGALoader works fine with ft2232 cable. However, I am not able to load the bitstream using CH347. I tried both Xilinx and Lattice FPGAs :

`openFPGALoader -c ch347_jtag   qmtech_xc7k325t.bit -v
JTAG TCK frequency set to 4 MHz

Xilinx
====

found 1 devices
index 0:
	idcode 0x3651093
	manufacturer xilinx
	family kintex7
	model  xc7k325t
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2023/04/29
design_name: qmtech_xc7k325t
hour: 20:31:53
part_name: 7k325tffg676
toolVersion: 2022.2
userID: 0XFFFFFFFF
load program
(stuck here)`

Lattice
=====
`openFPGALoader -c ch347_jtag colorlight_5a_75b.bit -vJTAG TCK frequency set to 4 MHz

found 1 devices
index 0:
	idcode 0x1111043
	manufacturer lattice
	family ECP5
	model  LFE5UM-25
	irlength 8
File type : bit
Open file: DONE
Parse file: DONE
bitstream header infos
Part: LFE5U-25F-6CABGA256
idcode: 41111043
mismatch between target's idcode and bitstream idcode
	bitstream has 0x41111043 hardware requires 0x00111043
Error: Failed to program FPGA: std::exception
`

LearnShareAlways, 2023-09-18:
Tried binary build from review branch created for https://github.com/trabucayre/openFPGALoader/pull/379  and I see it works for ECP5:
``
openFPGALoader -c ch347_jtag colorlight_5a_75b.bit -v
JTAG TCK frequency set to 7 MHz

found 1 devices
index 0:
	idcode 0x41111043
	manufacturer lattice
	family ECP5
	model  LFE5U-25
	irlength 8
File type : bit
Open file: DONE
Parse file: DONE
bitstream header infos
Part: LFE5U-25F-6CABGA256
idcode: 41111043
IDCode : 41111043
displayReadReg
	Config Target Selection : 0
	Done Flag
	Std PreAmble
	No err
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.00%
Done
userCode: 00000000
Disable configuration: DONE
displayReadReg
	Config Target Selection : 0
	Done Flag
	Std PreAmble
	No err
``
However, it still does not work for Xilinx FPGA:
``
openFPGALoader -c ch347_jtag  qmtech_xc7k325t.bit -v
JTAG TCK frequency set to 7 MHz

found 1 devices
index 0:
	idcode 0x3651093
	manufacturer xilinx
	family kintex7
	model  xc7k325t
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2023/04/29
design_name: qmtech_xc7k325t
hour: 20:31:53
part_name: 7k325tffg676
toolVersion: 2022.2
userID: 0XFFFFFFFF
load program
(stuck here)
``

aystarik, 2023-09-19:
https://github.com/trabucayre/openFPGALoader/pull/379/commits/51292f0b5535283463e281a9b1dc7533b1943d9c this might help

LearnShareAlways, 2023-09-19:
> [51292f0](https://github.com/trabucayre/openFPGALoader/commit/51292f0b5535283463e281a9b1dc7533b1943d9c) this might help

Now getting "input buffer underflow" error (used your repo->master):

JTAG TCK frequency set to 7 MHz

found 1 devices
index 0:
	idcode 0x3651093
	manufacturer xilinx
	family kintex7
	model  xc7k325t
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2023/05/06
design_name: qmtech_kintex.frames
hour: 21:11:19
part_name: xc7k325tffg676-1
toolVersion: xc7frames2bit
userID: xc7frames2bit
load program
Error: Failed to program FPGA: input buffer underflow

LearnShareAlways, 2023-09-19:
No buffer underflow error but It  stuck at load again:

JTAG TCK frequency set to 7 MHz

found 1 devices
index 0:
	idcode 0x3651093
	manufacturer xilinx
	family kintex7
	model  xc7k325t
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2023/04/29
design_name: qmtech_xc7k325t
hour: 20:31:53
part_name: 7k325tffg676
toolVersion: 2022.2
userID: 0XFFFFFFFF
load program

henrypenghong, 2023-12-15:
I also get "input buffer underflow" error in programming bit file to Xilinx FPGA while detecting is OK:

racerxdl, 2024-02-15:
Had the same problem, but #424 fixed it.