trabucayre/openFPGALoader #382: Can't load gowin tangnano9k under OpenBSD

nerfur, 2023-09-25:
Hello, I succesfully built openFPGAloader with all green strings and some warnings (git cloned on 25.09.2023), it can scan usb

> found 10 USB device
> Bus device vid:pid       probe type      manufacturer serial               product
> 000 001    0x0000:0x0000 xvc-client      Generic      none                 xHCI root hub
> 001 001    0x0000:0x0000 xvc-client      Generic      none                 xHCI root hub
> 001 005    0x0403:0x6010 FTDI2232        SIPEED       FactoryAIOT Pro      JTAG Debugger

**but fails to load file**

> fail to read data usb bulk read failed
> JTAG init failed with: low level FTDI init failed

**Partial Output from build process:**

> cmake .
> CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
>   Compatibility with CMake < 3.5 will be removed from a future version of
>   CMake.
>   Update the VERSION argument <min> value or use a ...<max> suffix to tell
>   CMake that the project does not need compatibility with older versions.
> -- The CXX compiler identification is Clang 13.0.0
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Check for working CXX compiler: /usr/bin/c++ - skipped
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
> -- Checking for module 'libftdi1'
> --   Found libftdi1, version 1.5
> -- Checking for module 'libusb-1.0'
> --   Found libusb-1.0, version 1.0.26
> -- Checking for module 'hidapi-hidraw'
> -- Checking for module 'hidapi-libusb'
> --   Found hidapi-libusb, version 0.13.1
> -- Checking for module 'zlib'
> --   Found zlib, version 1.3
> -- Checking for module 'libudev'
> --   Found libudev, version 243
> cmsis_dap support enabled
> Xilinx Virtual Server support disabled
> Remote bitbang client support disabled
> -- Configuring done (26.5s)
> -- Generating done (0.1s)
> -- Build files have been written to: /home/nerfur/openFPGALoader

> ...
> [100%] Linking CXX executable openFPGALoader
> dfu.cpp(CMakeFiles/openFPGALoader.dir/src/dfu.cpp.o:(DFU::searchDFUDevices())): warning: sprintf() is often misused, please use snprintf()
> ftdipp_mpsse.cpp(CMakeFiles/openFPGALoader.dir/src/ftdipp_mpsse.cpp.o:(FTDIpp_MPSSE::search_with_dev(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&))): warning: strcpy() is almost always misused, please use strlcpy()
> [100%] Built target openFPGALoader

trabucayre, 2023-09-26:
Displayed warnings, at build time, are here unrelated to your issue.
Message:
```
fail to read data usb bulk read failed
```
is related to an issue, at *libusb/libftdi* transport layer.
Could you share `dmesg` output? I have already seen issue (enumerate, instability) with the tangNano's onboard cable on rpi4 but don't remember for *OpenBSD*.

nerfur, 2023-09-26:
> is related to an issue, at _libusb/libftdi_ transport layer. Could you share `dmesg` output? 

Sure, here _dmesg_ 

> uftdi0 at uhub3 port 3 configuration 1 interface 0 "SIPEED JTAG Debugger" rev 2.00/5.00 addr 5
> ucom0 at uftdi0 portno 1
> uftdi1 at uhub3 port 3 configuration 1 interface 1 "SIPEED JTAG Debugger" rev 2.00/5.00 addr 5
> ucom1 at uftdi1 portno 2

and also _usbdevs_ outputs

> addr 05: 0403:6010 SIPEED, JTAG Debugger
>          full speed, power 90 mA, config 1, rev 5.00, iSerial FactoryAIOT Pro
>          driver: uftdi0
>          driver: uftdi1

nerfur, 2023-10-06:
Ok, I solved it.

With info from openbsd mailing list. 
Solution is simple:

openFPGAloader under openbsd need more "raw" usb device without attached uftdi driver. You can temporarily disable this driver by for example either 
booting with 'boot -c' command and typing in UKC> 'disable uftdi' 'quit'
or adding 'disable uftdi' to /etc/bsd.re-config

after this board will be in system as ugen device and will be accessible by openFPGAloader.

closed 2023-10-06 by nerfur