trabucayre/openFPGALoader #383: flash chip unknown, timeout

ottelo9, 2023-09-25:
I have a HDMI mod (64HD) for my Nintendo 64 and try to flash it (again) with a Pico and openFPGALoader running on Windows (UCRT64 MSYS2). But I actually get this error:

[image]

I use this command:
openFPGALoader -c dirtyJtag --fpga-part xc6slx9tqq144 -f FWUPD_64HD_v2.0.2.mcs

I already tested it with --unprotect-flash or -f FWUPD_64HD_v2.0.2.bin but with same results.

The strange thing is that I was already able to successfully flash the FPGA/Flash Chip a week ago. I double checked all wires and they seems to be okay.

Today I read something similar [here ](https://github.com/trabucayre/openFPGALoader/issues/124#issuecomment-1606002876). Maybe I have the same problem? Can I block the boot communication between the FPGA and flash ic while powering up (connect/short SPI pins)?

[image]

trabucayre, 2023-09-26:
Could you share exact flash model?
Compared to issue mentioned the flash return something different to 0xff: seems your able to communicate with the chip.
Your board is publicly available? Is it possible to order (or to have a physical access to one of them)?
Thanks

ottelo9, 2023-09-26:
@trabucayre 
It's a Micron [48LC16M16A2](https://www.mouser.de/datasheet/2/671/mict_s_a0001002289_1-2290690.pdf) flash. The 64HD mod board is actually hard to get (HDMI Nintendo 64 mods are always sold out in few seconds). A few others from the community have already been able to flash it successfully (like me).

But I have now ordered a Xilinx DLC10 Programmer to get it running.

aystarik, 2023-09-27:
What you are pointing at is RAM
SDR SDRAM
MT48LC64M4A2 – 16 Meg x 4 x 4 banks
MT48LC32M8A2 – 8 Meg x 8 x 4 banks
MT48LC16M16A2 – 4 Meg x 16 x 4 banks

SPI FLASH is a small 8-pin chip

ottelo9, 2023-09-28:
Ups, my fault. Its a Windbond 25064jvs10.

aystarik, 2023-09-28:
https://www.winbond.com/hq/product/code-storage-flash-memory/serial-nor-flash/?__locale=en&partNo=W25Q64JV -- this?

ottelo9, 2023-09-28:
> https://www.winbond.com/hq/product/code-storage-flash-memory/serial-nor-flash/?__locale=en&partNo=W25Q64JV -- this?
> 
> W25Q64JV

yes!

Today I received the Waveshare Xilinx platform cable and followed the original instructions from the maker of the 64HD mod.

But with the original way of flashing the chip it doesnt work, too. I get SPI flash chip errors:

```
iMPACT Version: Oct 13 2013 10:22:21

iMPACT log file Started on Thu Sep 28 14:12:42 2023

Preference Table
Name                 Setting             
StartupClock         Auto_Correction     
AutoSignature        False               
KeepSVF              False               
ConcurrentMode       False               
UseHighz             False               
ConfigOnFailure      Stop                
UserLevel            Novice              
MessageLevel         Detailed            
svfUseTime           false               
SpiByteSwap          Auto_Correction     
AutoInfer            false               
SvfPlayDisplayComments false               
AutoDetecting cable. Please wait.
*** WARNING ***: When port is set to auto detect mode, cable speed is set todefault 6 MHz regardless of explicit arguments supplied for setting the baudrates
Connecting to cable (Usb Port - USB21).
Checking cable driver.
 Driver file xusb22.sys found.
 File C:/Xilinx/14.7/LabTools/LabTools/bin/nt64/xusb22.sys not found.
 Driver file xusb_xp2.sys found.
 Driver version: src=2301, dest=2301.
 Driver windrvr6.sys version = 10.2.1.0. WinDriver v10.21 Jungo (c) 1997 - 2010 Build Date: Aug 31 2010 x86_64 64bit SYS14:14:44, version = 1021.
 Cable PID = 0008.
 Max current requested during enumeration is 300 mA.
Type = 0x0005.
write (count, cmdBuffer, dataBuffer) failed C0000004.
 Cable Type = 3, Revision = 0.
 Setting cable speed to 6 MHz.
Cable connection established.
Firmware version = 2301.
File version of C:/Xilinx/14.7/LabTools/LabTools/data/xusb_xp2.hex = 2401.
Firmware hex file version = 2401.
Downloading C:/Xilinx/14.7/LabTools/LabTools/data/xusb_xp2.hex.
Downloaded firmware version = 2401.
PLD file version = 200Dh.
 PLD version = 200Dh.
Type = 0x0005.
ESN option: 13620207920A61.
Identifying chain contents...'0': : Manufacturer's ID = Xilinx xc6slx9, Version : 2
INFO:iMPACT:1777 - 
   Reading C:/Xilinx/14.7/LabTools/LabTools/spartan6/data/xc6slx9.bsd...
INFO:iMPACT:501 - '1': Added Device xc6slx9 successfully.
----------------------------------------------------------------------
----------------------------------------------------------------------
done.
Elapsed time =      1 sec.
Elapsed time =      0 sec.
Elapsed time =      0 sec.
Elapsed time =      0 sec.
Unprotect sectors: FALSE
INFO:iMPACT - A CFI file is not detected. To ensure correct and safe   configuration,
    Please make sure a CFI file is present in the same directory as the PROM   file,
    or, regenerate the PROM file with the latest software.
'1': SPI access core not detected. SPI access core will be downloaded to thedevice to enable operations.
INFO:iMPACT - Downloading core file   C:/Xilinx/14.7/LabTools/LabTools/spartan6/data/xc6slx9_spi.cor.
'1': Downloading core...
 LCK_cycle = NoWait.
LCK cycle: NoWait
done.
'1': Reading status register contents...
INFO:iMPACT:2219 - Status register values:
INFO:iMPACT - 0011 1100 1110 1100 
INFO:iMPACT:2492 - '1': Completed downloading core to device.
'1': ID Check passed.
'1': ID Check passed.
'1': Erasing Device.
'1': Using Sector Erase.
'1': Erasing non-volatile quad-enable bit...
'1': Programming Flash.
'1': Reading device contents...
read count != nBytes, rc = C0000011.
read failed C0000011.
read count != nBytes, rc = 20000015.
read failed 20000015.
Failed at address, 90880
'1': Verification Terminated
INFO:iMPACT - '1': Flash was not programmed successfully.
write cmdbuffer failed 20000015.
write cmdbuffer failed 20000015.
write cmdbuffer failed 20000015.
Error reading reference voltage level.
VREF was lost during the current operation. It is recommended that theoperation be repeated.
write cmdbuffer failed 20000015.
write (count, cmdBuffer, dataBuffer) failed 20000015.
Elapsed time =     36 sec.
write (count, cmdBuffer, dataBuffer) failed 20000015.
write cmdbuffer failed 20000015.
```

ottelo9, 2023-09-28:
Ok I disabled the verify after programming the flash. Now it works. But why it worked the first time and now it doesn't, I don't know. Maybe the program loaded from flash at boot blocks the SPI in anyway? But I can erase the flash, program it but verify doesnt work...

trabucayre, 2023-10-01:
The first thing I see in your first post is the wrong ID: W25Q64 must have 0xef4017 -> it seems to have an issue with communication...
Another strange thing is: your bitstream seems working but when you use impact verify step didn't match: Could you try another totally different bitstream to check if the bitstream is really written?

Yes: it's possible your device is configured in QSPI mode and consequently you can't access to it with a tool supporting only SPI mode.

ottelo9, 2023-10-04:
The bitstream is really writen to the flash, I tested it with different 64HD Firmware versions. Because the 64 HDMI mod is working now and no one else has any problems with it, I won't investigate any time in it. I close it.

closed 2023-10-04 by ottelo9