trabucayre/openFPGALoader #387: Segmentation fault when programming GateMate FPGAs

machdyne, 2023-10-04:
Hello. I'm using openFPGALoader with pico-dirtyJtag to program GateMate FPGAs. I'm not sure if this is supposed to work, but it is working at least sometimes.

However, it seems that with some larger bitstreams there is a segfault:

```
$ sudo openFPGALoader -c dirtyJtag --verbose top_00.cfg.bit 
Jtag frequency : requested 6000000Hz -> real 6000000Hz
found 1 devices
index 0:
	idcode 0x20000001
	manufacturer colognechip
	family GateMate Series
	model  GM1Ax
	irlength 6
File type : bit
Segmentation fault
```
```
Thread 1 "openFPGALoader" received signal SIGSEGV, Segmentation fault.
0x00007ffff7f796ac in ftdi_write_data ()
   from /lib/x86_64-linux-gnu/libftdi1.so.2
```

I'm also sometimes seeing segfaults in libusb_exit. And sometimes using --verbose changes the behavior, as seen below.

Any ideas?

```
$ sudo openFPGALoader -c dirtyJtag count_00.cfg.bit
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Segmentation fault
```

```
$ sudo openFPGALoader -c dirtyJtag --verbose count_00.cfg.bit
Jtag frequency : requested 6000000Hz -> real 6000000Hz
found 1 devices
index 0:
	idcode 0x20000001
	manufacturer colognechip
	family GateMate Series
	model  GM1Ax
	irlength 6
File type : bit
Load SRAM via JTAG: [==================================================] 100.00%
Done
```

trabucayre, 2023-10-04:
In fact I have never tried to program my gatemate with an external interface.
There is no real reason why it's not working.
But by re-reading code, some predicate exists to have access to reset, done, fail and oen (and SPI access is ftdi dependant :().
I suspect something around trying to access unsupported/unavailable feature may be a cause of these segfaut (but it's maybe not the full answer since sometime your able to program).

I have to see if it's possible, with my eval board to bypass onboard jtag interface to use external one.

Note: I have noticed by using [picorv32](https://github.com/pu-cc/picorv32/tree/gatemate-demo) jtag is no more available when `p_r` is used with `-uCIO` so direct access to the SPI may be required (or a way to avoid self programming at start time).

pu-cc, 2023-10-04:
@machdyne could you try https://github.com/trabucayre/openFPGALoader/commit/18056180a8c5a9d3c944b203a5bdf5d3dfed9732?

Just clone into my repo:
```
$ git clone --branch gatemate-dirtyjtag https://github.com/pu-cc/openFPGALoader.git
```
Thanks!

machdyne, 2023-10-04:
@pu-cc Thanks. I tried that branch with several different bitstreams and I'm no longer seeing any segfaults.

trabucayre, 2023-10-04:
@pu-cc Thanks for this quick fix !

PR #388 (2023-10-04): gatemate: fix dirtJtag support
Fixes #387

Thanks for the report @machdyne
closed 2023-10-04 by trabucayre