trabucayre/openFPGALoader #389: write to flash on GW5A is not yet supported

chopin1998, 2023-10-10:
i got a tang mega 138k pro which use Gowin GW5AST 138 as main fpga chip,

but now openFPGALoader not supported to write its external flash, will support soon?

Thank you!

trabucayre, 2023-10-10:
I will receive a mega in few days (or a week), will be able to work on adding external flash support.

chopin1998, 2023-10-20:
got your board ? ^_^

trabucayre, 2023-10-20:
Yes! I have received my board and started to see how to access flash for this family. It's not really complex (a bit tricky) but I have first to review a PR for gowin devices.

chopin1998, 2023-10-21:
Good news! i’m a newbie on this, thanks for your help!

chopin1998, 2023-11-20:
i noticed that openFPGAloader support flash GW5A now! so happy!

but seems write to SRAM function bad now.... when Load a new fs file to SRAM, nothing happen, and READY led on board keeping ON, which should be off

trabucayre, 2023-11-20:
Currently SPI write is only supported for GW5A, not GW5AST (I don't know why but I have strange behaviour).
It's weird: I have tried yesterday with tangMega and not seen issue when trying to load SRAM, maybe device is in wrong state...
I have to investigate.

trabucayre, 2023-11-20:
I have also to update status display: currently bit/function are displayed according to non GW5A(ST). For this family only GW5A-25 has the correct register decoding.

chopin1998, 2023-11-20:
> Currently SPI write is only supported for GW5A, not GW5AST (I don't know why but I have strange behaviour). It's weird: I have tried yesterday with tangMega and not seen issue when trying to load SRAM, maybe device is in wrong state... I have to investigate.

but i can write flash on my mega138pro..

trabucayre, 2023-11-20:
Ok. So I have to recheck everything. SPI flash is seen but not all the time, it fails to write and I'm able to load SRAM. it's more or less totally opposite situation.
Could you try with `openFPGALoader --bulk-erase` to see if after your able to load a bitstream (ie device is in wrong state due to an issue when bitstream was written).

chopin1998, 2023-11-21:
when device under wrong state, other write is not work, even with --bulk-erase, need to repower it.

trabucayre, 2023-11-28:
I continue to search the reason to this regression. before this [commit](https://github.com/trabucayre/openFPGALoader/commit/f71858f96a75c7c0a96f31a4ed3f167e4b914ef0) there is no issue, after yes.
I have some results but it's not really stable and OK/KO depends on bitstream written to the flash...

wtjcn, 2023-12-10:
I have the same problem, when load to sram of 138k, nothing happen

trabucayre, 2023-12-10:
I try, as much as possible, to find why SRAM is not working. But I can't isolate the issue. It's a WIP.

trabucayre, 2023-12-10:
A strange thing I observe: depending of the bitstream written in flash, loading to SRAM works or not.
If I have one of the @sipeed bitstream: SRAM fails. With LiteX bitstream: SRAM is working.
Maybe something wrong with the security option.

trabucayre, 2023-12-11:
With my last commit I'm able to load a bitstream all the time (with a flash containing a bitstream where security bit is enable or not).
Could you confirm/infirm my tests ?
Thanks.

chopin1998, 2023-12-12:
c00l!! 

now SRAM write ~100% work, thank thee very much!

flash write need twice, 1st time failed(need to wait), then 2rd time work fine

trabucayre, 2023-12-12:
I have observed this issue too. I don't know why but `bulk-erase` works all the time.
Maybe the same trick must be applied on Flash side.

wtjcn, 2023-12-12:
for me, it not work.
when write sram, it show "Erase SRAM double eraseSRAM", progress to 100% and "DONE", but the board's LEDs light continue in old ways. Both litex and official led.fs .

trabucayre, 2023-12-12:
litex and led.fs uses leds in a similar way. Could you try with litex and [ws2812](https://github.com/sipeed/TangMega-138KPro-example/tree/main/ws2812). I use this demo to have real distinct behavior when I have to switch.
But If after the load you have something (ie no errors) it's a good news.

wtjcn, 2023-12-21:
1) I found I used linux's FT2232 driver, because it can give /dev/ttyUSB device, although it does not realy work on mega138k
2) I run Gowin_USB_Cable_Installer.sh and then reboot, /dev/ttyUSB disappear, but Gowin's Programmer works well, both for SRAM and FLASH. In this condition, I use openFPGAloader,  sram erase dose not work, and sram programming "DONE" but not work. Another small problem, openFPGAloader default use 6MHz jtag freq,  Gowin's doc said it should not more than 2.5MHz, I change it to 2.5MHz but no change.

allrighteveryday, 2024-09-08:
In version 0.12.1，Ubuntu , I also meet the problem: nothing happened after burning and READY led on board keeping ON, which should be off.