trabucayre/openFPGALoader #393: Cannot write flash on arty_a7_100t [Error: flash chip unknown: use basic protection detection, then timeout: ff ff ff]

inkdot7, 2023-10-15:
With an arty_a7_100t board, I am not able to write or erase flash.
Loading an image directly to the FPGA works.

Using openFPGALoader, current master, f964c33.

On the first attempt, I got:
```
load program
Flash SRAM: [==================================================] 100.00%
Done
Detail:
Jedec ID          : cd
memory type       : ef
memory capacity   : 1f
flash chip unknown: use basic protection detection
timeout: b a1 65
b
wait: Error
Erasing: [                                                  ] 0.00%
```
which then did not do much, and I aborted with Ctrl-C.  Perhaps a bad idea?

After that first attempt, the Jedec ID and following bytes are all 00:
```
OPENFPGALOADER_SOJ_DIR=~/sw/openFPGALoader/spiOverJtag ~/sw/openFPGALoader/build/openFPGALoader -b arty a100_fnet.bit -f --verbose-level 2
write to flash
Jtag frequency : requested 10.00MHz  -> real 10.00MHz 
Raw IDCODE:
- 0 -> 0x13631093
- 1 -> 0xffffffff
Fetched TDI, end-of-chain
found 1 devices
index 0:
	idcode 0x3631093
	manufacturer xilinx
	family artix a7 100t
	model  xc7a100
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2023/10/15
design_name: arty_a7_35_fakernet
hour: 18:57:07
part_name: 7a100tcsg324
toolVersion: TRUE;Version=2017.2
userID: 0XFFFFFFFF
use: /home/htj/sw/openFPGALoader/spiOverJtag/spiOverJtag_xc7a35tcsg324.bit.gz
load program
Load SRAM: [==================================================] 100.00%
Done
0 0 0 0 read 0
Detail: 
Jedec ID          : 00
memory type       : 00
memory capacity   : 00
0 0 0 40 read 40
Detail: 
Jedec ID          : 00
memory type       : 00
memory capacity   : 00
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
flash chip unknown: use basic protection detection
timeout: ff ff ff
ff
wait: Error
Erasing: [                                                  ] 0.00%timeout: ff ff ff
ff
wait: Error
Erasing: [==                                                ] 3.85%
```
(I now waited a bit longer before aborting.)

The arty_a7_100t has a product change notification regarding the flash: https://files.digilent.com/resources/programmable-logic/documents/S25FL127S_PCN.pdf
However, the board I have has a sticker: S25FL128SAG, so I would imagine it is the original kind.

trabucayre, 2023-10-15:
`arty` is for 35t version. I have left this entry for backward compatibility (but maybe i'ts an error). For the 100t version you have to use `-b arty_a7_100t` (it's required to have FPGA mode/size for loading correct bridge).
I suspect something like that (and I have to improve sanity check in direct part of the code to verify FPGA<->bitstream match and stop when jedec == 0xffffff or 0x000000).

inkdot7, 2023-10-15:
That was quick!  Thanks!  That did the trick!

Indeed it would be nice with all that sanity checking.
Also for the bitstream one tries to load into flash - I am making so many mistakes...

trabucayre, 2023-10-20:
It's planned to try to have a more global/simple way for sanity check between hardware and bitfile.
Thanks!