trabucayre/openFPGALoader #398: unable to open ftdi device: -6 (ftdi_usb_reset failed)

PedroAntunes178, 2023-10-25:
Hello,

I am trying to program the [IceSugar-pro](https://github.com/wuxx/icesugar-pro) board. That uses the ECP5 FPGA.
I have encountered several errors.

First, `unable to open ftdi device: -3 (device not found)` which happened if I called `openFPGALoader` without `--busdev-num`. 
The command I used that bypassed this issue was `openFPGALoader --busdev-num 003:005 -c digilent_hs2 blink_100.bit`.  
The bus and the device address can be obtained by executing `lsusb`.

Second, `unable to open ftdi device: -4 (libusb_open() failed)` which happened because my user did not have permissions to write to the Serial port. To solve this I followed issue https://github.com/trabucayre/openFPGALoader/issues/245 and added the following lines to `/etc/udev/rules.d/99-openfpgaloader.rules`:
> \# IceSugar-Pro  
ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="602b", MODE="664", GROUP="plugdev", TAG+="uaccess"

The current issue I am having trouble solving is `unable to open ftdi device: -6 (ftdi_usb_reset failed)`. There does not seem to be much information about it.

Any help is appreciated, 
Thank you in advance :)

Best,
Pedro

trabucayre, 2023-10-26:
Hello,

## First issue:
The second issue give the answer for your first issue: you use `-c digilent_hs2`. This cable is based on **ft232H** (with vid:pid = 0x0403:0x6014) but your interface has a vid:pid = 0x1d50:0x602b.

## Second issue: 

yes you needs to have access right on your USB interface as mentioned [here](https://trabucayre.github.io/openFPGALoader/guide/install.html#udev-rules) and [here](https://trabucayre.github.io/openFPGALoader/guide/troubleshooting.html#unable-to-open-ftdi-device-4-usb-open-failed-issue-245). 

I'm agree those sections are not much visible.

## Last issue:

I haven't an iceSugar-pro but have colorlight i5 (more or less similar) and the interface as mentioned in the iceSugar repo's README is: 

> the on board debugger iCELink (base on ARM Mbed DAPLink)

This isn't an FTDI (or clone) and you have to use the right protocol.

This is a specific protocol implemented on the onboard microcontroler (**U3** in board schematic). 
So maybe trying with `-b cmsisdap --vid=0x1d50 --pid0x602b` will resolve your issue.

PedroAntunes178, 2023-10-26:
Hey,

Nice, thank you! :) 

The following command line worked:
```
openFPGALoader -c cmsisdap --vid=0x1d50 --pid=0x602b blink_1000.bit
````

closed 2023-10-26 by PedroAntunes178