trabucayre/openFPGALoader #402: I can't communicate with my "tang nano 20k"

joselmy-diaz, 2023-11-03:
Hello, I have a problem with the "tang nano 20k" not communicating with it.
it returns the following error:
`$ openFPGALoader -m -b tangnano20k ~/Escritorio/ASM/counter.fs
write to ram
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Parse file Parse /home/joselmy/Escritorio/ASM/counter.fs:
Done
DONE
Error: Failed to claim FPGA device: mismatch between target's idcode and bitstream idcode
	bitstream has 0x1100481B hardware requires 0x0000081b
`

joselmy-diaz, 2023-11-03:
> Hola, tengo un problema con el "tang nano 20k" que no se comunica con él. devuelve el siguiente error: `$ openFPGALoader -m -b tangnano20k ~/Escritorio/ASM/counter.fs write to ram Jtag frequency : requested 6.00MHz -> real 6.00MHz Parse file Parse /home/joselmy/Escritorio/ASM/counter.fs: Done DONE Error: Failed to claim FPGA device: mismatch between target's idcode and bitstream idcode bitstream has 0x1100481B hardware requires 0x0000081b ` Lo intenté con otra computadora con Windows y funciona correctamente. Puedes ayudarme, me gustaría poder usarlo desde mi dispositivo con arch linux
> 
> `$ openFPGALoader -V openFPGALoader v0.11.0`

trabucayre, 2023-11-06:
A sanity check is done before load or flash to be sure bitstream match device so if the .fs was for another device openFPGALoader produces this type of messages

trabucayre, 2024-11-26:
Any news?
This issue seems clearly related to the bitstream targeted for an GW1N(R)-9?
I close this issue. Feel free to reopen or to create a new one
Thanks

closed 2024-11-26 by trabucayre