trabucayre/openFPGALoader #403: mismatch between target's idcode and bitstream idcode

multigcs, 2023-11-04:
strange,  found device with 'idcode 0x12bd043'
and  bitstream header infos with 'idcode: 012bd043'

but say: bitstream has 0x012BD043 hardware requires 0x002bd043

```
# openFPGALoader -v -c usb-blaster build/rio_build.bit 
found 1 devices
index 0:
	idcode 0x12bd043
	manufacturer lattice
	family MachXO2
	model  LCMXO2-7000HC
	irlength 8
File type : bit
Open file: DONE
Parse file: DONE
bitstream header infos
Architecture: xo2c00
Bits: 1533840
Bitstream CRC: 0x4394
Bitstream Status: Final Version 1.95
Cols: 1992
Date: Sat Nov  4 18:21:36 2023
Design name: rio_build.ncd
Part: LCMXO2-7000HC-4TQFP144
Readback: Off
Rows: 770
Security: Off
Version: Diamond (64-bit) 3.12.0.240.2
idcode: 012bd043
mismatch between target's idcode and bitstream idcode
	bitstream has 0x012BD043 hardware requires 0x002bd043
Error: Failed to program FPGA: std::exception

```

ravirajchilka, 2024-03-11:
Did you find the solution for this error ?

multigcs, 2024-03-11:
not realy, but it works on raspberry with gpio's (libgpiod)

openFPGALoader -c libgpiod --pins=19:26:13:6 -r -f --unprotect-flash rio_build.bit

ravirajchilka, 2024-03-12:
I fixed this issue yesterday by using GOWIN IDE which installed drivers as well, I was getting this error when using VScode. Its related driver issue