trabucayre/openFPGALoader #411: unable to open ftdi device: -4 (usd_open() failed) on Windows (tangnano1k)

alain6666, 2023-12-22:
I use openFPGALoader from yosys-oss-cad-suite on Windows. When I connect to tangnano1k via USB cable, I enter the command `openFPGALoader. exe - b tangnano1k .\Top.fs`, then, 
```
unable to open ftdi device: -4 (usd_open() failed)
JTAG init failed with: unable to open ftdi device.
```
I think my USB device is recognized properly because when I remove the board connection and enter the above command, it will prompt: 
```
unable to open ftdi device: -3 (device not found)
JTAG init failed with: unable to open ftdi device.
```
[image]

When I connect the board, the following content will be displayed in my device manager
[image]

Additionally, I believe that my board is functioning properly as I can use OpenFPGALoader to burn bitstream normally under MacOS.

angelaacabral, 2024-01-15:
Hello,

Did you manage to find out what was happening? I have the same problem.

If you are working from Windows, you need Zadig. So, launching the executable worked for me. However, when installing openFPGALoader and running it, I get the same problem.

> openFPGALoader -b gatemate_evb_jtag blink.cfg.bit
empty
unable to open ftdi device: -3 (device not found)
JTAG init failed with: unable to open ftdi device

Thanks.

trabucayre, 2024-01-15:
Yes, for windows you have to use zadig (and my knowledge stop more or less here for this OS).

@angelaacabral it's weird. If you are able to use openFPGALoader with a tangnano you must be able to do the same thing for the gatemate. Both uses interfaceA for programming.

angelaacabral, 2024-01-15:
It's quite strange. When I do it with the example zip provided by Cologne Chip, since it has the executable it works correctly, but when installing the tool and executing exactly the same command that is carried out from the Makefile, it gives that error. I have tried changing USB and user permissions, but there is no way.

I had faith that someone would find a solution. In the meantime I will continue investigating.

trabucayre, 2024-01-15:
And with nightly/release asset it's the same result ?

angelaacabral, 2024-01-15:
mmm I don't think I know how that would be done.

alain6666, 2024-01-16:
> Hello,
>
> Did you manage to find out what was happening? I have the same problem.
>
> If you are working from Windows, you need Zadig. So, launching the executable worked for me. However, when installing [...]

After using Zadig, I can now use OpenFPGALaoader normally.

angelaacabral, 2024-01-16:
I have already gotten it to work, thank you very much for your help.

angelaacabral, 2024-01-16:
Certainly, does anyone know if the GateMate primitives' code can be found anywhere? No matter how much I search, I haven't been able to locate them, and I think they are necessary for GHDL.

trabucayre, 2024-01-18:
@alain6666 : great!
@angelaacabral : synthesis is done by yosys. primitives are in yosys share directory. Check @colognechip config.mk to see how to use it

closed 2024-06-07 by trabucayre