trabucayre/openFPGALoader #413: colorlight i5 and WSL2

timksf, 2023-12-24:
Hi,
I'm having some trouble getting openFPGALoader to work with my colorlight i5. I'm working inside WSL2 with a custom kernel that has
CONFIG_USB_HIDDEV  enabled. The usb device is forwarded with usbipd.
Here is my dmesg output:
```
[  +4.153175] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[  +0.000004] vhci_hcd vhci_hcd.0: devid(196621) speed(2) speed_str(full-speed)
[  +0.000042] vhci_hcd vhci_hcd.0: Device attached
[  +0.283290] vhci_hcd: vhci_device speed not set
[  +0.069954] usb 1-1: new full-speed USB device number 4 using vhci_hcd
[  +0.080066] vhci_hcd: vhci_device speed not set
[  +0.069994] usb 1-1: SetAddress Request (4) to port 0
[  +0.035266] usb 1-1: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice= 1.00
[  +0.000005] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000001] usb 1-1: Product: DAPLink CMSIS-DAP
[  +0.000001] usb 1-1: Manufacturer: ARM
[  +0.000001] usb 1-1: SerialNumber: 0700000105daff303334594e43135065a5a5a5a597969908
[  +0.002229] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  +0.002671] hid-generic 0003:0D28:0204.0006: hiddev96: USB HID v1.00 Device [ARM DAPLink CMSIS-DAP] on usb-vhci_hcd.0-1/input2
```
This looks fine to me, here is the lsusb output:
`Bus 001 Device 004: ID 0d28:0204 NXP ARM mbed`

Now when I try to detect the FPGA I get the following output:
```
>:~$ openFPGALoader --detect -c cmsisdap
empty
JTAG init failed with: No device found
```
I also tried ecpdap, which seems to work just fine, after I installed the udev rules (https://github.com/adamgreig/ecpdap/tree/master/drivers):
```
>:~$ ecpdap scan
Detected JTAG chain, closest to TDO first:
 - 0: 0x41111043 (Lattice Semi.) [IR length: 8] [LFE5U-25]
```
Any help would be greatly appreciated.

stavinsky, 2024-04-09:
same here with sipeed slogic in daplink mode. 
works with ecpdap, I can program and write flash. 
cant work with openFPGAloader even if I set pid and vid explicitly

trabucayre, 2024-04-11:
I have acquired a little bit old windows computer. I will try to reproduces but I'm not sure if this computer is compatible with WSL2.
And in fact I have near 0 knowledge about windows...

stavinsky, 2024-04-12:
in my case it is OS X, sorry forgot to mention. No emulation. native libraries

trabucayre, 2024-04-12:
Ok I have to check both case. Thanks.
Until recently my only OS was linux so supports was only tested in this OS.

trabucayre, 2024-04-13:
@stavinsky I have tried with my macbookAir and the colorlight i5 bitstream is correctly loaded. `--detect`is also working.
I have to check which version of dap the slogic support (currently openFPGALoader supports only v1).
I have now to see how to build openFPGALoader for windows and to do the same thing...

trabucayre, 2024-04-14:
@timksf I have switch hidapi library detection. Problem was an issue with cable detection: not working with hidraw, working with libusb.
I'm able to load a bitstream on my colorlight i5 with WSL2 on win10.

trabucayre, 2024-04-14:
Fixed by https://github.com/trabucayre/openFPGALoader/commit/2535c28e0443553660c15975a0da52aa9d6c7467

timksf, 2024-04-14:
thank you for the effort, I'll be able to test it over the coming days

timksf, 2024-04-15:
works now, thanks again

closed 2024-04-15 by timksf