trabucayre/openFPGALoader #418: Which USB<->JTAG interface is the easiest/best/preferrred to implement?

harbaum, 2024-01-18:
I am working with the Tang Nano 20k. This doesn't use a real ft2232 but instead runs some emulation of the ft2232 on a bl616 risc-v CPU. Now this chip is pretty powerful and i'd like to use it for other purposes as well. So I need to replace its firmware which can easily be done. The Tang Nano 20k even has the "Update" button for this purpose. But doing so makes the board loose its ability to load the FPGA. 

Sipeed (the maker of the Tang Nano 20k) will now allow to distribute the source code of the ft2232 emulation, so it's not possible to include this functionality into some custom extended firmware for the Tang Nano 20k. One solution is to restore the original firmware whenever the FPGA (or its flash) is to be updated and then flash the custom firmware again afterwards. This works but is a pain to use ...

What I am now searching for is a replacement solution for the FPGA-loader part inside the BL616 MCU on the Tang Nano 20k. I don't like the ft2232 emulation that much as it re-uses the FTDI USB vendor ID and potentially violates FTDIs rights.

So here's my question: What is the USB/JTAG protocol that you guys think is the one to implement? There is e.g. a CDC ACM example for the BL616 CPU and all the JTAG pins are also accessible. But I'd need something in between these two parts. Which protocol should one implement on top of CDC ACM (or anything else, but CDC ACM seems to be a good choice) to be able to use openFPGAloader to upload code to the Tang Nano 20k.

I am doing this under Linux. But if openFPGAloader would use CDC ACM (as opposed to libusb directly), then this should imho work nicely under other OSs as well without the need to use tools like Zadig under Windows oder give special libusb permissions under Linux.

trabucayre, 2024-01-18:
CDC ACM is for emulate serial interface no?
My first idea is dirtyJTAG but this protocol is slower than ftdi.

harbaum, 2024-01-18:
Yes, CDC ACM emulates a serial interface. The advantage is that it's a USB standard protocol and doesn't need any special drivers. It'll give you a COM port under windows and a /dev/ttyACM under Linux. And MacOS also has a standard way to deal with it.

Speed is imho no issue.

harbaum, 2024-01-18:
DirtyJTAG is a vendor class device and also seems to use a custom protocol on USB side and would require libusb (and zadig on windows) etc. While this is not a no-go I think it would be more end-user friendly to use something that runs over a regular COM/tty.