trabucayre/openFPGALoader #421: Loading Tang Primer 25K does not work both for SRAM and flash

jgroman, 2024-01-21:
I have a Sipeed Tang Primer 25K board which can be successfully programmed using official Gowin programmer (both GUI and CLI).
However this board fails to be programmed using openFPGALoader. Loading SRAM finishes without errors but there is no change in board behavior afterwards. Loading flash fails with an error. Please see included logs:
`sudo openFPGALoader -b tangprimer25k --verbose impl/pnr/project.fs`: [prg_sram.log](https://github.com/trabucayre/openFPGALoader/files/14000542/prg_sram.log)
`sudo openFPGALoader -b tangprimer25k -f --verbose impl/pnr/project.fs`: [prg_flash.log](https://github.com/trabucayre/openFPGALoader/files/14000541/prg_flash.log)

I am running Fedora 39, openFPGALoader is built from the latest master. Tried connecting the board to USB directly and through USB hub with no change. The official programmer uses 15MHz clock but setting this speed on openFPGALoader has no visible effect.
Tang Primer 25K board uses BL616 for USB-JTAG/UART, emulating FT2232C/D/H.

trabucayre, 2024-01-23:
Hi,
Sorry for the delay, I have spent time to check openFPGALoader with my tang primer 25k without being able to reproduce your issue.
Your test is really interessting because it show a regression at BL616 level (I assume).
Could you dump dmesg output when you plug your board : something like:
```
[87927.537898] usb 3-8.1: Product: FPGA PARTNER
[87927.537899] usb 3-8.1: Manufacturer: SIPEED
[87927.537899] usb 3-8.1: SerialNumber: 2023081411
```
I suspect SerialNumber will start twith **202311**

In fact I have observed another wrong behavior with a tangMega and an BL616 with this a serialNumber more recent than mine.

Yes a better detection (maybe based on manufacturer / product is required to add a workaround here.
I have also to find where to download firmware version to be able to update my tangPrimer (or my tangMega) to see but your idea to use an opcode only supported by an 'H' version make sense.

Thanks

jgroman, 2024-01-23:
Hi @trabucayre , thank you for getting back to me!
Yes, you are right, my board reports this new serial number:
```
[23802.581729] usb 1-2: New USB device found, idVendor=0403, idProduct=6010, bcdDevice= 7.00
[23802.581746] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[23802.581754] usb 1-2: Product: USB Debugger
[23802.581759] usb 1-2: Manufacturer: SIPEED
[23802.581765] usb 1-2: SerialNumber: 2023112818
```

libftdi uses **bcdDevice** only to distinguish between FT2232 types: [ftdi.c source](https://www.intra2net.com/en/developer/libftdi/documentation/ftdi_8c_source.html) (line 677).

Here is the problematic part in board communication:
```
595392600 S Bo:1:015:2 -115 14 = 4b05ff4b 03823903 00ffffff ff87    # TAP-Reset; Shift-DR; R/W 4 bytes 0xFFFFFFFF; Flush
595392629 C Bo:1:015:2 0 14 >
595392637 S Bi:1:015:1 -115 512 <
595392800 C Bi:1:015:1 0 6 = 02601b28 0100                          # -> 0x1B280100
595392809 S Bo:1:015:2 -115 8 = 390300ff ffffff87                   # R/W 4 bytes 0xFFFFFFFF; Flush
595392839 C Bo:1:015:2 0 8 >
595392847 S Bi:1:015:1 -115 512 <
595392887 C Bi:1:015:1 0 6 = 0260ffff ffff                          # -> 0xFFFFFFFF
595392895 S Bo:1:015:2 -115 3 = 4b04bf                              # [11111] Test-Logic-Reset
595392926 C Bo:1:015:2 0 3 >
595415291 S Bo:1:015:2 -115 9 = 4b04861b 06414b00 01                # [01100] Shift-IR; Cmd 0x41 Read Status Reg; [1] Exit1-IR;
595415326 C Bo:1:015:2 0 9 >
595415341 S Bo:1:015:2 -115 15 = 4b01818e 054b0281 390200ff ffff87  # [10] Run-Test-Idle; Wait 6 clk; [100] Shift-DR; R/W 3 bytes 0xFFFFFF; Flush
595415358 C Bo:1:015:2 0 15 >
595415367 S Bi:1:015:1 -115 512 <
595415398 C Bi:1:015:1 0 2 = 0260
595415406 S Bi:1:015:1 -115 512 <
595415438 C Bi:1:015:1 0 2 = 0260
595415445 S Bi:1:015:1 -115 512 <
595415477 C Bi:1:015:1 0 4 = 0260fa05                               # -> 0xFA05     ! MPSSE Error
595415485 S Bi:1:015:1 -115 512 <
595415517 C Bi:1:015:1 0 5 = 02602060 02                            # -> 0x206002
```

I have no experience with BL616. If there is a way to dump its firmware I could send  you a copy then.

trabucayre, 2024-01-25:
Hi,
In fact I may access to one board with the buggy firmware so I can test a workaround. My idea is to use manufacturer and serial number to force ftdi type.
It also make sense to report this issue to @sipeed.

jgroman, 2024-01-27:
After looking at libftdi code I think that overriding FTDI type could subtly break it as there are some buffers and settings derived from this type at the init time.
I have prepared a PR which implements your idea to use manufacturer and serial number for activating a workaround flag which is in turn used in `FtdiJtagMPSSE::toggleClk()` to fall back to legacy toggleClk code.

I can confirm that this solution works with my Tang Primer 25K board.

trabucayre, 2024-01-29:
Thanks for your commit.
I have to review again GW5 devices: I have a lot a issues with this family (the GW5AST is my nightmare).
Have you tried to use uart interface?

closed 2024-01-29 by trabucayre

jgroman, 2024-01-29:
Glad I could help!

Regarding UART: I created a simple project just containing loopback between UART RX and TX pins and it seems to be working fine. I was also able to drop into BL616 UART console (`Ctrl-X` `Ctrl-C` `Enter`).

I'll have a look at this problem with SRAM loading when flash is empty and eventually open a new issue if I find anything worthwhile.