trabucayre/openFPGALoader #428: Terasic DE0-Nano (Cyclone IV) board

sevan, 2024-02-04:
I have a 1st gen de0-nano board, and `openFPGALoader` sees the device when I scan the USB bus, but if I try to detect it, it is not detected unless I specify the cable to use. Is that expected or should the utility make the correct decision by default based on what's scanned? I'm using openFPGALoader v0.11.0

I've yet to try writing to it :) getting there, slowly.
```
openFPGALoader --scan-usb
Bus device vid:pid       probe type      manufacturer serial               product
020 013    0x09fb:0x6001 usb-blaster     Altera                           USB-Blaster
```

```
openFPGALoader --detect
No cable or board specified: using direct ft2232 interface
unable to open ftdi device: -3 (device not found)
JTAG init failed with: unable to open ftdi device
```

```
openFPGALoader --detect -c usb-blaster
index 0:
	idcode 0x20f30dd
	manufacturer altera
	family cyclone 10 LP
	model  10CL025
	irlength 10
```

trabucayre, 2024-02-06:
`scan-usb` is used to detect all cables known by openFPGALoader.

But when you try `detect` or loading/writing a bitstream if no board (`-b`) or cable (`-c`) is provided openFPGALoader will uses the default value **ft2232** (FTDI FT2232H without buffer/level shifter).
A message is displayed to explains this situation:
```
No cable or board specified: using direct ft2232 interface
```

So in your case you have to explicitly provides `-c usb-blaster` or `-b de0nano` to be able to load/write a bitstream.

I know the default behavior is error prone but I'm hesitant to remove that to avoid breaking backward compatibility.

sevan, 2024-02-06:
> I know the default behavior is error prone but I'm hesitant to remove that to avoid breaking backward compatibility.

That's fine, just double checking.
Thanks for the explanation.

closed 2024-02-06 by sevan