trabucayre/openFPGALoader #429: How to add new spiOverJtag.bit?

nobody-1024, 2024-02-05:
Hello, I'm working on the Cmod-a7 (15t - xc7a15t-cpg236), which isn't included in the chip list. 
And I try to use xc7a35t-cpg236 interface to do it, seems work well in loading to ram, but fail in flash operation. it shows as below:

**"
Jedec ID          : eb
memory type       : 6c
memory capacity   : 00
flash chip unknown: use basic protection detection
timeout: ff ff ff
ff
wait: Error
Erasing: [                                                  ] 0.00% 
...

"**
I assume that 35t spiOverJtag.bit didn't work well with 15t?BTW the flash I'm using  is micron N25Q32, which is included in flashdb.hpp, the Jedec should be 0x0020ba17.
So the question is that how I can get my own spiOverJtag.bit for xc7a15t if my assumption is right. And this .bit file functionality is just for the operation of R/W flash by SPI? If not, I would appreciate it if you can please provide an example for it.
Thx in advance.

trabucayre, 2024-02-12:
Yes a dedicated bitstream for this model must be integrated.
This [commit](https://github.com/trabucayre/openFPGALoader/commit/d2f31860cda89e063e7ede6c1dd2b14f0e9f2f09) is a good example on how to integrate a new device into spiOverJtag. For your specific case the xdc isn't required since already present (same constraints files for all devices with the same package). So you have only to update the `Makefile` and the `build.py`.

Thanks

nobody-1024, 2024-02-21:
So you have only to update the Makefile and the build.py.

Hello.
As you mentioned, I already did so before. 
Besides  I added  
`	{0x0362e093, {"xilinx", "artix a7 15t", "xc7a15", 6}},`
into fpga_list in part.hpp and rebuild this tool for update.

Basically, I use the command " openFPGALoader -b **cmoda7_35t** -f OptSensorInterface.bit " to finish the flash process.
I think it will ok for xc7a15t to use spiOverJtag_xc7a35tcpg236.bit.gz, am I right?
If no, how can I get or generate  spiOverJtag_xc7a15tcpg236.bit?
But somehow it didn't work, also showed as below:
“
empty
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/**spiOverJtag_xc7a35tcpg236**.bit.gz
load program
Load SRAM: [==================================================] 100.00%
Done
Detail: 
Jedec ID          : a6
memory type       : c3
memory capacity   : 80
flash chip unknown: use basic protection detection
”
And every time I got different Jedec ID, which will cause fail operation to flash. Is there any other change I need to do ?

Thx in advance.

trabucayre, 2024-02-21:
35t and 15t have different internal size/capacity, bitstream from former can't be used on latter (and I have to add a check to avoid this possibility).
Has mentionned in my previous message if you read the commit mentionned you have everything to add a new bitstream.

But if you prefer I can do that by myself.

nobody-1024, 2024-02-22:
sorry, I'm new to FPGA and vivado  things, I may get some simple and primary question.

What I did:

1.Copying constr_xc7a_cpg236.xdc existed to generate new .bit file for 15t by vivado with a totally  new project, then add spiOverJtag_xc7a15tcpg236.bit.gz.

2. Also follow the commit you mentioned, make changes for both makefile and build.py to rebuild and update the tool.

What I tried:

I tried to use
 ` openFPGALoader  -c digilent -b cmoda7_15t  -f OptSensorInterface.bit --verbose-level 2`

But it still showed flash chip unkown, which means it'll read flash fail? So I stop it. The debug info showed as below
`openFPGALoader  -c digilent -b cmoda7_15t  -f OptSensorInterface.bit --verbose-level 2
empty
write to flash
Board default cable overridden with digilent
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Raw IDCODE:
- 0 -> 0x0362e093
- 1 -> 0xffffffff
Fetched TDI, end-of-chain
found 1 devices
index 0:
        idcode 0x362e093
        manufacturer xilinx
        family artix a7 15t
        model  xc7a15
        irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2024/02/04
design_name: OptSensorInterface
hour: 15:47:18
part_name: 7a15tcpg236
toolVersion: 2022.2
userID: 0XFFFFFFFF
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a15tcpg236.bit.gz
load program
Load SRAM: [==================================================] 100.00%
Done
7e fe 0 21 read 7efe0021
Detail: 
Jedec ID          : 7e
memory type       : fe
memory capacity   : 00
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
flash chip unknown: use basic protection detection
timeout: ff ff ff
ff
wait: Error
Erasing: [                                                  ] 0.00%^C`

I'm confused why I cannot get the right flash Jedec ID.
Did I use the openFPGAloader option in a wrong way? cuz when I use `openFPGALoader --scan-usb` , 
It showed 
`empty
found 5 USB device
Bus device vid:pid       probe type      manufacturer serial               product
001 009    0x0403:0x6010 FTDI2232        Digilent     210328B560F7         Digilent Adept USB Device`

However when I tried in ft2232 way with 
` openFPGALoader  -c FT2232 -b cmoda7_15t  -f OptSensorInterface.bit --verbose-level 2
empty
write to flash
Board default cable overridden with FT2232
error : FT2232 not found`

I just wondering if I use it in wrong way or I just didn't make correct spiOverJtag_xc7a15tcpg236.bit or others...

Thanks for your patience!

trabucayre, 2024-02-23:
First: about interface: you have to use `digilent` cable definition, `ft2232` is the bare minimal ftdi interface without buffer and/or level shifter. cmod A7 has a specific interface (has mentionned into scan (Yes these informations must be improved)) corresponding to the `digilent` cable.

Why **a totally new project**? If you have followed commit way no need to copy the xdc, you have just to add informations for your device and using
```
make spiOverJtag_xc7a15tcpg236.bit.gz
```
into *spiOverJtag* sub-directory

trabucayre, 2024-02-26:
Hi!
Sorry for the delay: I have just added support for the cmodA7 15t (but not tested since I haven't this board).
this [commit](https://github.com/trabucayre/openFPGALoader/commit/9e2edeb6c188eccd5e7d71f07466d227a086aaf5) adds spiOverJtag bitstream and this [commit](https://github.com/trabucayre/openFPGALoader/commit/85d9ca5d20ee80a15164c45ec86c63091b087998) adds the board support.

Now you can use:
```
openFPGALoader -b cmoda7_15t -f yourBitstream
```
to write a new bitstream to the flash

nobody-1024, 2024-02-27:
Thanks for providing this bitstream, otherwise I may need to download vivado in my server to support the make(which I'm busy with it)
And I tried your instruction, which was workable for Cmod-A7_15T. It shows as below

"
openFPGALoader -b cmoda7_15t -f OptSensorInterface.bit 
empty
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file DONE
use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a15tcpg236.bit.gz
load program
Load SRAM: [==================================================] 100.00%
Done
Detected: Macronix MX25L3233F 256 sectors size: 128Mb
00000000 00000000 00000000 00
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
"

Thank you again!

closed 2024-02-28 by nobody-1024