trabucayre/openFPGALoader #430: Unable to program CertusPro NX dv

vishnutskumar, 2024-02-08:
[image]

The error is thrown at line 71 in latticeBitParser.cpp
[image]

The bitstream generated by Radiant(lattice tool) has 0xbe instead of 0xbd or 0xbf 
[image]

trabucayre, 2024-02-08:
Have you enabled some options?
maybe @bg-gsl has more idea. In fact I haven't an FPGA to this family...
If nothing I have to study documentation to this something like that.

Thanks

bg-gsl, 2024-02-09:
Hej! :)

May I ask whether the bitstream is encoded? i.e. whether you generated it with the setting "IP Evaluation" being active in Project -> Active Strategy -> Bitstream Setting in Radiant?

vishnutskumar, 2024-02-09:
hey @bg-gsl , yes the bitstream is set with IP evaluation ON. If it isn't ON, I get the following error while trying to export the design
[image]

bg-gsl, 2024-02-10:
Hej!
Thanks for the info.
This behaviour is the expected one. I have plans to look into add the capability of loading encrypted bitstreams to Lattice Nexus FPGAs, but I cannot guarantee when nor if it will work. Probably I should have documented it somehow: @trabucayre do you know how this could be done?

About your error: I guess it's because you don't have a license for Radiant for CertusPro-NX devices. While it is not needed if you're targeting Crosslink or Certus FPGAs, you have to buy a license for targeting CertusPro https://www.latticesemi.com/en/Support/Licensing

bg-gsl, 2024-02-13:
Hej!

I've created a pull request that should enable downloading of encrypted bitstreams on Lattice Nexus FPGAs https://github.com/trabucayre/openFPGALoader/pull/432
Let me know if things work!

vishnutskumar, 2024-02-13:
Hey,

Thanks for the update, it works really well now!

closed 2024-02-13 by vishnutskumar