trabucayre/openFPGALoader #433: Enable to flash spartan-6

Dv-p, 2024-02-21:
Hello !!

I am trying to flash my spartan-6 FPGA without ISE, finally I got this repo. And it works great for programming.
But not able to flash my bit file. Its prompting "Wait: error  timeout: ff ff".  Is there anything I should be aware of ?
Attaching log file for your reference.

Can you help me with this ? 

Thanks !!

[flashing_log.txt](https://github.com/trabucayre/openFPGALoader/files/14359807/flashing_log.txt)

trabucayre, 2024-02-21:
SPI Flash access is done using a specific bitstream loaded into RAM. This one is used as a bridge between JTAG and SPI. If you check **spiOverJtag** directory you could see many bitstreams because it depends on the family, size and package. openFPGALoader know which bitstream must be used according to the board definition or the information provided by the user.

I see you use `-b arty` ie an artix A7 35T. openFPGALoader will use corresponding spiOverJtag bitstream not compatible with your FPGA (spartan 6).

Could you share the exact model you use?

trabucayre, 2024-02-22:
Yes you have to use `spiOverjtag_xc6slx9qg144.bit.gz` with
```
openFPGALoader --fpga-part xc6slx9qg144 -f youBitFile.bit
```

Dv-p, 2024-02-22:
I did but unable to open it, gave permission using "chomd 777" but didn't work.
[image]

Dv-p, 2024-02-22:
Ohh my bad, I copied your cmd and added my bit file.

I used this command:
`sudo openFPGALoader --fpga-part xc6slx9tqg144 -f yourbitfile.bit`

Thanks for your help !!

trabucayre, 2024-02-22:
sorry for the typo.
Great!
Thanks for the feedback

Dv-p, 2024-03-09:
Hello !!

Sorry to bother again but now I am trying to flash my spartan6 using pi4 via JTAG (I have jtag cable on board). I have read installing process so "-DENABLE_LIBGPIOD=ON" enabled it and built accordingly. Then I the run following command:
`openFPGALoader --cable=libgpiod --pins=26:24:25:27 --detect -v`

But its not working and my pi gets shutdown, adding snapshot of error:
[image]

Can you please help me with this? Is there any issue while builting openFPGAloader? Or is the pin I have assigned are not proper?

Thanks !!

closed 2024-03-18 by Dv-p