trabucayre/openFPGALoader #434: Feature request: Support Lilygo T-FPGA board with GWU2X USB-to-JTAG bridge

janschiefer, 2024-02-21:
Hi,

I just picked up an interesting and cheap FPGA board - Lilygo T-FPGA with an Gowin GW1NSR-LV4CQN48PC6/15 and an ESP-S3 as an IoT solution.

Interestingly this board seems t use the relatively new GWU2X application specific chip as an USB to JTAG-brigde.

Any chance flashing this board over USB will be supported in the future?

Specs: https://www.lilygo.cc/products/t-fpga
GIthub repository: https://github.com/Xinyuan-LilyGO/T-FPGA/tree/main
Board schematic: https://github.com/Xinyuan-LilyGO/T-FPGA/blob/main/schematic/T-FPGA-V1.0%20CORE.pdf
Documentation for the GWU2X USB to JTAG bridge: https://alcom.be/uploads/Gowin-UG1000-1.0E_GWU2X-Datasheet.pdf

trabucayre, 2024-02-21:
I have already seen this board, and I think I have to bought one (or ask for free sample ;-) ).
The main issue with gowin's stuff is the totally lake of documentations, most of the time you have partial/incomplete, wrong documentations.

janschiefer, 2024-02-21:
Do you speak Mandarin? ... https://cdn.gowinsemi.com.cn/UG1003.pdf

trabucayre, 2024-02-22:
Not really, but this doc explain how to use a gowin's library to use this chip no?
My interest isn't to use an external library but implement the protocol ;-)
But I assume wireshark will be my friend.

trabucayre, 2024-03-05:
I have received my board today.
After a quick search on gowin website I have found a doc about protocol (instructions to send).
It looks really similar to FTDI's MPSSE.
I think it's not possible to use directly libftdi but this seems not really hard to implement. I will try it ASAP.

janschiefer, 2024-03-05:
Very nice!

zJudGenie, 2024-03-14:
I also bought this cool board, I'm trying to use it with the toolchain yosis+nextpnr+gowin_pack+openFPGAloader.
The GW1NSR-4C is supported by the apicula project but I'm struggling to flash the bitstream to the FPGA through the Gowin Programmer as it is complaining about a wrong ID CODE and I don't know why.

So I'm waiting that the GWU2X will be supported by openFPGAloader because it is a really cool project, I can help with testing stuff.

Also, I found a little bit of documentation [here](https://www.gowinsemi.com/en/support/database/1685/) that should be helpful to add the support.

Thank you.

trabucayre, 2024-05-21:
Hi.
Gowin GWU2X and lilygo T-FPGA are now supported.
Is it good to you?

zJudGenie, 2024-05-22:
Yes, thank you, I tried flashing to SRAM with `-m` and it works, when I try to write to flash with `-f` I get, during the Erase FLASH phase, `Write failed with error -7, Failed to program FPGA: std:exception`

trabucayre, 2024-05-22:
Thanks for your feedback: I will try with a different computer to see if it's an hardware issue.
Could you share your setup ? OS / VM ?
Thnaks

zJudGenie, 2024-05-22:
I tried with this example https://github.com/Xinyuan-LilyGO/T-FPGA/tree/main/example/FPGA/led/src.
I synthesized it with the Gowin IDE educational 1.9.9 on Windows and then I used Ubuntu 22.04 on WSL to flash it, using openFPGALoader

trabucayre, 2024-05-22:
Thanks for informations. Is it possible to you to test with a native linux? I have seen lot of issues with WSL (and I have to find why).

zJudGenie, 2024-05-22:
I noticed this strange behavior, it seems that when I use the verbose flag the IDCODE gets corrupted
[image]

zJudGenie, 2024-05-22:
And then, trying to detect the FPGA with `openFPGALoader --detect --cable gwu2x` gives the same error

zJudGenie, 2024-05-22:
This is the output I get when, after rebooting the device, I try to flash it with `-f -v` so in verbose mode
[image]

trabucayre, 2024-05-22:
This timeout (-7) is fixed: stupid optimisation before pushing: 65356== 0x10000 ie needs 17bits but I have used a 16bits variable...
But now it seems to have a problem with the bitstream: I obtain an error (mismatch between read checksum and computed). Maybe another issue with a variable or a size. But I have to check if the same FPGA and an FTDI I have a similar error or if it's gwu2x related bug.

zJudGenie, 2024-05-22:
If you have an FTDI and a 13-pin flat cable you can directly access JTAG lanes using the white connector as you can see here
[image]

trabucayre, 2024-05-24:
With tangnano4k Flash sequence is working: I have to recheck everything here to see where is issue with gwu2x.