trabucayre/openFPGALoader #436: Missing delay after loading spioverjtag?

UweBonnes, 2024-02-26:
Hello,
first loading the spioverjtag bitfile and then writing the flash with skipping the load step detects the flash, also the flash is not know yet.

> ./openFPGALoader -b vmm3 ../spiOverJtag/spiOverJtag_xc7s50csga324.bit.gz 
empty
Can't read iSerialNumber field from FTDI: considered as empty string
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file DONE
load program
Load SRAM: [==================================================] 100.00%
Done
> ./openFPGALoader --skip-load-bridge -b vmm3 -f /tmp/44MHz.mcs
empty
write to flash
Can't read iSerialNumber field from FTDI: considered as empty string
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file DONE
Skip loading bridge for spiOverjtag
Detail: 
Jedec ID          : 01
memory type       : 60
memory capacity   : 18
flash chip unknown: use basic protection detection
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done

Doing it in one step fails to detect the SPI flash:
> env OPENFPGALOADER_SOJ_DIR=../spiOverJtag/ ./openFPGALoader -b vmm3 -f /tmp/44MHz.mcs
empty
write to flash
Can't read iSerialNumber field from FTDI: considered as empty string
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Open file DONE
Parse file DONE
use: ../spiOverJtag//spiOverJtag_xc7s50csga324.bit.gz
load program
Load SRAM: [==================================================] 100.00%
Done
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff

If there is a way to detect fpga reboot after loading the fpga sram, does openFPGALoader wait for that event? Or must we add some delay?

trabucayre, 2024-02-26:
There is a delay after `JTSTART` and I have already tried to flash spartan7 without seeing this type of issue (I have digilent arty s7 50).
The delay is based on artix7 datasheet so maybe to small for the device you use (but this code is used for biggest devices without any problem...).
I have to look at bsd files to see if a longer delay is required or if there is a regression somewhere.

UweBonnes, 2024-02-26:
Maybe 
 attribute INSTRUCTION_CAPTURE of XC6VLX130T_FF784 : entity is
-- Bit 5 is 1 when DONE is released (part of startup sequence)
-- Bit 4 is 1 if house-cleaning is complete
-- Bit 3 is ISC_Enabled
-- Bit 2 is ISC_Done
        "XXXXXXXX01";
is usefull to wait for "DONE"?

trabucayre, 2024-03-03:
I have seen your commit and UG470. It look like but there is maybe a limitation for xc7v ?

trabucayre, 2024-03-03:
Do you plan to open a PR to integrate this?

UweBonnes, 2024-03-03:
Yes, but I must get back the hardware. Will try soon.

trabucayre, 2024-03-03:
Thanks!
In fact I have notice on ug470, a poll is also required after JPROGRAM (it's not to request this modification ;-) )

UweBonnes, 2024-03-29:
Fixed by #440

closed 2024-03-29 by UweBonnes