trabucayre/openFPGALoader #438: flash.read_id() should be reachable from the command line

UweBonnes, 2024-02-28:
Hello,

something like
openFPGALoader -c xxx -B yyy --detect-spi
will be usefull and make it easier to get into touch with a new board and check if things work. One can also use something like
openFPGALoader -c xxx -B yyy --dump-flash /tmp/bla --file-size 1
but that needs more knowledge.

trabucayre, 2024-02-28:
It's true and it's in my todo list. I have already done tests but I have to cleanup before.

trabucayre, 2024-06-09:
Hi, with [commit](https://github.com/trabucayre/openFPGALoader/commit/c468a69fc91c899abd652b074678a3a98afafb22) it's now possible to display jedec id and status register by using `--detect -f`.

UweBonnes, 2024-06-20:
Close by  implementation in commit c468a69fc91c899a

closed 2024-06-20 by UweBonnes