trabucayre/openFPGALoader #441: Cannot find board tangprimer25k

pradheepkaliraj, 2024-03-11:
Hi,

I want to use openFPGALoaded for tangprimer25k. The tool couldn't find thisboard. 
Any suggestions on how to use tang primer 25k dock will be much appreciated.

CMD : openFPGALoader -b tangprimer25k impl/pnr/fpga_project.fs
Error: cannot find board 'tangprimer25k'

trabucayre, 2024-03-11:
Hi,
Could you share openFPGALoader's version?

This board was introduces after v0.11.0 released.

pradheepkaliraj, 2024-03-14:
Hi, 

sorry for the delay. 

I have the following version.
openFPGALoader v0.11.0. 

Is there a latest version after this?

pradheepkaliraj, 2024-03-14:
I see that there is a v.0.12.0.
I will try with that once I get permissions to install it.

pradheepkaliraj, 2024-03-31:
"sudo pacman -S openFPGALoader"  always installs the version v0.11.0.

I uninstalled openFPGALoader with the following command
 sudo pacman -R openFPGALoader

Can you please tell me how to get Version [v0.12.1](https://github.com/trabucayre/openFPGALoader/releases/tag/v0.12.1)

trabucayre, 2024-04-01:
Could you try to build openFPGALoader from sources?
Or with assets ?

trabucayre, 2024-04-11:
Any news/update?

pradheepkaliraj, 2024-04-11:
Thanks for following up. 

I tried to build from the source. But I am getting this error message after cmake --

 "-   No package 'libftdi1' found".

However, when I install libftdi1, I see that it is already installed "libftdi1 is already the newest version."

I am trying to figure out what is going on.

trabucayre, 2024-07-31:
@pradheepkaliraj any news?

pradheepkaliraj, 2024-07-31:
Hi, 

Thanks for checking. I was able to install the latest version.
I am able to program Tang25k with the latest version.

closed 2024-07-31 by pradheepkaliraj

trabucayre, 2024-07-31:
Thanks for your feedback!