trabucayre/openFPGALoader #443: olimex_gatemateevb: unable to flash

schnommus, 2024-03-13:
Firstly, thanks for your fantastic work on openFPGALoader thus far!

I have been trying to flash some bitstreams to my Olimex Gatemate A1 board as follows (note: using your published v0.12 release binaries for Ubuntu 22.04):

```
[seb@zenner bin]$ sudo ./openFPGALoader -b olimex_gatemateevb ~/dev/litex-venv/litex/build/olimex_gatemate_a1_evb/gateware/olimex_gatemate_a1_evb_00.cfg.bit 
empty
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Load SRAM via JTAG: [==================================================] 100.00%
Done
Wait for CFG_DONE
```

It hangs on this line 'Wait for CFG_DONE' forever until I terminate it.
Interestingly, the `CFG_DONE_LED1` does light up, so presumably the configuration actually happened successfully (interestingly the LED chaser doesn't flash and I get no UART traffic as one should for a litex example, but that's another kettle of fish... I built with `PATH=/opt/cc-toolchain-linux/bin/p_r:$PATH PATH=/opt/cc-toolchain-linux/bin/yosys:$PATH python3 -m litex_boards.targets.olimex_gatemate_a1_evb --cpu-type vexriscv --build` if that matters).

The more interesting issue I encountered is a segfault on adding the `-f` option:

```
[seb@zenner bin]$ sudo ./openFPGALoader -b olimex_gatemateevb -f ~/dev/litex-venv/litex/build/olimex_gatemate_a1_evb/gateware/olimex_gatemate_a1_evb_00.cfg.bit 
empty
write to flash
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Segmentation fault
```

Is writing to flash supported on this board? It has an SPI flash on it so presumably it is possible but unclear to me if it is supposed to be supported.

Thanks!

trabucayre, 2024-03-13:
Thanks for your feedback!

For the first part it's weird: I have used this without being bloked or something similar, but I have to retest.
For the flash aspect I must admit not having tested by myself: I do!.

For LiteX it's also partially weird: sometime p_r produces a bitstream not fully working (but led chaser is always working it's more issues with the CPU/UART).

Could you try to build manually from sources, maybe something wrong with the binary version? Anyway I have to fix every issues mentionned here.

Thanks!

trabucayre, 2024-03-14:
Hi,
The problem is again related to use of gpios. This feature is only available for ftdi but way to detect if gpio must be used or not is no more working.
I have to rework that.
Some FPGA's gpios are connected to the RPI2040 but unfortunately dirtyJtag doesn't provides anything to have access.

pu-cc, 2024-03-14:
@trabucayre You are right, I see what is wrong.

This commit should fix the unintended gpio access if there is no ftdi: 1304f67f1b3c4f7cfd66043ea8fdc5b8e8ce9a7b
If that helps, I'll open a PR. I still need to take a look at the flash access via the jtag-spi bypass. My board also had no flash and no pullups or pulldowns populated...

@schnommus Some Olimex boards had an incorrect VDD_PLL filter, which results in a non-locked/unstable PLL. I can offer to investigate this with you in parallel if you don't mind. Feel free to contact me so that we can keep this thread free for related topics.

schnommus, 2024-03-14:
thanks @trabucayre and @pu-cc - I think a hardware problem would match my other symptoms as I tried some examples that do not depend on the PLL lock, and they seemed to work. I will contact you separately to keep this thread clean. I guess this thread should focus on these 2 points:
1) unintended GPIO access (thanks for the patch!)
2) segfault on attempting a write to flash

trabucayre, 2024-03-14:
Thanks @pu-cc I have written same piece of code but delayed to push it to verify everything.
I'm happy to applied your patch ;-)
I have planned to check jtag-spi but I suppose you know much better this FPGA :)

For the filter I'm interested too and I think opening an issue on the board repository may help users.

Thanks!

trabucayre, 2024-03-17:
@schnommus Could you confirm @pu-cc's commits fixes openFPGALoader's issues?
Once validated I will create a new sub-release.

Thanks!

schnommus, 2024-03-17:
@trabucayre confirming I rebuilt at `1d276e` and everything looks good now, thanks!

```
[seb@zenner openFPGALoader]$ sudo ./openFPGALoader -b olimex_gatemateevb ~/dev/litex-venv/litex/build/olimex_gatemate_a1_evb/gateware/olimex_gatemate_a1_evb_00.cfg.bit 
empty
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Load SRAM via JTAG: [==================================================] 100.00%
Done
```

For flashing I can't really test it as upon a closer look, U3 isn't actually populated on my board (even though the [product page](https://www.olimex.com/Products/FPGA/GateMate/GateMateA1-EVB/open-source-hardware) states there is configuration flash!)

```
[seb@zenner openFPGALoader]$ sudo ./openFPGALoader -b olimex_gatemateevb -f ~/dev/litex-venv/litex/build/olimex_gatemate_a1_evb/gateware/olimex_gatemate_a1_evb_00.cfg.bit --unprotect-flash
empty
write to flash
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
flash chip unknown: use basic protection detection
unlock blocks
timeout: ff 1000
ff
wait: Error
```

Also worth noting after some tips from @pu-cc  that the LiteX example now works after stacking an 0603 22u on top of C39, and replaced R5 with 4R7 so that the PLL filter is better and it locks reliably. Olimex would do well to halt selling these until that's fixed...

In any case, as far as `openFPGALoader` is concerned all looks OK :)

trabucayre, 2024-03-17:
Thanks all for your help / confirm.
I haven't checked C39 nor R5 but will do (My board works as expected maybe this one is fixed ?)
U3 isn't populated too here :-/

pu-cc, 2024-03-18:
As far as I have now learned, only some of the first boards are affected by the insufficient filter.

They will open an issue in their repo and describe the workaround.

The info about flash has also been updated on their website. Unfortunately, there is no way around equipping U3 (and the pullups and pulldowns, if necessary) yourself. I did this with an MX25R643F and was able to write the flash via the jtag-spi-bypass without any problems and also verified the contents. The FPGA also starts from flash, if CFG_MD=0x0.

goran-mahovlic, 2024-03-18:
I have tried with 2x10 ohm in parallel as I do not have 4.7ohm, also placed 22uF over C39 but still did not get vga sample running.

Placed U3 - and it flash it succesfully but I probably need resistors for proper boot setings

pu-cc, 2024-03-21:
@goran-mahovlic Thanks for confirming that it worked for you, too! I would actually not set the resistance value of R5 too high, as it results in a voltage drop at high PLL loads.

But that is certainly a different matter, which has nothing to do with the openFPGALoader-related issue mentioned here. I think this is fixed and - from my side - the thread can be marked as solved.

trabucayre, 2024-03-23:
I have published a new new sub-release with all fixes. So know everything openFPGALoader's related is fixed.
Thanks all!

playduck, 2024-04-08:
@pu-cc 
> Unfortunately, there is no way around equipping U3 (and the pullups and pulldowns, if necessary) yourself.

Out of curiosity: What does "if necessary" mean here?
Are the external pull-up/down resistors R14, R15, R16 and RM3 strictly required or are there internal pull-ups such tht U3 (with C84) is sufficient?

pu-cc, 2024-04-08:
GateMate has internal pullups and pulldowns with nominal 50 kOhm, see page 79:
https://colognechip.com/docs/ds1001-gatemate1-datasheet-latest.pdf

To be on the safe side, I would still recommend to place them externally.

closed 2024-04-13 by schnommus