trabucayre/openFPGALoader #454: unable to open ftdi device: -6 (ftdi_usb_reset failed) tang nano 20k

H4-b, 2024-04-27:
Whatever i try, i still get a ftdi_usb_reset failed error when i try to load something to my tang nano 20k. (the board is brand new and working) 
[image]

It is detected by lsusb with the right IDs, and i have setup the udev rules as needed by copying the 99-ftdi.rules.
[image]

I am on debian testing.

I cant find any more steps to do in the other tickets about ftdi problems. 

What could i do?

trabucayre, 2024-04-27:
HI,
just to be sure could you try using `sudo` (and with `-b tangnano20k`?
This issue seems to be at libftdi level.

H4-b, 2024-04-27:
hello! thank you for your quick response. I get the same error messages using sudo:
 
[image]

H4-b, 2024-04-27:
Hello again!
I just tried to use it on a live Lubuntu USB, and it works there. What packets could i have on my main install that would be broken/ missing? I have installed all the packets marked as needed to install from source:
[image]

zJudGenie, 2024-04-27:
@H4-b Can you check the `dmesg` output when you plug the tang nano ?

H4-b, 2024-04-27:
here is the dmesg output:
[image]

zJudGenie, 2024-04-27:
Seems good, I've noticed that [here](https://github.com/trabucayre/openFPGALoader/issues/454#issuecomment-2080886016) you didn't use `sudo` as trabucayre said.

H4-b, 2024-04-28:
I just tried to see the dmesg after trying to use openFPGALoader -b tangnano20k --detect
It says that it disconnected the device, and only on one of the two ports associated. I have no idea if that's normal or not.
[image]

trabucayre, 2024-04-29:
Using `sudo` is finally not required: your user seems to have correct right.
the device is disconnected because to have access to them, userspace needs to detach the driver: this is the reason of this message.
The most interesting thing is you are able to use openFPGALoader with a live OS (ie more or less native): your Linux is a WSL2? I have recently observed issues in this mode: cmsis-dap not working and debugging a CPU was impossible.

If so it's why gowin is able to work: it not in WSL2 environment. When openFPGALoader shows an error associated to libftdi/libusb it's usually a lowlevel issue (ie between userspace application and hardware.

H4-b, 2024-05-01:
It seems like the bug is back. I manage to make it work sometimes, but other it just doesn't. I'm going to update this message with the dmesg and terminal output when it will want to magically work to see if there is any difference

H4-b, 2024-05-01:
I choose to upgrade to unstable to test if the issue is gone, and it is. It seems like it's one of the current testing packages that is somewhat broken. 

Tell me if you want to close this issue or not, as it's not really resolved but more circumvented.

I still have a full backup of the non-working setup if you want me to try some things to actually resolve the issue for others, or at least know where it's from.