trabucayre/openFPGALoader #456: Refresh: FAIL on Lattice MachXO2

mchitre, 2024-04-30:
I have had great success in using `openFPGALoader` to burn a program to RAM on my Lattice MachXO2 board. However, when I try to burn to flash, it fails:
```
# openFPGALoader --busdev-num 000:005 --freq 6000000 -c ft2232_b -f build_impl/DAM_build_impl.bit
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Enable configuration: DONE
Flash erase: DONE
Writing data: [==================================================] 100.00%
Done
Write program Done: DONE
Disable configuration: DONE
Enable configuration: DONE
SRAM erase: DONE
Detail:
Jedec ID          : 00
memory type       : 00
memory capacity   : 00
Skip resetting device
Refresh: FAIL
displayReadReg
	Config Target Selection : 0
	ISC Enable
	Write Enable
	Read Enable
	SDM Enable
	No err
Error: Failed to program FPGA: std::exception
```
Burning to flash from Lattice Diamond work fine. Any idea on what the problem is?

trabucayre, 2024-05-01:
I have to check with my machX02 device. Could you provides a dump with `--verbose` and the model of machXO2 you use ?

mchitre, 2024-05-01:
I am using a `MachXO2 LCMXO2-7000-ZE-3TG144I`

With verbose:
```
# openFPGALoader --busdev-num 000:005 --freq 6000000 -c ft2232_b --verbose -f build_impl/DAM_build_impl.bit
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
found 1 devices
index 0:
	idcode 0x12b5043
	manufacturer lattice
	family MachXO2
	model  LCMXO2-7000HE
	irlength 8
File type : bit
IDCode : 12b5043
displayReadReg
	Config Target Selection : 0
	ISC Enable
	Write Enable
	Read Enable
	SDM Enable
	No err
Enable configuration: DONE
Flash erase: DONE
Writing data: [==================================================] 100.00%
Done
Write program Done: DONE
Disable configuration: DONE
Enable configuration: DONE
SRAM erase: DONE
0 0 0 0 read 0
Detail:
Jedec ID          : 00
memory type       : 00
memory capacity   : 00
Skip resetting device
Refresh: FAIL
displayReadReg
	Config Target Selection : 0
	ISC Enable
	Write Enable
	Read Enable
	SDM Enable
	No err
Error: Failed to program FPGA: std::exception
```

trabucayre, 2024-05-01:
For a similar device I haven't a different dump: your looks like something for ECP5.
Which version you use: I see `LCMXO2-7000HE` but since a recent commit there is no more HE but ZE, lattice support was also improved/fixed.

mchitre, 2024-05-01:
Upgraded to latest. Problem persists.
```
# openFPGALoader -V
openFPGALoader v0.12.1

# openFPGALoader --busdev-num 000:005 --freq 6000000 -c ft2232_b --verbose -f build_impl/DAM_build_impl.bit
empty
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
found 1 devices
index 0:
	idcode 0x12b5043
	manufacturer lattice
	family MachXO2
	model  LCMXO2-7000ZE
	irlength 8
File type : bit
IDCode : 12b5043
displayReadReg
	Config Target Selection : 0
	ISC Enable
	Write Enable
	Read Enable
	SDM Enable
	BSE Error Code
		No err
b3bdffff
Enable configuration: DONE
Flash erase: DONE
Writing data: [==================================================] 100.00%
Done
Write program Done: DONE
Disable configuration: DONE
Enable configuration: DONE
SRAM erase: DONE
0 0 0 0 read 0
Detail:
Jedec ID          : 00
memory type       : 00
memory capacity   : 00
Skip resetting device
Refresh: FAIL
displayReadReg
	Config Target Selection : 0
	ISC Enable
	Write Enable
	Read Enable
	SDM Enable
	BSE Error Code
		No err
Error: Failed to program FPGA: std::exception
```

mchitre, 2024-05-09:
Anything I can do to help debug?

trabucayre, 2024-05-11:
Look like the issue is near *lattice.cpp*: `post_flash_access()`
If refresh fails it will try to read flash but this is unsupported for *machXO2*/*machXO3(D)*.
Is it possible to share your bitstream, maybe using it I will be able to reproduce.

mchitre, 2024-05-11:
Here's a simplified version of my code that I can't burn into flash but works fine in RAM:
https://drive.google.com/file/d/1HF8rapJ1EhFltFHy86KzNOAWVLjb8WVR/view?usp=sharing

mchitre, 2024-05-11:
Hmm. If I try many times, sometimes the flash writing succeeds. But the code doesn't work from the flash. Loading to RAM for the same code works.

BlueSyncLine, 2024-06-09:
I observe similar behavior on a MachXO3 with flashing in general (similar zero flash IDs and capacity). v0.11.0 works fine, 0.12.0 fails.

trabucayre, 2024-06-09:
I have an issue with my machXO2 but will try with machXO3. I have rechecked the code and don't see why messages related to spiflash are displayed... Could you try with .jed to check if it's the same situation?

trabucayre, 2024-06-10:
@BlueSyncLine: could you provides details about your board. machXO FPGAs have only internal flash.

trabucayre, 2024-06-29:
@mchitre After a (too long) time I have retried your bitstream without any issues:
```bash
openFPGALoader --busdev-num 003:016 --freq 6000000 -c ft2232 --verbose -f ../../machxo2_fail/fails.bit 
empty
write to flash
Can't read iSerialNumber field from FTDI: considered as empty string
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 1 devices
index 0:
        idcode 0x12b5043
        manufacturer lattice
        family MachXO2
        model  LCMXO2-7000ZE
        irlength 8
File type : bit
IDCode : 12b5043
displayReadReg
        Config Target Selection : 0
        Done Flag
        SDM Enable
        BSE Error Code
                No err
b3bdffff
Enable configuration: DONE
Flash erase: DONE
Writing data: [==================================================] 100.00%
Done
Write program Done: DONE
Disable configuration: DONE
Refresh: DONE
```

I am just unable to see why, in your case,  openFPGALoader try to access to an external flash.
Just to see: is it an native linux?

mchitre, 2024-06-29:
It is a Mac OS X

trabucayre, 2024-06-29:
Will try with my (montherey)