trabucayre/openFPGALoader #460: Attempting to Flash SPI PROM of a custom board - Dragon-L

AnantaSrikar, 2024-05-28:
I am attempting to flash a bitstream to a custom FPGA board, called the [Dragon-L](https://www.knjn.com/FPGA-Dragon-L.html). It has a `Spartan-6 XC6SLX25T-CSG324` FPGA on board. I am using an Altera USB Blaster as the JTAG adapter.

I have built and tried the latest openFPGALoader. Upon trying to flash a bitstream, I got the following error:
```
❯ openFPGALoader -c usb-blaster ledblink.bit
empty
JTAG init failed with: Unknown device with IDCODE: 0x24024093 (manufacturer: 0x049 (Xilinx), part: 0x20 vers: 0x2
```

Then, I did some research and figured out I could add my the IDCODE to `src/part.hpp`. I added the following and recompiled:
```cpp
        {0x04004093, {"xilinx", "spartan6", "xc6slx25",        6}},
+       {0x24024093, {"xilinx", "spartan6", "xc6slx25T",       6}},
        {0x04008093, {"xilinx", "spartan6", "xc6slx45",        6}},
```

Then, I was able to flash to the SRAM of the FPGA with the following output:
```
❯ ./openFPGALoader -c usb-blaster ledblink.bit --verbose 
empty
found 1 devices
index 0:
	idcode 0x24024093
	manufacturer xilinx
	family spartan6
	model  xc6slx25T
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2024/05/22
design_name: ledblink.ncd
hour: 10:26:28
part_name: 6slx25tcsg324
toolVersion: 0xFFFFFFFF
userID: 0xFFFFFFFF
load program
Load SRAM: [==================================================] 100.00%
Done
Shift IR 35
ir: 1 isc_done 1 isc_ena 0 init 1 done 1
```

I wanted to take it a step ahead and see if I could flash the same onto the onboard SPI flash, and tried adding the `-f` flag, and errored out with the following output:
```
❯ ./openFPGALoader -c usb-blaster -f ledblink.bit --verbose
empty
write to flash
found 1 devices
index 0:
	idcode 0x24024093
	manufacturer xilinx
	family spartan6
	model  xc6slx25T
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2024/05/22
design_name: ledblink.ncd
hour: 10:26:28
part_name: 6slx25tcsg324
toolVersion: 0xFFFFFFFF
userID: 0xFFFFFFFF
Can't program SPI flash: missing device-package information
```

I did some more research and I figured out I could add the `--fpga-part ` flag, and ran the same, but I still got an error:
```
❯ ./openFPGALoader -c usb-blaster --fpga-part xc6slx9csg324 -f ../../fpgaconf/captures/ledblink.bit --verbose
empty
write to flash
found 1 devices
index 0:
	idcode 0x24024093
	manufacturer xilinx
	family spartan6
	model  xc6slx25T
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2024/05/22
design_name: ledblink.ncd
hour: 10:26:28
part_name: 6slx25tcsg324
toolVersion: 0xFFFFFFFF
userID: 0xFFFFFFFF
use: /home/srikar/projects/dragon-l/openFPGALoader/spiOverJtag/spiOverJtag_xc6slx9csg324.bit.gz
load program
Load SRAM: [==================================================] 100.00%
Done
Shift IR 11
ir: 1 isc_done 0 isc_ena 0 init 1 done 0
Register raw value: 0x0
CRC Error      No CRC error
Part Secured   0
MMCM lock      0
DCI match      0
EOS            0
GTS CFG B      0
GWE            0
GHIGH B        0
MODE           0
INIT Complete  0
INIT B         0
Release Done   0
Done           0
ID Error       No ID error
DEC Error      0
XADC Over temp 0
STARTUP State  0
Reserved       0
BUS Width      x1
Reserved       0
cd 96 76 80 read cd967680
Detail: 
Jedec ID          : cd
memory type       : 96
memory capacity   : 76
RDSR : 68
WIP  : 0
WEL  : 0
BP   : a
TB   : 1
SRWD : 0
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash

```

Since the error mentioned the flag --uprotect-flash, I tried that too, but still failed with a timeout error:
```
❯ ./openFPGALoader -c usb-blaster --fpga-part xc6slx9csg324 -f ../../fpgaconf/captures/ledblink.bit --unprotect-flash --verbose 
empty
write to flash
found 1 devices
index 0:
	idcode 0x24024093
	manufacturer xilinx
	family spartan6
	model  xc6slx25T
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2024/05/22
design_name: ledblink.ncd
hour: 10:26:28
part_name: 6slx25tcsg324
toolVersion: 0xFFFFFFFF
userID: 0xFFFFFFFF
use: /home/srikar/projects/dragon-l/openFPGALoader/spiOverJtag/spiOverJtag_xc6slx9csg324.bit.gz
load program
Load SRAM: [==================================================] 100.00%
Done
Shift IR 11
ir: 1 isc_done 0 isc_ena 0 init 1 done 0
Register raw value: 0x0
CRC Error      No CRC error
Part Secured   0
MMCM lock      0
DCI match      0
EOS            0
GTS CFG B      0
GWE            0
GHIGH B        0
MODE           0
INIT Complete  0
INIT B         0
Release Done   0
Done           0
ID Error       No ID error
DEC Error      0
XADC Over temp 0
STARTUP State  0
Reserved       0
BUS Width      x1
Reserved       0
ad b4 40 80 read adb44080
Detail: 
Jedec ID          : ad
memory type       : b4
memory capacity   : 40
RDSR : 2c
WIP  : 0
WEL  : 0
BP   : 3
TB   : 1
SRWD : 0
flash chip unknown: use basic protection detection
unlock blocks
timeout: ff ff ff
ff
wait: Error
```

I've double checked the pinout for the SPI PROM and it should be fine. I am not sure what I'm missing here, and I hope to make some changes and add official support to this board.

The flash chip in use is an `25P80VP` by STMicroelectronics. I believe [this](https://www.digikey.com/htmldatasheets/production/8267/0/0/1/m25p80.html) is the datasheet for the same.

I'm not sure what I'm missing here, hope to find some insights and direction for the same. Thanks!

trabucayre, 2024-05-28:
Yes. To be able to use openFPGALoader with a FPGA not already supported you have to update list of idcode (Yes, A better way is to import for some family all variants to avoid this type of issue).

But to update the flash with a bitstream you needs a sort of bridge (called spiOverJtag). For Xilinx you can't directly have access to the flash by using JTAG and you needs to load a gateware to interconnect these two interfaces. But unfortunately one bitstream is required by FPGA type (family, size, package), this why spiOverJtag directory contains a bunch of bitstreams.
Here you have tried to load a bitstream dedicated to another model (size/model) this why openFPGALoader is unable to access FLASH chip.

For this step you have to follow approach similar to this [commit](https://github.com/trabucayre/openFPGALoader/commit/7bbaef0c879c9f3c30b69fb78a39a044d95d06b5)

AnantaSrikar, 2024-05-28:
Thank you so much for getting back to me. The way Xilinx accesses the SPI over JTAG is interesting. I will try looking around and getting the bitstream for `xc6slx25tcsg324.bit.gz`. I think it should be pretty straightforward after that. Any idea where I should be looking for the same?

AnantaSrikar, 2024-05-28:
I figured out how to generate the bitstream:

- Make a new file named `constr_xc6s_t_csg324.ucf` with the same contents of `spiOverJtag/constr_xc6s_csg324.ucf` since the pinout is the same (I triple checked it [here](http://www.xilinx.com/support/documentation/user_guides/ug385.pdf)). 
- Made the appropriate changes to the `spiOverJtag/build.py` to accomodate for the new chip.
- Setup a python virtual environment with `edalize` and appended the `$PATH` with the Xilinx ISE binaries
- Ran `python3 build.py xc6slx25tcsg324`. This generated the `spiOverJtag.bit` file in the `tmp_xc6slx25tcsg324`
- Ran `gzip -9 -c tmp_xc6slx25tcsg324/spiOverJtag.bit > spiOverJtag_xc6slx25tcsg324.bit.gz` to get the `.bit.gz` version of the bitstream.
- Finally, I ran the flash command to get a similar error:
```
❯ ./openFPGALoader -c usb-blaster --fpga-part xc6slx25tcsg324 -f ledblink.bit --verbose --unprotect-flash
empty
write to flash
found 1 devices
index 0:
	idcode 0x24024093
	manufacturer xilinx
	family spartan6
	model  xc6slx25T
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2024/05/22
design_name: ledblink.ncd
hour: 10:26:28
part_name: 6slx25tcsg324
toolVersion: 0xFFFFFFFF
userID: 0xFFFFFFFF
use: /home/srikar/projects/dragon-l/openFPGALoader/spiOverJtag/spiOverJtag_xc6slx25tcsg324.bit.gz
load program
Load SRAM: [==================================================] 100.00%
Done
Shift IR 35
ir: 1 isc_done 1 isc_ena 0 init 1 done 1
a1 3d bb 80 read a13dbb80
Detail: 
Jedec ID          : a1
memory type       : 3d
memory capacity   : bb
RDSR : bc
WIP  : 0
WEL  : 0
BP   : 7
TB   : 1
SRWD : 1
flash chip unknown: use basic protection detection
unlock blocks
timeout: ff ff ff
ff
wait: Error
```
I've pushed all the changes I've made [here](https://github.com/AnantaSrikar/openFPGALoader/commit/4fd3a46cc84a87567f65040b9e662ec6c3b1dc7b) for your reference.
Any idea what I'm missing here?

trabucayre, 2024-06-01:
Jedec ID seems wrong must be 0x2014. There is a problem with communication but I suspect the chip is in quad mode instead of SPI mode. But I must check if it's possible for your board.
Maybe I have to order one, but it's a bit expensive.

trabucayre, 2024-06-01:
In fact there is no schematics available... Could you also try with other JTAG cable? like ftdi?

AnantaSrikar, 2024-06-01:
I'm not too sure about quad mode. I will also contact the board manufacturer to see if I can get more details and maybe the schematic. Regarding another JTAG cable, I'll have to order one. 

Regarding another JTAG cable, do you think I can use [this](https://www.adafruit.com/product/2264) to program the FPGA? I've never worked with FPGAs before, so I'm a little unsure about how I should go ahead.

AnantaSrikar, 2024-06-02:
I've contacted the Dragon-L board developer regarding the SPI pinout. I'm attaching the schematic for the same. The board **does not** support the quad mode.

[image]

trabucayre, 2024-06-02:
any FTDI do the job, the only you have to check is the pin voltage: usually for a jtag probe, jtag pins are powered by a power pins from the board.
Your schematics confirms constraints file pinout, it's weird openFPGALoader and spiOverJtag are perfectly compatible with spartan6...

AnantaSrikar, 2024-06-02:
That is weird. I will try dirty JTAG on Bluepill, an ST-Link v2 clone and a Raspberry Pi pico to see if  I can get it to work. In the event it doesn't, I will purchase [this](https://my.mouser.com/ProductDetail/FTDI/C232HM-DDHSL-0?qs=EB8w3yL7SqMfAQpmyuyp%252Bw%3D%3D) FTDI cable and try again. 

In the meantime, is there any signal that you would want me to verify by using an oscilloscope?

trabucayre, 2024-06-02:
I think the best way is I have a physical to a similar board. Its not always easy to see reasons why an issue like here.

AnantaSrikar, 2024-06-04:
I flashed a Bluepill board I had lying around with the `dirtyJtag` firmware. The SRAM flashing works just fine, like the USB Blaster. It fails whenever I try to flash to SPI flash.

I'm starting to think this is because I'm not using all the JTAG pins. I'm only using the `TCK`, `TDO`, `TDI`, and `TMS` pins that can be easily connected by the 2.54 headers. The JTAG port uses a smaller form factor 14-pin connector, which I'm still on the hunt for. 

I will try buying a smaller 14-pin connector and connecting the remaining pins (`TRST` and `SRST`) to see if it works. Either that or the generated bitfile is somehow not the right one. 

Attaching a picture for your reference.

[image]

trabucayre, 2024-06-04:
Don't buy anything: it's not required. Tests with altera blaster and dirtyJTAG are enough: the problem isn't related to the cable...

trabucayre, 2024-06-04:
I must admit its hard to said. Flow looks good and your constraints file too... Maybe trying to disable flash at powerup time using a wire between CS and VCC to have an empty FPGA (flash be reserved by gateware loaded at boot time).

AnantaSrikar, 2024-06-08:
I will try disabling the flash as the FPGA powers up and let you know how that.

I've had success flashing the SPI of the FPGA using the [KNJN USBFX2-JTAG development board](https://www.knjn.com/ShopJTAGcables_USB.html) and the [KNJN JTAG cable](https://www.knjn.com/ShopJTAGcables.html) along with the `FPGAconf` portable executable. That `exe` application allows me to select the bitfile I want to flash, and click on flash boot for it to write it to SPI. So what I can also try writing the SpiOverJtag.bit to the flash and see how openFPGALoader works with that.

AnantaSrikar, 2024-07-11:
A small update on this. I was trying to get OpenOCD running with the altera blaster and my FPGA. I think the Altera Blaster is at fault here, but I'm not sure.

Flashing to SRAM works perfectly as expected. 

Here is dragon-l-sram.cfg
```
source [find interface/altera-usb-blaster.cfg]
source [find cpld/xilinx-xc6s.cfg]

adapter speed 10000

init

xc6s_program xc6s.tap

pld load 0 ledglow.bit
shutdown
```

The command run for the above is as follows:
```bash
openocd -f dragon-l-sram.cfg
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : usb blaster interface using libftdi
Info : This adapter doesn't support configurable speed
Info : JTAG tap: xc6s.tap tap/device found: 0x24024093 (mfg: 0x049 (Xilinx), part: 0x4024, ver: 0x2)
Warn : gdb services need one or more targets defined
shutdown command invoke
```

Now, I have a configuration that is probably working, since the SPI flash gets corrupted as after running the command, and the SPI flash is detected correctly.

Below is configuration
```
source [find interface/altera-usb-blaster.cfg]
source [find cpld/xilinx-xc6s.cfg]
source [find cpld/jtagspi.cfg]

adapter speed 100

init
jtagspi_init 0 bscan_spi_xc6slx25t.bit
jtagspi_program ledglow.bit 0x0
xc6s_program xc6s.tap
flash verify_bank 0 ledglow.bit
shutdown
```

This is the output of the same:
```bash
openocd -f dragon-l-spi.cfg
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : usb blaster interface using libftdi
Info : This adapter doesn't support configurable speed
Info : JTAG tap: xc6s.tap tap/device found: 0x24024093 (mfg: 0x049 (Xilinx), part: 0x4024, ver: 0x2)
Info : JTAG tap: xc6s.tap tap/device found: 0x24024093 (mfg: 0x049 (Xilinx), part: 0x4024, ver: 0x2)
Info : Found flash device 'st m25p80' (ID 0x142020)
Info : sector 0 took 636 ms
Info : sector 1 took 591 ms
Info : sector 2 took 558 ms
Info : sector 3 took 576 ms
read 244520 bytes from file ledglow.bit and flash bank 0 at offset 0x00000000 in 2.178538s (109.610 KiB/s)
[... 110 lines cut ...]
diff 109 address 0x0000007b. Was 0x00 instead of 0x88
diff 110 address 0x0000007c. Was 0x00 instead of 0x31
diff 111 address 0x0000007d. Was 0x00 instead of 0x41
diff 112 address 0x0000007e. Was 0x00 instead of 0x3d
diff 113 address 0x0000007f. Was 0x00 instead of 0x08
diff 114 address 0x00000080. Was 0x00 instead of 0x31
diff 115 address 0x00000081. Was 0x00 instead of 0x61
diff 116 address 0x00000082. Was 0x00 instead of 0x09
diff 117 address 0x00000083. Was 0x00 instead of 0xee
diff 118 address 0x00000084. Was 0x00 instead of 0x31
diff 119 address 0x00000085. Was 0x00 instead of 0xc2
diff 120 address 0x00000086. Was 0x00 instead of 0x04
diff 121 address 0x00000087. Was 0x00 instead of 0x02
diff 122 address 0x00000088. Was 0x00 instead of 0x40
diff 123 address 0x00000089. Was 0x00 instead of 0x93
diff 124 address 0x0000008a. Was 0x00 instead of 0x30
diff 125 address 0x0000008b. Was 0x00 instead of 0xe1
diff 126 address 0x0000008d. Was 0x00 instead of 0xcf
diff 127 address 0x0000008e. Was 0x00 instead of 0x30
More than 128 errors, the rest are not printed.
```

I'm not sure what I'm missing here, but I'm able to detect the flash chip, and I'm able to (sort of) wipe it. Any idea what the issue might be?

I got the `bscan_spi_xc6slx25t.bit` from [here](https://github.com/quartiq/bscan_spi_bitstreams/blob/master/bscan_spi_xc6slx25t.bit).

AnantaSrikar, 2024-07-11:
I think I figured out why its happening. I modified the configuration file as follows:
```
source [find interface/altera-usb-blaster.cfg]
source [find cpld/xilinx-xc6s.cfg]
source [find cpld/jtagspi.cfg]

adapter speed 100

init
jtagspi_init 0 bscan_spi_xc6slx25t.bit
xc6s_program xc6s.tap
flash write_image erase ledglow.bit 0;
flash verify_bank 0 ledglow.bit
shutdown
```
and upon running the above configuration, I got the following output:
```
openocd -f dragon-l-spi.cfg
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : usb blaster interface using libftdi
Info : This adapter doesn't support configurable speed
Info : JTAG tap: xc6s.tap tap/device found: 0x24024093 (mfg: 0x049 (Xilinx), part: 0x4024, ver: 0x2)
Info : JTAG tap: xc6s.tap tap/device found: 0x24024093 (mfg: 0x049 (Xilinx), part: 0x4024, ver: 0x2)
Info : Found flash device 'st m25p80' (ID 0x142020)
Error: Cannot enable write to flash. Status=0x00
Error: Sector erase failed.
Error: failed erasing sectors 0 to 3
auto erase enabled
```

I think we have to send a WREN (Write enable) instruction before we attempt to write to the flash (as per the [datasheet](https://www.digikey.com/htmldatasheets/production/8267/0/0/1/m25p80.html#pf13))

trabucayre, 2024-07-12:
Hi,
I have received a board to do some tests, but unfortunately I have an issue with this board (maybe around the USB connector).
the WREN bit is always set before erasing or writing a sector. But this command (and many others) needs to have exactly 8bits (no more, no less), otherwise command is discarded.

AnantaSrikar, 2024-07-12:
Thank you for the update. I'm glad you received a board for testing, as that should indeed help with faster debugging. It's interesting that the WREN bit is always set before erasing or writing a sector.

Since you mentioned that the WREN isn't the issue, could you elaborate on what you think might be causing the problem?

Any additional information you can provide would be helpful in understanding the issue better. Let me know how else I can assist in troubleshooting this problem.

trabucayre, 2024-08-03:
Hi Sorry for late answer: Issue with my board was related to a wrong USB connector (VCC connected to GND, and GND to VCC). 
After some tests I have observed the FPGA immediately reboot after spiOverJtag's bitstream load. And managed to test some different bitstream with a classic flow instead of using edalize: to conclude the issue was related to *bitgen*'s configuration where *Unused Pins* was not set to *float*.
With https://github.com/trabucayre/openFPGALoader/commit/bdaba6e7da1006fb0bd5892d2ee265a7074777a3 I'm able to detect and program the SPI flash.

Next step: adding spiflash and board definition, and a edalize's PR to allows user to provides more configurations.

trabucayre, 2024-08-03:
- spi flash definition added with https://github.com/trabucayre/openFPGALoader/commit/0d657f0f652e798f674cd58ad30d11be9cbe8039
- board definition added with https://github.com/trabucayre/openFPGALoader/commit/9edf1edb3fceab6a833198c87ae48d33f2425bac

closed 2024-08-03 by trabucayre

epoch101codes, 2024-08-03:
This is such good news! Thank you so much for debugging and finding a fix. However, I still have one small issue. I'm trying to use the `-B dragonL` flag and it doesn't seem to be working well with the `-f` flag. I'm able to flash to SPI ROM if i mention the `fpga-part` seperately. Below is the log file for your reference:

```
srikar@testboi:~/bitfiles $ sudo openFPGALoader -c usb-blaster -B dragonL ledglow.bit --verbose
empty
found 1 devices
index 0:
	idcode 0x24024093
	manufacturer xilinx
	family spartan6
	model  xc6slx25T
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2015/12/11
design_name: LEDglow.ncd
hour: 09:09:44
part_name: 6slx25tcsg324
toolVersion: 0xFFFFFFFF
userID: 0xFFFFFFFF
load program
Load SRAM: [==================================================] 100.00%
Done
Shift IR 35
ir: 1 isc_done 1 isc_ena 0 init 1 done 1
srikar@testboi:~/bitfiles $ sudo openFPGALoader -c usb-blaster --fpga-part xc6slx25tcsg324 -f ledblink.bit --verbose
empty
write to flash
found 1 devices
index 0:
	idcode 0x24024093
	manufacturer xilinx
	family spartan6
	model  xc6slx25T
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2024/05/22
design_name: ledblink.ncd
hour: 10:26:28
part_name: 6slx25tcsg324
toolVersion: 0xFFFFFFFF
userID: 0xFFFFFFFF
use: /usr/local/share/openFPGALoader/spiOverJtag_xc6slx25tcsg324.bit.gz
load program
Load SRAM: [==================================================] 100.00%
Done
Shift IR 35
ir: 1 isc_done 1 isc_ena 0 init 1 done 1
20 20 14 10 read 20201410
JEDEC ID: 0x202014
Detected: ST M25P80 16 sectors size: 8Mb
RDSR : 0x00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
00000000 00000000 00000000 00
start addr: 00000000, end_addr: 000d0000
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
srikar@testboi:~/bitfiles $ sudo openFPGALoader -c usb-blaster -B dragonL -f ledglow.bit --verbose
empty
write to flash
found 1 devices
index 0:
	idcode 0x24024093
	manufacturer xilinx
	family spartan6
	model  xc6slx25T
	irlength 6
File type : bit
Open file DONE
Parse file DONE
bitstream header infos
date: 2015/12/11
design_name: LEDglow.ncd
hour: 09:09:44
part_name: 6slx25tcsg324
toolVersion: 0xFFFFFFFF
userID: 0xFFFFFFFF
use: dragonL
Error: fail to open dragonL
Error: Failed to program FPGA: Error: fail to open dragonL
srikar@testboi:~/bitfiles $ 
```

Please let me know if I'm missing something. Thank you once again!

trabucayre, 2024-08-03:
- `-B`(upper 'b') is to specify the bitstream to you use as bridge. openFPGALoader will try to open the corresponding bitstream, and here, fails because there is no file called *dragonL* in the current directory.
- `-b` (lower 'b') is to specify the board

To program the dragonL you have simply to use:
```bash
openFPGALoader -c usb-blaster -b dragonL -f YourBitStream
```

epoch101codes, 2024-08-03:
Oh got it, that works perfectly. Thank you!