trabucayre/openFPGALoader #464: XC7A200 on TE0712: Boot from flash only after one time programming with Xilinx tools

UweBonnes, 2024-06-21:
I got new TE0712 modules. On a working setup, I replaced an old programmed  modules with the new module. I could load the bitfile to FPGA, and the  board worked as expected until reboot/repower. Then I wrote the binfile to flash. It compared okay. But the module, more exactly both modules i bought,  did not boot from flash, neither by reset or repower.

To get the module bootable, I needed to use the Xilinix programmer once to program the flash. After the one-time xilinx programming, I can erase and reprogram flash and the setup boots from flash as expected.

Any ideas what openFPGAloader is missing on a virgin setup of FPGA/Flash to get flash bootable?

trabucayre, 2024-06-21:
Is your bitstream is configured in quad spi mode? If so, maybe the SPI flash has quad mode disabled?
I have observed this situation after a wrong status register write where the quad enable bit was unset.

trabucayre, 2024-06-29:
Any news? Thanks

UweBonnes, 2024-07-01:
Sorry, I am missing knowledge to answer you question. Are these settings related?
1:set_property BITSTREAM.GENERAL.COMPRESS TRUE [current_design]
2:set_property BITSTREAM.CONFIG.CONFIGRATE 66 [current_design]
6:set_property BITSTREAM.CONFIG.SPI_32BIT_ADDR YES [current_design]
7:set_property BITSTREAM.CONFIG.SPI_BUSWIDTH 4 [current_design]
8:set_property BITSTREAM.CONFIG.UNUSEDPIN PULLDOWN [current_design]

trabucayre, 2024-07-30:
Sorry for the delay.
According to the configuration the bitstream is read in quad mode. So if the flash chip is not configured to support this mode, loading the file will fails.
Could you check your flash configuration (openFPGALoader may do that with `-f --detect`.

UweBonnes, 2024-07-30:
This is the flash after one time Xilinx programming and then multiple times openFPGALoader programming:> ~/devel/openFPGALoader/build/openFPGALoader -b te0712_8 -f --detect
empty
write to flash
No cable or board specified: using direct ft2232 interface
Jtag frequency : requested 15.00MHz  -> real 15.00MHz 
protect_flash: use: /usr/local/share/openFPGALoader/spiOverJtag_xc7a200tfbg484.bit.gz
load program
Load SRAM: [==================================================] 100.00%
Done
Shift IR 35
ir: 1 isc_done 1 isc_ena 0 init 1 done 1
JEDEC ID: 0x010219
Detected: spansion S25FL256S 512 sectors size: 256Mb
RDSR : 0x00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0

Configuration Register
RDCR   : 02
FREEZE : 0
QUAD   : 1
TBPARM : 0
BPNV   : 0
TBPROT : 0
Done

"QUAD :1" seems to be loaded. Can openFPGALoader set it? A grep through the code did not ring a bell for me.

trabucayre, 2024-08-02:
Currently the only modification applied by openFPGALoader is about protection. Enabling/disabling quad mode isn't supported (but it's possible to integrate this option).

UweBonnes, 2024-11-15:
Should this be fixed by faa1fc76fcf "core,xilinx,device: added option/methods to enable/disable quad mode on SPI Flash"?

trabucayre, 2024-11-16:
Sorry to have missed to update to issue. Indeed, this commit allows users to enable/disable quad mode.

Sorry again

closed 2025-07-09 by UweBonnes