trabucayre/openFPGALoader #466: Artix Ultrascale XCAU25P support for flash(mcs file)

cshekharOlympus, 2024-06-25:
I am trying to add support for flash for Artix Ultrascale XCAU25P, could some one point me in the right direction. I see I need to generate a bit file to add support. Thanks.

petersn, 2025-02-12:
I second this, I'd like to flash my xcau10p parts. I see xcau25p and xcau15p support now, but what about other Artix UltraScale+ parts? If I'd like to add support for such parts how can I help?

trabucayre, 2025-02-12:
Principle is similar to this commit https://github.com/trabucayre/openFPGALoader/commit/29d2264e1594b279b49dc883b1f35fa8652efa6e
- you have to create the xdc with constraints if not already present
- Makefile and build.py must be updated with model part size
- and, if required, verilog code must also be updated if some subilities needs to be added