trabucayre/openFPGALoader #467: Flashing a MT25QL256 attached to Cyclone 10CL025 fails (mostly)

kawk, 2024-06-30:
Hi,

I tried to program a  QSPI Flash (MT25QL256) attached to a Intel 10LP (10CL025YU256I7G) using openFPGALoader (branch master as of today, 1b00940). The output indicates a successful upload of the spiOverJtag config (spiOverJtag_10cl025256.rbf) and writing to flash (identified as Micron N25Q256).

However, the content of the flash is only "touched", but not correctly copied from the input RBF: Mostly 0xFF, with contiguous blocks of 256 zero bytes at 0 and then repeated every 64 kByte. Only the last byte of these blocks sometimes is neither 0 or 0xff...

Command: `openFPGALoader -v -c usb-blaster -B spiOverJtag_10cl025256.rbf --external-flash top_auto.rpd`

I normally program the device using Quartus Programmer with "Factory default enhanced SFL image" for "10CL025Y" and device "MT25QL256". Does specific information about this flash have to be added somewhere? 

Kolja

kawk, 2024-06-30:
I learned that Quartus Programmer changes the flash NVCFG from 0xFFFF to 0x8FEE when programming it for the first time. That changes initial addressing mode, for example, and is also needed for further operation (a NIOS2 running from this flash). I suspect that the addressing mode might cause such issues.

kawk, 2024-06-30:
Indeed: After the nonvolatile config register (NVCFG) was forced back to 0xFFFF, programming with openFPGALoader succeeded. Now however my NIOS can't run. Even if it's possible to change the logic and firmware to run with NVCFG 0xFFFF, that would make it incompatible with boards programmed with Quartus Programmer.. Maybe the flash programming code should be enhanced to observe the addressing mode currently active?

kawk, 2024-06-30:
I validated two possible approaches for the case that flash is configured with default 4-byte adressing in NVCFG:
 - Issue "`EXIT 4-BYTE ADDRESSING (0xB7)`" once at the very beginning (I put it at the end of `read_id()`) to put the flash in 3-byte addressing mode (volatile, until next reset), or
 - If flash is currently in 4-byte addressing mode (LSB of _flag_ status register is set), use 4-byte address commands even if address is below 2^24 (no manipulation of active configuration).
Either way, I additionally have to implement some method to change the NVCFG register, because I need to have it changed from the factory default for normal operation, when flashing brand new chips for the first time. What would be be an appropriate way to  provide such functionality? An extra command line argument?

kawk, 2024-06-30:
Nb - the NVCFG may put the flash into much more exotic operational modes that prevent programming, such as automatic address wraparound after so many bytes (e.g. 32). I presume it might be a good idea to check for these and disable them at least volatile. However, I don't oversee what flash chips provide what modes...

blurbdust, 2026-11-03:
Same problem on a Stratix V board (Microsoft Storey Peak, 5SGSD5, on-board FT232H) with a Micron N25Q256A (JEDEC ID 0x20ba19), with the second approach above implemented and tested. Patch at the end, it applies to current master (e3d8d94).

**Cause.** The flash's nonvolatile configuration register reads 0xAFFE. Bit 0 = 0 selects 4-byte address mode at power-up and after a software reset, which is what AS x4 boot of these flashes uses. `SPIFlash` sends the 3-byte address instructions (03h, 02h, 20h, D8h) below 16 MB, so the flash takes the next byte on the bus as the fourth address byte. The 66h/99h reset in the `SPIFlash` constructor reloads 4-byte mode from that register, so sending E9h earlier, in `Altera::prepare_flash_access()`, does not help. I tried it.

**Reproduce.** `--dump-flash --file-size 1048576` at offset 0: 963,460 of the 1,048,576 bytes differ from the image in the flash, and nothing reports an error. Each 256-byte read at address A returns one lost byte, then the flash content at `(A << 8) | x`. Flag status register bit 0 reads 1 (4-byte mode) right after the constructor's reset.

To show the addressing directly, fill a spare sector with a pattern where every 4-byte word is its own address (written with the patched build, since the stock one cannot write there):

```
python3 -c "import struct; open('pattern.bin','wb').write(b''.join(struct.pack('>I', 0xC00000 + 4*i) for i in range(16384)))"
openFPGALoader -c ft232 --fpga-part 5sgsd5 -f -o 0xC00000 pattern.bin
openFPGALoader -c ft232 --fpga-part 5sgsd5 --dump-flash -o 0xC0C000 --file-size 256 read.bin
xxd -l 32 -g 4 read.bin
```

Expected at 0xC0C000:

```
00000000: 00c0c000 00c0c004 00c0c008 00c0c00c  ................
00000010: 00c0c010 00c0c014 00c0c018 00c0c01c  ................
```

v1.0.0 returns one lost byte (`ff`), then the words from 0xC000F8, which is `(0xC0C000 << 8) | 0xF8` modulo the 32 MB size:

```
00000000: ff00c000 f800c000 fc00c001 0000c001  ................
00000010: 0400c001 0800c001 0c00c001 1000c001  ................
```

With the patch the same read returns the expected words.

**Fix.** For Micron flashes larger than 16 MB, read flag status register bit 0 after the reset and the ID read. When it is set, send the standard instructions (03h, 02h, 20h, 52h, D8h) with a 4-byte address at every address. Otherwise nothing changes. The flash is left in 4-byte mode, which AS boot needs. I used the standard instructions because 12h, 21h and DCh exist only on the N25Q256A83 line items (datasheet Rev. T, table 18, note 14).

With the patch, on the same board:
- 1 MB dumps at offset 0 and at 16 MB match the image
- a full 32 MB dump matches
- `-f --verify` of a 12 MB image succeeds, and the FPGA boots from the rewritten flash

Tested with v1.0.0 and master e3d8d94, on Ubuntu 20.04. I can open this as a PR if that is easier.

<details>
<summary>spiFlash: Micron: use 4-byte addresses when the flash is in 4-byte address mode (spiFlash.cpp, spiFlash.hpp, +26 -13)</summary>

```diff
diff --git a/src/spiFlash.cpp b/src/spiFlash.cpp
index 9d57554..e0ac2f1 100644
--- a/src/spiFlash.cpp
+++ b/src/spiFlash.cpp
@@ -87,11 +87,23 @@
 SPIFlash::SPIFlash(FlashInterface *spi, bool unprotect, int8_t verbose):
 	_spi(spi), _verbose(verbose), _jedec_id(0),
 	_flash_model(NULL), _unprotect(unprotect), _must_relock(false),
-	_status(0)
+	_status(0), _addr_4b_mode(false)
 {
 	reset();
 	power_up();
 	read_id();
+
+	/* A Micron flash larger than 16MB leaves reset in the address mode set
+	 * by its nonvolatile configuration register, and Intel Active Serial
+	 * configuration needs 4-byte address mode. In this mode every
+	 * instruction takes a 4-byte address: check flag status register bit 0.
+	 */
+	if (_flash_model && (_jedec_id >> 24) == 0x20 &&
+			_flash_model->nr_sector > 256) {
+		uint8_t fsr;
+		_spi->spi_put(FLASH_RFSR, NULL, &fsr, 1);
+		_addr_4b_mode = (fsr & 0x01) != 0;
+	}
[... 69 lines cut ...]
 		addr_len = 4;
@@ -308,7 +320,7 @@ int SPIFlash::read(int base_addr, uint8_t *data, int len)
 	uint8_t tx[len+addr_len];
 	uint8_t rx[len+addr_len];
 
-	if (read_cmd == FLASH_4READ)
+	if (addr_len == 4)
 		tx[i++] = (uint8_t)(0xff & (base_addr >> 24));
 	tx[i++] = (uint8_t)(0xff & (base_addr >> 16));
 	tx[i++] = (uint8_t)(0xff & (base_addr >>  8));
diff --git a/src/spiFlash.hpp b/src/spiFlash.hpp
index a767ab3..9ebb84f 100644
--- a/src/spiFlash.hpp
+++ b/src/spiFlash.hpp
@@ -180,6 +180,7 @@ class SPIFlash {
 		bool _unprotect; /**< allows to unprotect memory before write */
 		bool _must_relock;
 		uint8_t _status;
+		bool _addr_4b_mode; /**< 4-byte address mode: 4-byte address for all instructions */
 };
 
 #endif  // SRC_SPIFLASH_HPP_
```

</details>