trabucayre/openFPGALoader #468: Segmentation fault with unknown flash chip ID

festlv, 2024-07-03:
I'm trying to program an external flash for ECP5 with current master (1b00940d13f5c1df5832197a3ccdb6ed6d9dce74) but I'm getting a segfault during execution of openFPGALoader.

GDB session:
```gdb
Reading symbols from ./openFPGALoader...
(gdb) run --cable cmsisdap --write-flash --unprotect-flash --bitstream /home/reinis/LinuxCNC-RIO/Output/Concrete/Firmware/rio.bit -v
Starting program: /home/reinis/openFPGALoader/build/openFPGALoader --cable cmsisdap --write-flash --unprotect-flash --bitstream /home/reinis/LinuxCNC-RIO/Output/Concrete/Firmware/rio.bit -v
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
empty
write to flash
[New Thread 0x7ffff7a1c6c0 (LWP 4806)]
Found 1 compatible device:
	0x0d28 0x0204 0x2 DAPLink CMSIS-DAP
[New Thread 0x7ffff721b6c0 (LWP 4807)]
	VID: 0d28
	PID: 0204
	serial number : 0700000105d4ff393733424e43155972a5a5a5a597969908
	firmware version : 0254
	hardware capabilities : 13
	SWO trace buffer size : NA
	max packet cnt : 04
	max packet size : 64
Hardware cap 00 01 13
found 1 devices
index 0:
	idcode 0x41111043
	manufacturer lattice
	family ECP5
	model  LFE5U-25
	irlength 8
File type : bit
IDCode : 41111043
displayReadReg
	Config Target Selection : 0
	Std PreAmble
	BSE Error Code
		No err
Open file DONE
Parse file b3bdffff
DONE
bitstream header infos
Part: LFE5U-25F-6CABGA381
idcode: 41111043
Enable configuration: DONE
SRAM erase: DONE
c8 40 15 c8 read c84015c8
Detail: 
Jedec ID          : c8
memory type       : 40
memory capacity   : 15
RDSR : 0x1c
WIP  : 0
WEL  : 0
BP   : 7
TB   : 0
SRWD : 0
flash chip unknown: use basic protection detection
unlock blocks

Thread 1 "openFPGALoader" received signal SIGSEGV, Segmentation fault.
SPIFlash::disable_protection (this=this@entry=0x7fffffffd900) at /home/reinis/openFPGALoader/src/spiFlash.cpp:705
705		for (int i = 0; i < _flash_model->bp_len; i++)
(gdb) bt
#0  SPIFlash::disable_protection (this=this@entry=0x7fffffffd900) at /home/reinis/openFPGALoader/src/spiFlash.cpp:705
#1  0x0000555555782dc2 in SPIFlash::erase_and_prog (this=this@entry=0x7fffffffd900, base_addr=base_addr@entry=0, data=data@entry=0x7ffff68fd010 "\377\377\377\275\263\377\377\377\377;", 
    len=len@entry=582341) at /home/reinis/openFPGALoader/src/spiFlash.cpp:430
#2  0x000055555578485e in SPIInterface::write (this=this@entry=0x5555559d0bd8, offset=offset@entry=0, data=0x7ffff68fd010 "\377\377\377\275\263\377\377\377\377;", len=582341, 
    unprotect_flash=unprotect_flash@entry=true) at /home/reinis/openFPGALoader/src/spiInterface.cpp:159
#3  0x00005555557e16d4 in Lattice::program_extFlash (this=0x5555559d0b80, offset=0, unprotect_flash=<optimized out>) at /home/reinis/openFPGALoader/src/lattice.cpp:800
#4  0x00005555557ec578 in Lattice::program (this=<optimized out>, offset=<optimized out>, unprotect_flash=<optimized out>) at /home/reinis/openFPGALoader/src/lattice.cpp:883
#5  0x0000555555682830 in main (argc=<optimized out>, argv=<optimized out>) at /home/reinis/openFPGALoader/src/main.cpp:606
(gdb) 

```

trabucayre, 2024-07-04:
When the spi flash chip isn't supported there is no _flash_model this why bp_len/bp_offset access was wrong.

May be fixed by https://github.com/trabucayre/openFPGALoader/commit/f6b6f6f9d8e3de198b5258829fffc7e844ef75c6.

Thanks!

festlv, 2024-07-04:
It is indeed fixed. Thank you for your effort!

FYI: one minor warning is emitted after the fix: unused variable https://github.com/trabucayre/openFPGALoader/blob/master/src/spiFlash.cpp#L897

closed 2024-07-04 by festlv

trabucayre, 2024-07-06:
Indeed!
It's now fixed.
Thanks again!