trabucayre/openFPGALoader #469: Cannot find Board or Programming cable

Tommy0295, 2024-07-04:
Hello,

i tried using this project for Programming a custom artix-7 board via a J-Link Programming Cable.
I installed it in my msys2 with command "pacman -S mingw-w64-ucrt-x86_64-openFPGALoader"

when i try to program a arty-a7 board, an error occurs:
`$ openFPGALoader -b arty_a7_a35t ./FPGA_Programming/test.bit
Error: cannot find board 'arty_a7_a35t'`

As well when i try to program my custrom board via the J-Link Cable.
`$ openFPGALoader -c JLink --vid 1366 --pid 0101 ./FPGA_Programming/test.bit
error : JLink not found
`

Do i need to do something that the USB devices are reachable for the msys2 shell? Unfortunatley I have poor understanding of this Linux/OS stuff :(

Thanks for your time

trabucayre, 2024-07-04:
There is a typo:
- `arty_a7_a35t` -> `arty_a7_35t`
- `JLink` -> `jlink` or `jlink_base`

Unlike windows, GNU/Linux and openFPGALoader are case sensitive ;-)

Tommy0295, 2024-07-04:
Thank you so much for the fast answer <3

The cable as well as the ftdi chip on the arty board seem to be found. But now it seems to get no connection to the FPGA. Both FPGAs are XC7A35T. They should be supported.
`$ openFPGALoader -b arty_a7_35t --vid 0403 --pid 6010 ./FPGA_Programming/test.bit`
`Cable VID overridden`
`Cable PID overridden`
`unable to open ftdi device: -3 (device not found)`
`JTAG init failed with: unable to open ftdi device`

`$ openFPGALoader -c jlink ./FPGA_Programming/test.bit`
`Error: no device found`
`JTAG init failed with: can't find compatible device`

Do i need to pass some other parameter to the program?

trabucayre, 2024-07-30:
Could you try using `sudo`? If your user has no write access to the device you can't use the JTAG interface.
These errors are classic when the user is not in the required group