trabucayre/openFPGALoader #470: How to add a Xilinx Spartan 7 Part?

trlgha, 2024-07-12:
I was excited to find this tool for a small GUI project on a Mac with Homebrew.  I have the tool downloaded and the Diligent HS3 programmer.  I checked support for the Spartan 7 series and attempted to run the app for a Spartan 7 xc7s6-2ftgb196.  When I run the following command:

`openFPGALoader -f --fpga-part xc7s6 -c digilent_hs3 my_file.mcs`

the following output is generated:

```
empty
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
JTAG init failed with: Unknown device with IDCODE: 0x03622093 (manufacturer: 0x049 (Xilinx), part: 0x1b vers: 0x0
```

I assume the issue is likely find that the specific part I need is not supported.  I am wondering if there is a path or instructions about how to add a new FPGA part to the Spartan 7 family?  I scanned all the documentation but maybe I missed it.

trabucayre, 2024-07-13:
To add a new IDCODE for an FPGA family already supported you have to update [part.hpp](https://github.com/trabucayre/openFPGALoader/blob/master/src/part.hpp#L65) with the new IDCODE.
See [this commit](https://github.com/trabucayre/openFPGALoader/commit/1042b6c2697e984d0c7f2821ef312613139a55f3) for an example on how to do it.

Thanks

trabucayre, 2024-07-23:
Updating `part.hpp` is mainly required by openFPGALoader to know which vendor driver must be used. This step is enough to load a bitstream to SRAM (volatile memory).
Here you wish to write the bitstream in the SPI flash.
This requires an additional step because with FPGA's like spartan7 it's not possible to access the SPI flash without a piece of logic loaded into the FPGA(see [this commit](https://github.com/trabucayre/openFPGALoader/commit/dc437957985852d8a83b04192a904e0ec8d4204a)).
Since for an FPGA for a specific vendor, family and size it may have more than one package you have to provides a more complete detail: this why [this line](https://github.com/trabucayre/openFPGALoader/blob/master/src/board.hpp#L122) provides not only the family and size but also the package format.

trlgha, 2024-07-24:
Well, I spent all day on this sequence and trying to figure out what steps are missing from the process.  I read through many past issues and the closest I could find was this one ([spiOverJTAG.bit](https://github.com/trabucayre/openFPGALoader/issues/429)) I understand that I need to program the SPI Flash through an additional step.  I assume the additional changes you referenced in the previous post would perform that step.

I assume the error message from my second post is that the file is not being generated.  I have performed the following steps several times over from scratch and still get the same error message:

- brew install --only-dependencies openfpgaloader
- brew install cmake pkg-config zlib gzip
- git clone https://github.com/trabucayre/openFPGALoader
- Modified src/part.hpp
[image]

- Modified src/board.hpp
[image]

- Modified spiOverJTAG/build.py
[image]

- Modified spiOverJTAG/Makefile
[image]

- Created spiOverJTAG/constr_xc7s_ftgb196.xdc
```
set_property CFGBVS VCCO [current_design]
set_property CONFIG_VOLTAGE 3.3 [current_design]
set_property BITSTREAM.CONFIG.SPI_BUSWIDTH {4} [current_design]
set_property BITSTREAM.GENERAL.COMPRESS TRUE [current_design]

set_property -dict {PACKAGE_PIN C11 IOSTANDARD LVCMOS33} [get_ports {csn}];
set_property -dict {PACKAGE_PIN B11 IOSTANDARD LVCMOS33} [get_ports {sdi_dq0}];
set_property -dict {PACKAGE_PIN B12 IOSTANDARD LVCMOS33} [get_ports {sdo_dq1}];
set_property -dict {PACKAGE_PIN D10 IOSTANDARD LVCMOS33} [get_ports {wpn_dq2}];
set_property -dict {PACKAGE_PIN C10 IOSTANDARD LVCMOS33} [get_ports {hldn_dq3}];
``` 
- cd openFPGALoader
- mkdir build
- cd build
- cmake ..
- make -j
- ./openFPGALoader -f --fpga-part xc7s6 -c digilent_hs3 ../my_file.mcs
[image]

Still worth mentioning that the folder location referenced in the error does not contain any files related to the openFPGALoader tool.  I'll ask again if there is a path problem?
[image]

Also, as referenced in the linked thread, I thought maybe the makefile in the spiOverJTAG needed to be called independent of the build/makefile but it generated an error.  I installed python and edalize toolset in a virtual environment. It creates a tmp_xc7s6... folder but not a .bit.gz file.
```shell
(venv) trlgha@MacBook-Pro-16-2021-M1 spiOverJtag % make spiOverJtag_xc7s6ftgb196.bit.gz                                
./build.py xc7s6ftgb196
vivado -notrace -mode batch -source spiOverJtag.tcl
make[1]: vivado: No such file or directory
make[1]: *** [spiOverJtag.xpr] Error 1
Traceback (most recent call last):
  File "/Users/trlgha/Documents/openFPGALoader/spiOverJtag/venv/lib/python3.12/site-packages/edalize/edatool.py", line 472, in _run_tool
    cp = run(
         ^^^^
  File "/opt/homebrew/Cellar/python@3.12/3.12.4/Frameworks/Python.framework/Versions/3.12/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['make']' returned non-zero exit status 2.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/trlgha/Documents/openFPGALoader/spiOverJtag/./build.py", line 213, in <module>
    backend.build()
  File "/Users/trlgha/Documents/openFPGALoader/spiOverJtag/venv/lib/python3.12/site-packages/edalize/edatool.py", line 245, in build
    self.build_main()
  File "/Users/trlgha/Documents/openFPGALoader/spiOverJtag/venv/lib/python3.12/site-packages/edalize/vivado.py", line 113, in build_main
    self._run_tool("make", args)
  File "/Users/trlgha/Documents/openFPGALoader/spiOverJtag/venv/lib/python3.12/site-packages/edalize/edatool.py", line 495, in _run_tool
    raise RuntimeError(_s)
RuntimeError: '['make']' exited with an error: 2
make: *** [tmp_xc7s6ftgb196/spiOverJtag.bit] Error 1
(venv) trlgha@MacBook-Pro-16-2021-M1 spiOverJtag %
``` 

Is there any detailed documentation that outlines the full process?  I would love to see similar successful output of loading the SPI Flash chip shown in the linked post.

- Tim

trabucayre, 2024-07-27:
`SpiOverJtag`is more or less independant to build step: bitstreams are build manually and versionned.
So yes you have to do a `make  spiOverJtag_xc7s6ftgb196.bit.gz`only once.
In your case, since you build a bitstream for a spartan7, you needs to have `vivado` in your `$PATH` (it's usually done with `source /VIVADO_INSTALL_PATH/VIVADO_VERSION/settings64.sh`

trlgha, 2024-07-28:
Since I am working on a Mac and don't have Vivado, is there a chance this chip will be added soon?  Or is someone able to run the make command and post the file for the Spartan 7 s6 chip?

trabucayre, 2024-07-30:
Could you try with this file ?
[spiOverJtag_xc7s6ftgb196.bit.gz](https://github.com/user-attachments/files/16422259/spiOverJtag_xc7s6ftgb196.bit.gz)

trlgha, 2024-07-31:
Yes, that file works.

I didn't know after building the source files with make that I also had to run the following command
`sudo make install`
which put the bit.gz files in the empty directory I asked about earlier.  

Also, I had to change the run command to include the full chip part name.  After doing these things, I was able to successfully program the chip.  This is good news and a great tool for those with Mac's that need FPGA programming capability.  Thanks!

trabucayre, 2024-08-01:
I pushed support: now this spartan7 is officially supported.
Thanks

closed 2024-08-03 by trabucayre