trabucayre/openFPGALoader #472: Gowin GW2A-LV18EQ144C8 external flash troubles

andrew-palamar, 2024-07-21:
Hello. Trying to use your program to flash / dump external flash **25Q64** on custom board with **GW2A-LV18EQ144C8** FPGA part in LQ144 package. The reading seems to work fine at first look. But I`m unable to flash correctly. What can I do to debug and / or fix this issue? Could you point me in right direction? Thanks in advance!

## Writing
```
sudo ./openFPGALoader -c ft2232 --fpga-part GW2A-18C --external-flash bitstream.fs
```
### Results
```
$ sudo ./openFPGALoader -c ft2232 --fpga-part GW2A-18C --external-flash bitstream.fs

empty

write to flash

Can't read iSerialNumber field from FTDI: considered as empty string

Jtag frequency : requested 6.00MHz   -> real 6.00MHz  

Parse file Parse bitstream.fs: 

Done

DONE

after program flash: displayReadReg 00006020

  Memory Erase

  Done Final

  Security Final

Erase SRAM DONE

Jtag frequency : requested 10.00MHz  -> real 10.00MHz 

JEDEC ID: 0xef4017

Detected: Winbond W25Q64 128 sectors size: 64Mb

JEDEC ID: 0xef4017

Detected: Winbond W25Q64 128 sectors size: 64Mb

RDSR : 0x00

WIP  : 0

WEL  : 0

BP   : 0

TB   : 0

SRWD : 0

00000000 00000000 00000000 00

start addr: 00000000, end_addr: 000e0000

timeout: 0 ff 1

00

wait: Error

write en: Error

Erasing: [==================================================] 100.00%

Fail

Error: write to flash failed

Error: Failed to program FPGA: Error: write to flash failed
```

## Reading

```
sudo ./openFPGALoader -c ft2232 --fpga-part GW2A-18C --dump-flash --external-flash dump.bin --file-size 8388608
```

### Results

```
$ sudo ./openFPGALoader -c ft2232 --fpga-part GW2A-18C --dump-flash --external-flash dump.fs --file-size 8388608

empty

Can't read iSerialNumber field from FTDI: considered as empty string

Jtag frequency : requested 6.00MHz   -> real 6.00MHz  

Erase SRAM Jtag frequency : requested 10.00MHz  -> real 10.00MHz 

JEDEC ID: 0xef4017

Detected: Winbond W25Q64 128 sectors size: 64Mb

dump flash (May take time)

Open dump file DONE

Read flash : [==================================================] 100.00%

Done
```

Had no chance to compare dumped binary with bitstream file because it needs to be converted first.

trabucayre, 2024-07-21:
Is it's possible you share the board's schematic? I observe a timeout, I suspect is it related to write enable command. This may happen if some glitches appears during the WEN send operation.

andrew-palamar, 2024-07-21:
Thank you for fast reply. 
Please look at attached SPI flash connection schematic.
[image]

Using Gowin programmer this flash writes just fine
```
Info:  Target Cable: Gowin USB Cable(FT2CH)/0/0/null
Info:  Operation "exFlash Erase,Program,Verify thru GAO-Bridge" is starting on device-1...
Info:  Loading GAO-Bridge...
Info:  spi flash Winbond W25Q64BV found!
Info:  Erasing 221 sectors from address 0x000000 to 0x0dd000...
Info:  Program and Verify flash successfully.
Info:  Cost 28.99 second(s)
```

Could it be that W25Q64**BV** needs special treatment?

Could I just Increase Timeout delay in the code? If so, please kindly tell where it should be done.

trabucayre, 2024-07-21:
Is the board is publicly available?
If the spiflash is known by openFPGALoader this mean it has already used with another board: spiFlash driver is not related to gowin but it's a independant piece of code using FPGA's driver as lowlevel.

In fact it's hard to say without the hardware because gowin provides documentation with a really poor quality, so maybe something it's not enough stable in your specific case....

andrew-palamar, 2024-07-21:
No. This board is not available publicly.
I see. Still could I somehow increase timeout delay to check if it helps somehow?

Well I could try to inspect flashing process of the same bitstream on Gowin programmer and openFPGALoader using Logic Analyzer to see the difference. Will post information if will wind something interesting.

One more question. Could I set different JTAG Frequency instead of 10MHz to match Gowin Programmer`s frequency?
P.S. Datasheet says that W25Q64BV could be used @ 80MHz.

trabucayre, 2024-07-30:
Frequency for Gowin's FPGAs is fixed at constructor time: it's something I have to modify to adapt/check requested frequency.
I have tested with two sipeed boards (nano20k and primer20k) without issues. I suspect an issue related to the hardware (but maybe I'm wrong).