trabucayre/openFPGALoader #476: openfpgaloader reports success (returns 0) when programming fails

AndrewD, 2024-07-25:
Efinix: programming errors still result in a success return code so there is no way for a script to detect errors from the return value.

This interface needs to be able to propagate errors (not return void):
https://github.com/trabucayre/openFPGALoader/blob/a39524636bcc0846a799ed1fb874c891fc350c1b/src/efinix.hpp#L30

trabucayre, 2024-07-25:
In fact for all vendor's drivers `program` method throw exceptions when something is wrong.
But it's true, changing `program` to a non void method may be more easy to manage
And for efinix most of errors doesn't produces exception.

AndrewD, 2024-07-25:
If throwing an exception is the existing pattern then following this for Efinix would be sufficient for now - openfpgaloader just returning a non-zero value on error is all we need.

trabucayre, 2024-07-29:
with [this commit](https://github.com/trabucayre/openFPGALoader/commit/a7cb7ec050a0f7a3fe6319b0a0bf1e5b7b38f736) all errors are correctly handled by program() method.
It's good to you?

harbaum, 2025-02-13:
Not meaning to hijack this. But does the efinix jtag loading work at all? I do have a t20-bga256 demo board and this is IMHO supposed to be supported.

`openFPGALoader -b trion_t20_bga256_jtag ./outflow/efinix_blink.bit`

Looks like everything worked well and the FPGA even shortly stops running. But afterwards it continues to execute the bitstream that was already installed before. The bitstream was built using the current efinity software and uploads successfully using the closed source programmer.

I can dig into this a little further as my [usbmon script](https://github.com/harbaum/PICO-MPSSE/tree/main/usbmon) allows me to look into both USB communications and compare them. But before doing that I'd like to make sure that I am not doing someting fundamentally wrong here. The way I am trying to upload is supposed to work?

harbaum, 2025-02-13:
I did a quick investigation on this. 

openFPGAloader actually loads the bit file and i see its contents arriving on USB (although bit reversed as JTAG is LSB first). But it seems to be missing a byte. The bitsteam hex file is 2035950 bytes in total. These are 2035950 / 3 = 678650 bytes decoded. With my tool I actually see 678649 bytes to be sent into the FPGA, so the first observation is that one byte seems to be missing. usbmon only captures the first 32 bytes of each bulk transfer so I cannot exactly tell what byte is missing but it seems it's the last one.

The efinity programmer on the other hand sends a total of 679926 bytes which is an additional 1286 bytes over the bitfile itself. I see that it e.g. prepends the first transfer of 4096 bytes with 20 zero bytes. The following few bytes look like they come directly from the bitfile. It would need further investigation where the other 1266 bytes are being inserted and if they're also 0 bytes.