trabucayre/openFPGALoader #478: Trying to Flash xc6slx25, But Found No spiOverJtag File

JerryIchido, 2024-08-01:
Hi, 

I'm trying to flash a standalone Xilinx xc6slx25 through a Digilent HS2 cable. From browsing previous issues I learned that I need  a dedicated spiOverJtag file and specify the --fpga-part for the board to work. However, I can't find a dedicated file for xc6slx25. (There is one exist for 45 though). Trying to just specify the part number will cause a fail to open file error:

openFPGALoader --fpga-part xc6slx25csg324 -f -c digilent_hs2 Flash_Prog.mcs --verbose-level 2
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Raw IDCODE:
\- 0 -> 0x24004093
\- 1 -> 0xffffffff
\- 2 -> 0xffffffff
\- 3 -> 0xffffffff
\- 4 -> 0xffffffff
found 1 devices
index 0:
        idcode 0x4004093
        manufacturer xilinx
        family spartan6
        model  xc6slx25
        irlength 6
File type : mcs
Open file DONE
Parse file DONE
use: D:/a/msys64/ucrt64/share/openFPGALoader/spiOverJtag_xc6slx25csg324.bit.gz
Error: fail to open D:/a/msys64/ucrt64/share/openFPGALoader/spiOverJtag_xc6slx25csg324.bit.gz
Error: Failed to program FPGA: Error: fail to open D:/a/msys64/ucrt64/share/openFPGALoader/spiOverJtag_xc6slx25csg324.bit.gz

Trying to use the one for LX45 will fail as well, so they are not compatible with each other:
 openFPGALoader --fpga-part xc6slx45csg324 -f -c digilent_hs2 Flash_Prog.mcs --verbose-level 2
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Raw IDCODE:
\- 0 -> 0x24004093
\- 1 -> 0xffffffff
\- 2 -> 0xffffffff
\- 3 -> 0xffffffff
\- 4 -> 0xffffffff
found 1 devices
index 0:
        idcode 0x4004093
        manufacturer xilinx
        family spartan6
        model  xc6slx25
        irlength 6
File type : mcs
Open file DONE
Parse file DONE
use: D:/a/msys64/ucrt64/share/openFPGALoader/spiOverJtag_xc6slx45csg324.bit.gz
load program
Flash SRAM: [==================================================] 100.00%
Done
80 0 0 0 read 80000000
Detail:
Jedec ID          : 80
memory type       : 00
memory capacity   : 00
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
flash chip unknown: use basic protection detection
timeout: c 61 e2
c
wait: Error
write en: Error
Erasing: [==================================================] 100.00%
Fail

On the website xc6slx25 is listed as supported. What should I do? Should I try to create my own spiOverJtag file? Or can I use something compatible?

Thank you very much!

JerryIchido, 2024-08-02:
We were able to create the xc6slx25 SPI over JTAG file ourselves from browsing some previous issues. The flash was then successful!

closed 2024-08-02 by JerryIchido

trabucayre, 2024-08-02:
Great!
Could you open a PR with modifications and bitstream to integrate this support directly?

JerryIchido, 2024-08-02:
Sure! It will be my very first time doing this, hope I don't mess things up too badly... I will create a branch, switch to that new branch and push things there.

JerryIchido, 2024-08-02:
I was made reluctant to do the request due to commit author concerns (Not all commits were done through a personal credential). I sincerely apologize for the inconvenience. But in case future people need to replicate the bitstream I'm putting what I done in writing here:

- Clone openFPGALoader to a local folder.
- Go to /spiOverJtag, And edit build.py and Makefile the same way as shown in this [commit](https://github.com/trabucayre/openFPGALoader/commit/7bbaef0c879c9f3c30b69fb78a39a044d95d06b5). Add the board names, etc. 
- Open a command line, do python build.py [Board_name]. In the linked commit, we would replace Board_name with xc6slx45tfgg484. Unfortunately we will need to have Xilinx ISE Design Suite installed. 
- Go to the newly generated folder and copy the .bit file into the spiOverJtag folder and rename it to have the same format as other files in the folder. 
- Run the flash! No compilation is needed if we already have openFPGALoader installed.

Once again I'm sorry for not being able to contribute:(

trabucayre, 2024-08-03:
Don't worry: I will adds this bitstream by myself.

trabucayre, 2024-08-03:
Done with https://github.com/trabucayre/openFPGALoader/commit/7fc222d50f92b0a921d6838fe1442ca93fe494f7