trabucayre/openFPGALoader #487: Device Not Recognized on Windows Without Using Zadig

freshmakerzhao, 2024-09-26:
I compiled the project using MSYS2 and attempted to run it on Windows. However, the application fails to recognize the device unless I install the driver using Zadig. After installing the driver with Zadig, the device is recognized, and the download process works correctly.

Is there an alternative method to complete this process without relying on external programs like Zadig?

trabucayre, 2024-09-26:
Unfortunately with windows and MSYS2 Zadig can't be avoided. I'm not a windows specialist  but it's maybe not required when using WSL2 (but you have to install a quite full GNU/Linux distribution to avoid to install one program...).

ZhiyuanYuanNJ, 2024-10-23:
> I compiled the project using MSYS2 and attempted to run it on Windows. However, the application fails to recognize the device unless I install the driver using Zadig. After installing the driver [...]

There's no other way, hardware work requires drivers, so if you don't use manufacturer drivers, you can only use Zadig's winUSB driver.

trabucayre, 2024-10-27:
Thanks @ZhiyuanYuanNJ 
@freshmakerzhao It's fine for you. Can I close this issue?

freshmakerzhao, 2024-10-27:
@ZhiyuanYuanNJ，@trabucayre 
Thank you both for your help. You can close the issue now.

closed 2024-10-27 by trabucayre