trabucayre/openFPGALoader #494: Cannot download xc7vx690t

ZhiyuanYuanNJ, 2024-10-23:
Hi, I have a Xilinx xc7vx690t, but I couldn't find the corresponding `spiOverJtag` file in the program. The only approximate file is `spiOverJtag_xc7vx330tffg1157.bit.gz`. What should I do?

trabucayre, 2024-10-27:
You have to update spiOverJtag Makefile & build.sh to add support for your model/size.
See https://github.com/trabucayre/openFPGALoader/commit/f3a48fb3b117a123b7cdc92b558ac2bf64a6440c as an example. Not all steps are required for all devices dependending if an constraints file exist or not.

ZhiyuanYuanNJ, 2024-10-30:
Thank you, but before that, I encountered an error in writing to SRAM. 
```bash
openFPGALoader -c ch347_jtag -m spiOverJtag_xc7v_ffg1761.bit
empty
write to ram
CH347 Open Success.
JTAG TCK frequency set to 7.500 MHz

Open file DONE
Parse file DONE
load program
Load SRAM: [==================================================] 100.00%
Done
Shift IR 91
ir: 1 isc_done 0 isc_ena 0 init 1 done 0
Register raw value: 0x4000190c
CRC Error       No CRC error
Part Secured    0x0
MMCM lock       0x1
DCI match       0x1
EOS             0x0
GTS CFG B       0x0
GWE             0x0
GHIGH B         0x0
MODE            0x1
INIT Complete   0x1
INIT B          0x1
Release Done    0x0
Done            0x0
ID Error        No ID error
DEC Error       0x0
XADC Over temp  0x0
STARTUP State   0x0
Reserved        0x0
BUS Width       x1
Reserved        0x8
Close the CH347.
```