trabucayre/openFPGALoader #495: gatemate-evb : unable to open ftdi device

Suk1268, 2024-11-12:
i have been trying to configure FPGA of the olimex gatemateA1-EVB where i'm getting the following error
[image]

i am able to detect in the device manager after using zadig 
[image]

i have also referred this git issue (https://github.com/trabucayre/openFPGALoader/issues/245#issue-1337570131)
but the error persists

trabucayre, 2024-11-13:
Examples and scripts shipped with colognechip toolchain are targeted for CologneChip's gatemateEVB board.
Your board is an olimex with a different JTAG interface and a different pin mapping so you have to use the correct target.
For openFPGALoader you have to use `-b olimex_gatemateevb`

Suk1268, 2024-11-14:
thank you for the help.

closed 2024-11-14 by Suk1268