trabucayre/openFPGALoader #501: ESP32S3 USB JTAG cable maybe possible

emard, 2024-12-28:
Currently I'm experimenting to get ESP32S3 USB JTAG which is supposedly
able to route its native JTAG to external GPIO, if that is possible then we
can use it as "cable" to program FPGA.

https://github.com/emard/esp32s3-jtag

Miniature ESP32S3 with USB-C are available for 7$ like
XIAO ESP32S3 MINI Mouser P/N: 713-113991114

So here is my wish/idea for openFPGALoader to support it if possible.

emard, 2025-01-01:
openocd driver esp_usb_jtag is fixed here's my source
https://github.com/emard/openocd/blob/master/src/jtag/drivers/esp_usb_jtag.c

with this esp jtag it is very fast, speed is comparable to ft2232.

So I tried to make openfpgaloader driver
https://github.com/emard/openFPGALoader/blob/master/src/esp_usb_jtag.cpp

but  it doesn't work, esp protocol is simple but has details
which I don't understant so it's not so easy =>expert needed

trabucayre, 2025-01-01:
I have ordered one waveshare boards based on the same devices: will try as soon as I have received samples.
Thanks to point this feature!

emard, 2025-01-01:
Great, I recommend they are so cheap and so fast.

Supported should be esp32 S2, S3, C3, C6 and maybe
few others. "Magic" iomux values that route jtag to external
pins for s3 are undocumented but on the forums obviously
I found required info :)

trabucayre, 2025-01-12:
I have received my board, I suppose firmware is correctly flashed. I observe issues for read command. Could you provides a dump to compare?

emard, 2025-01-12:
hardware runs this JTAG so firmware is here only
to initialize registers, assigns signals to pins and wait.

openocd was only that worked (after I fixed)
I haven't tried any longer reads but now I can pull
your source and check if I can see something strendge

emard, 2025-01-12:
Here is compiled dump of my firmware
small 14-pin S3 board pinout is on my git
https://github.com/emard/esp32s3-jtag

emard, 2025-01-12:
I have also some old devkit-1 ESP32S3 board

https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/user_guide_v1.0.html

If you have this board and can't get it working
I can compile and see what happens

trabucayre, 2025-01-14:
Sorry for the delay to answer: no needs to prepare code/firmware. I'm now able to read IDCODE \o/.

I have to cleanup my local directly before pushing modifications in your branch. After that I see some improvements.

Thanks!

emard, 2025-01-14:
eheheeee wonderful!

One notice while I was experimienting with driver in openocd,
this usb-jtag works good only if repeated commands are
repeated using its special usb command for repeating

RUNTEST IDLE    2 TCK   1.00E-02 SEC;

here is this function used for repeating see function
esp_usb_jtag_write_rlestream()

https://github.com/espressif/openocd-esp32/blob/6bcb287a9deec4f9f68f6a8939b60792f55e9927/src/jtag/drivers/esp_usb_jtag.c#L482

I if replace it this CMD_REP command with simple for-loop
that repeats "cmd" command "ct" number of times, then
something doesn't work and usb timeout error appears.

edanuff, 2025-03-21:
Very curious about any updates around this.  Currently designing a board that uses the ESP32-S3-Mini-1 and a Tang Mega 60 and would love to be able to use the code from [esp-usb-bridge](https://github.com/espressif/esp-usb-bridge) along with openfpgaloader to program the FPGA.

emard, 2025-03-21:
Last time I remember it returns correct fpga chip id
but bitstream programming stops with some usb errors
So the basic mechanism is working but needs a bit
more fiddling to work. there is openocd code which
works with S3. Also you can check my https://github.com/emard/esp32ecp5
it works on S3 same as on classic ESP32. S3 USB can additional
run  https://github.com/emard/esp32ecp5/blob/master/dfu.py
so with any DFU client you can program FPGA and FLASH
very fast using USB

emard, 2025-03-21:
Ups only detail esp32ecp5 supports ecp5 artix7 cycloe5
no gw yet. openocd could work if board has jtag and bitstream
can be converted so .svf file

you can patch and compile openocd
https://github.com/emard/esp32s3-jtag/blob/main/openocd/espfix.patch
I did it on linux its easy

trabucayre, 2025-03-21:
@edanuff I confirm: basics functions are working (FPGA detect and idCode) but writing a full bitstream is not working. I try to find with small tests the cause of this failure. But not observed issue with USB.

trabucayre, 2025-03-23:
@edanuff @emard with latest commit I'm able to load a bitstream with ECP5 based board. Latest issue was related to the last TDI bit sent with tms transaction (SHIFTIR/SHIFTDR -> EXIT1IR/EXIT1DR).
I have now to test if it's working with SPI Flash, check for TDO and FLUSH command before merging this support.
Could you test in your side to confirm/infirm my results? Thanks.

edanuff, 2025-03-24:
> [@edanuff](https://github.com/edanuff) [@emard](https://github.com/emard) with latest commit I'm able to load a bitstream with ECP5 based board. 

Is that the [https://github.com/trabucayre/openFPGALoader/tree/esp_usb_jtag](esp_usb_jtag) branch? I'll give it a shot

emard, 2025-03-24:
It works for ECP5 25F, and it is very fast!

esp32s3 xiao mini is connected to jtag of ulx3s ecp5 25f
with arduino i compiled and uploaded https://github.com/emard/esp32s3-jtag to S3
I pulled openFPGALoader branch esp_usb_jtag, compiled

time /tmp/openFPGALoader/build/openFPGALoader -c esp32s3
ulx3s_25f_oscg_blink.bit
empty
 0x1 0xa 0x1 0x8 0x40 0x1f 0x1 0x0 0xff 0x0
esp_usb_jtag: Device found. Base speed 40000 KHz, div range 1 to 255
Jtag frequency : requested 6000000Hz -> real 6000000Hz divisor=6
Open file: DONE
b3bdffff
Parse file: DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.00%
Done
Disable configuration: DONE
drain_in

real    0m2,982s
user    0m0,008s
sys    0m0,124s

edanuff, 2025-03-24:
I'm getting the following:

```
% ./openFPGALoader --scan-usb                
empty
Bus device vid:pid       probe type      manufacturer serial               product
% ./openFPGALoader  -c esp32s3 --detect      
empty
fails to open esp_usb_jtag device vid:pid 0x303a:0x
JTAG init failed with: std::exception
```

however, if I change the pid to 1002 in cable.hpp to match the [Espressif USB JTAG bridge](https://github.com/espressif/esp-usb-bridge), I get this:

```
% ./openFPGALoader --scan-usb
empty
Bus device vid:pid       probe type      manufacturer serial               product
005 002    0x303a:0x1002 esp32s3         Espressif Systems Co. Ltd. 64E8335D11A0         ESP USB Bridge
% ./openFPGALoader  -c esp32s3 --detect
empty
fails to open esp_usb_jtag device vid:pid 0x303a:0x
JTAG init failed with: std::exception
% ./openFPGALoader  -c esp32s3 ../pmod_led.fs
empty
fails to open esp_usb_jtag device vid:pid 0x303a:0x
JTAG init failed with: std::exception
```

Not sure is the is due to differences between @emard's bridge versus the Espressif code on the ESP32?

edanuff, 2025-03-24:
@trabucayre I don't actually see the firmware in [https://github.com/emard/esp32s3-jtag](https://github.com/emard/esp32s3-jtag).  Is it the [esp32s3-jtag.ino](https://github.com/emard/esp32s3-jtag/blob/main/esp32s3-jtag.ino) file? If so, I wasn't able to get that to build.

emard, 2025-03-24:
.ino is the C source not binary
atart arduino ide and load this .ino some editor window will
open this file.

follow hints in .ino file comments to select your board
e.g. if it is XIAO MINI ESP32S3

Tools->Board->Boards Manager->esp32 by espressif
Tools->USB CDC On Boot: Enabled
Tools->CPU Frequency: 80 MHz (WiFi)
Tools->Board: XIAO_ESP32S3
Tools->JTAG Adapter: Integrated USB JTAG
Tools->USB Mode: Hardware CDC and JTAG

near top left corner of the window will be
a circular blue "button" with right arrow (->), click this
it will compile source and upload to your board

edanuff, 2025-03-24:
@emard thanks, I was able to get it to program.  I am using the ESP32-S3-DevKitM-1 board with this, however I do need to set the board as XIAO_ESP32S3 for it to compile in the Arduino IDE.  Otherwise, I get "error: no matching function for call to 'HardwareSerial::begin()'".  I'm not seeing anything significantly different between the two board types so it shoudl be okl to just keep the XIAO_ESP32S3 board type with this.

Once I do this, I get:

```
% ./openFPGALoader --scan-usb

empty
Bus device vid:pid       probe type      manufacturer serial               product
005 001    0x303a:0x1001 esp32s3         Espressif    64:E8:33:5D:11:A0    USB JTAG/serial debug unit
% ./openFPGALoader  -c esp32s3 --detect
empty
 0x1 0xa 0x1 0x8 0x40 0x1f 0x1 0x0 0xff 0x0
esp_usb_jtag: Device found. Base speed 40000 KHz, div range 1 to 255
Jtag frequency : requested 6000000Hz -> real 6000000Hz divisor=6
index 0:
	idcode 0x9604093
	manufacturer xilinx
	family xc9500xl
	model  xc9572xl
	irlength 8
drain_in
% ./openFPGALoader  -c esp32s3 ../../../GitHub/a2bridge/a2bridge.jed
empty
 0x1 0xa 0x1 0x8 0x40 0x1f 0x1 0x0 0xff 0x0
esp_usb_jtag: Device found. Base speed 40000 KHz, div range 1 to 255
Jtag frequency : requested 6000000Hz -> real 6000000Hz divisor=6
Open file DONE
Erase flash zsh: segmentation fault  ./openFPGALoader -c esp32s3 ../../../GitHub/a2bridge/a2bridge.jed
```

edanuff, 2025-03-24:
and trying with a different device (Tang Mega 60K), I get:

```
% ./openFPGALoader  -c esp32s3 --detect
empty
 0x1 0xa 0x1 0x8 0x40 0x1f 0x1 0x0 0xff 0x0
esp_usb_jtag: Device found. Base speed 40000 KHz, div range 1 to 255
Jtag frequency : requested 6000000Hz -> real 6000000Hz divisor=6
JTAG init failed with: Unknown device with IDCODE: 0x0001481b (manufacturer: 0x40d (Gowin), part: 0x00 vers: 0x0
```

which is probably to be expected since openfpgaloader doesn't support Gowin GW5AT-60 parts yet, only GW5AT-138.

edanuff, 2025-03-24:
One more try with a supported chip (Tang Primer 20K):

```
% ./openFPGALoader  -c esp32s3 --detect      
empty
 0x1 0xa 0x1 0x8 0x40 0x1f 0x1 0x0 0xff 0x0
esp_usb_jtag: Device found. Base speed 40000 KHz, div range 1 to 255
Jtag frequency : requested 6000000Hz -> real 6000000Hz divisor=6
index 0:
	idcode 0x81b
	manufacturer Gowin
	family GW2A
	model  GW2A(R)-18(C)
	irlength 8
drain_in
% ./openFPGALoader  -c esp32s3 ../uart_Hello.fs 
empty
 0x1 0xa 0x1 0x8 0x40 0x1f 0x1 0x0 0xff 0x0
esp_usb_jtag: Device found. Base speed 40000 KHz, div range 1 to 255
Jtag frequency : requested 6000000Hz -> real 6000000Hz divisor=6
Parse file Parse ../uart_Hello.fs: 
Done
DONE
Load SRAM: [==================================================] 100.00%
Done
DONE
xfer: usb bulk write failed with error -7
ESP USB Jtag drain_in failed
```

emard, 2025-03-24:
here is now binary release of esp32s3-jtag firmware
https://github.com/emard/esp32s3-jtag/tree/main/binary

For such simple source all boards are the same, some have onboard
LED, SERIAL some don't, it doesn't matter for JTAG, it's good enough to have
compiled binary for any ESP32S3 board.

Looking at issues, I think Gwen would have few small bugs to fix in order
to support different chips than ECP5 :) but I think we are close to getting
it fully
working.

emard, 2025-03-24:
> Hi and thanks for these tests. Gowin 60K is supported but in master
> branch: I have to rebase, squash commits.
> I have to also check or removes drain_in: seems not required.
> My idea is in first time to verify everything with ECP5 before starting to
> test with another FPGA.
> For the 20k drain_in fails at destructor level: but I suppose the FPGAs is
> not working?
> @emard <https://github.com/emard> indeed there is small bugs to fix,
> maybe not visible with ECP5.

For Lattice ECP5 on ULX3S programming FPGA RAM and FLASH works faster
than ever before :)

emard, 2025-03-24:
> @emard <https://github.com/emard> Great!
> Maybe I have to move to another device to check why it fails.

Yes try different boards, for ECP5 on ULX3S everything works great.

trabucayre, 2025-03-25:
@edanuff I have tested the esp32s3 with an board based on spartan6: I observe the same error with drain_in but I'm able to load and/or flash the bitstream successfully

trabucayre, 2025-03-25:
With https://github.com/trabucayre/openFPGALoader/commit/c71668a1a2fc11db60ae0554f7f5c88d85a6e734 the error is discarded when drain_in is used to flush ESP32 device (error code mean timeout)

trabucayre, 2025-03-25:
@edanuff I have found the issue with XC95x: some writeTDI calls are done with tx/tdi NULL, but tx/tdi was always used without checking status.
Fixed with https://github.com/trabucayre/openFPGALoader/commit/26037ebb16ee3e01a604aa30eb15d9a87108c61a
I think this commit will also fixes error with gowin FPGAs

edanuff, 2025-03-26:
Xilinx xc9572xl works.  Tang Primer 20K works.  Looking forward to the merge for the rest of the Gowin parts.

```
% ./openFPGALoader  -c esp32s3 --detect
empty
 0x1 0xa 0x1 0x8 0x40 0x1f 0x1 0x0 0xff 0x0
esp_usb_jtag: Device found. Base speed 40000 KHz, div range 1 to 255
Jtag frequency : requested 6000000Hz -> real 6000000Hz divisor=6
index 0:
	idcode 0x9604093
	manufacturer xilinx
	family xc9500xl
	model  xc9572xl
	irlength 8
% ./openFPGALoader  -c esp32s3 ../a2bridge-v2-03-03-2024.jed
empty
 0x1 0xa 0x1 0x8 0x40 0x1f 0x1 0x0 0xff 0x0
esp_usb_jtag: Device found. Base speed 40000 KHz, div range 1 to 255
Jtag frequency : requested 6000000Hz -> real 6000000Hz divisor=6
Open file DONE
Erase flash DONE
Write Flash: [==================================================] 100.00%
Done

```

```
% ./openFPGALoader  -c esp32s3 --detect                     
empty
 0x1 0xa 0x1 0x8 0x40 0x1f 0x1 0x0 0xff 0x0
esp_usb_jtag: Device found. Base speed 40000 KHz, div range 1 to 255
Jtag frequency : requested 6000000Hz -> real 6000000Hz divisor=6
index 0:
	idcode 0x81b
	manufacturer Gowin
	family GW2A
	model  GW2A(R)-18(C)
	irlength 8
% ./openFPGALoader  -c esp32s3 ../uart_Hello.fs             
empty
 0x1 0xa 0x1 0x8 0x40 0x1f 0x1 0x0 0xff 0x0
esp_usb_jtag: Device found. Base speed 40000 KHz, div range 1 to 255
Jtag frequency : requested 6000000Hz -> real 6000000Hz divisor=6
Parse file Parse ../uart_Hello.fs: 
Done
DONE
Load SRAM: [==================================================] 100.00%
Done
DONE
```

trabucayre, 2025-04-20:
Now esp32s3 support is merged to master. I think it time to close this issue.

emard, 2025-04-27:
I see some exception after programming,
probably flashing and verifying is OK just
some happy error after everything done successfuly

ofl-esp32s3 -c esp32s3 -f --verify esp32s3-ulx3s-jtag.bin
empty
write to flash
 0x1 0xa 0x1 0x8 0x40 0x1f 0x1 0x0 0xff 0x0
esp_usb_jtag: Device found. Base speed 40000 KHz, div range 1 to 255
Jtag frequency : requested 6000000Hz -> real 6000000Hz divisor=6
Open file DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
JEDEC ID: 0x9d6018
Detected: ISSI IS25LP128 256 sectors size: 128Mb
00000000 00000000 00000000 00
start addr: 00000000, end_addr: 00800000
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Done
Enable configuration: DONE
SRAM erase: DONE
JEDEC ID: 0x9d6018
Detected: ISSI IS25LP128 256 sectors size: 128Mb
Skip resetting device
Refresh: FAIL
displayReadReg
Config Target Selection : 0
SPIm Fail1
BSE Error Code
Preamble ERR
EXEC Error
Error: Failed to program FPGA: std::exception

emard, 2025-04-27:
here is the same file (8MB esp32 binary) flashing
to ulx3s flash. Normally this flash is not valid ecp5
bitstream maybe that's reason of final error but I see
some errors timeout during flashing

esp32s3 is module mounted directly on ulx3s and
few wires patched kinda early prototype

ofl-esp32s3 -c esp32s3 -f --verify esp32s3-ulx3s-jtag.bin
empty
write to flash
 0x1 0xa 0x1 0x8 0x40 0x1f 0x1 0x0 0xff 0x0
esp_usb_jtag: Device found. Base speed 40000 KHz, div range 1 to 255
Jtag frequency : requested 6000000Hz -> real 6000000Hz divisor=6
Open file DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
JEDEC ID: 0x9d6018
Detected: ISSI IS25LP128 256 sectors size: 128Mb
00000000 00000000 00000000 00
start addr: 00000000, end_addr: 00800000
Erasing: [==================================================] 100.00%
Done
Writing: [                                                  ] 0.00%timeout:
ff ff 1000
ff
wait: Error
timeout: ff ff 1000
ff
wait: Error
timeout: ff ff 1000
ff
wait: Error
Writing: [=                                                 ] 0.53%timeout:
ff ff 1000
ff
wait: Error
timeout: ff ff 1000
ff
wait: Error
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Done
Enable configuration: DONE
SRAM erase: DONE
JEDEC ID: 0x9d6018
Detected: ISSI IS25LP128 256 sectors size: 128Mb
Skip resetting device
Refresh: FAIL
displayReadReg
Config Target Selection : 0
SPIm Fail1
BSE Error Code
Preamble ERR
EXEC Error
Error: Failed to program FPGA: std::exception

trabucayre, 2025-04-28:
I have tried the master branch with one colorlight board successfully.
I have tried two cables:
- with the first cable: lot of errors
- with the second cable: no errors at all and flash correctly updated.

Maybe your issue is related to the cable and/or wires between esp32s3 and the ulx3s FPGA?

emard, 2025-04-28:
ok, it could be, I should make more tests.
Used cable was of a better quality,
so far those errors were not fatal and
verify passed

edanuff, 2025-06-08:
I'm seeing difficulties with the Tang Primer 25K using this.  Not sure how to debug.  I'm using it with the --bulk-erase option and the done line gets asserted almost immediately before programming starts.

edanuff, 2025-06-08:
Also, with the Tang Mega 60K, I can program the sram but not able to program the flash.

trabucayre, 2025-06-10:
It's weird: I have just tried with my Tang primer 25k and openFPGALoader mainline without any issues.
This you are able to load bitstream this mean the bl616 is correctly disabled.

edanuff, 2025-06-21:
> It's weird: I have just tried with my Tang primer 25k and openFPGALoader mainline without any issues. This you are able to load bitstream this mean the bl616 is correctly disabled.

It's working via the ESP32-based interface\? I'm doing this on my own board that just has an ESP32S3 and doesn't have a BL616.

edanuff, 2025-06-21:
> It's working via the ESP32-based interface? I'm doing this on my own board that just has an ESP32S3 and doesn't have a BL616.

Never mind, I had the wrong pins defined in the CST.  SRAM configuration on the Primer 25K is confirmed working on my board.

edanuff, 2025-06-21:
However, still not able to flash the Primer 25K (using openFPGALoader v0.13.1), only SRAM configuration works.

trabucayre, 2025-06-22:
Thanks for your feedback.
Will try for Flash

mohammed-nurulhoque, 2025-08-12:
https://github.com/trabucayre/openFPGALoader/issues/578

mohammed-nurulhoque, 2025-08-12:
openFPGALoader -c "esp32s3" --bitstream blink.bit --verbose
empty
fails to open esp_usb_jtag device vid:pid 0x303a:0x

even though the device can be seen by openocd.