trabucayre/openFPGALoader #507: Intel Max10: configuring the FPGA in volatile mode (.sof file) affects CFM

yacremann, 2025-01-10:
Dear all,

First of all thanks for the great project, it is very valuable for me!

I think I found a strange bug: I configure the FPGA (CRAM) on my DE10-Lite board using a .sof file from Quartus (converted to a .svf file using quartus_cpf). The new design is correctly loaded into the CRAM of the FPGA. However, it also affects the CFM or some other nonvolatile memory of the FPGA: A previously stored design on the CFM does not run after a power cycle. It looks like the Flash memory of the FPGA gets erased by configuring the CRAM of the Max10 using openFPGALoader.

Here the commands:
* Generate .svf fle: quartus_cpf -c -q 24.0MHz -g 3.3 -n p project_name.sof project_name.svf
* configure FPGA: openFPGALoader -b de10lite project_name.svf

The reason why I care: I use the boards for my lecture. The students will try many different configurations, but the FLASH memory of the FPGA will not survive an infinite number of write cycles.

The attached .zip file contains the svf file.

Thanks a lot!

Best,
Yves
[Stopwatch.zip](https://github.com/user-attachments/files/18376243/Stopwatch.zip)

trabucayre, 2025-01-11:
Hi,
I have also observed this situation.
When you use an `.svf` file, openFPGALoader act as a player: it read file's content line by line and execute instructions.
This issue isn't related to openFPGALoader by itself but to instructions/operations contained into the `.svf`.
If you read begining of the file you see DSM/ICB section is cleared and rewritten with a new content.
Since there is no documentation related to this section it's hard to know meaning of this operation.

trabucayre, 2025-01-11:
Not your svf but another I have just produces.
It contains:
```svf
!                                                                                                                                                                                                                                                                 
!Max 10 Program ICB                                             
!                
SIR 10 TDI (203);
RUNTEST 123 TCK;                                                                                                                                                                                                                                                
SDR 23 TDI (480000);                                                                                                                                                                                                                                               
SIR 10 TDI (3F4);                                                                         
RUNTEST 123 TCK; 
SDR 32 TDI (6C48A50F);
RUNTEST 7680 TCK;
!                
!Max 10 DSM Verify
!          
SIR 10 TDI (307);
RUNTEST 123 TCK;
SDR 1 TDI (0) TDO (1) MASK (1);
!
!Max 10 Program ICB
!
SIR 10 TDI (203);
RUNTEST 123 TCK;
SDR 23 TDI (680000);
SIR 10 TDI (3F4);
RUNTEST 123 TCK;
SDR 32 TDI (6C48A50F);
RUNTEST 7680 TCK;
```

First **Max 10 Program ICB** section match *Max 10 Program Done Bit* and second match *Max 10 ICB Program Success* found in an svf produces from the pof file.

yacremann, 2025-01-13:
Ah, OK. I apologize for missing that the issue is with the .svf file. I will try to find the problem there.
Thanks a lot for your help! And thanks for this great project!

Best,
Yves

closed 2025-01-13 by yacremann

trabucayre, 2025-01-14:
Don't worry. It's not really intuitive since svf is the only file type, known by openFPGALoader, where the logic is in the file instead of in the software. It's why I try as much as possible to adds native support.